bluez/bluez #1125: Intel AX211 fails to Pair on first attempt but succeeds on second attempt (utilizing software-controlled bluez API via Dbus)

millerH112, 2025-03-07:
Attempting to Pair with a BLE device fails on the first Pairing attempt, but succeeds on the second Pairing attempt when utilizing Bluez Dbus API.

When utilizing bluetoothctl directly, this issue does not occur.

Note that this issue does not arise when using Intel AC 9462, but only arises utilizing Intel AX211.

[bluetooth_debug_using_bluetoothctl_displayonly.log](https://github.com/user-attachments/files/19134548/bluetooth_debug_using_bluetoothctl_displayonly.log)
Bluetooth syslog utilizing bluetoothctl.

[bluetooth_debug_bluez_api_usage.log](https://github.com/user-attachments/files/19134547/bluetooth_debug_bluez_api_usage.log)
Bluetooth syslog utilizing Bluez API.

[computer_capture_btmon.log](https://github.com/user-attachments/files/19134594/computer_capture_btmon.log)
Wireshark capture when using Bluez API.

From the syslog when using Bluez API, the only difference I can see is that the discover_incl_cb() method is called directly after discovery_parse_services(), while discover_incl_cb() is called after bonding has been completed when utilizing bluetoothctl.

I'm unsure where to go from here, but open to suggestions on how to debug this issue further.

millerH112, 2025-03-12:
```
Bluetooth monitor ver 5.53
= Note: Linux version 6.13.2-arch1-1 (x86_64)                                                                                                                                                             0.820009
= Note: Bluetooth subsystem version 2.22                                                                                                                                                                  0.820012
= New Index: D4:F3:2D:33:29:8E (Primary,USB,hci0)                                                                                                                                                  [hci0] 0.820013
= Open Index: D4:F3:2D:33:29:8E                                                                                                                                                                    [hci0] 0.820013
= Index Info: D4:F3:2D:33:29:8E (Intel Corp.)                                                                                                                                                      [hci0] 0.820014
@ MGMT Open: bluetoothd (privileged) version 1.23                                                                                                                                                {0x0001} 0.820015
> HCI Event: LE Meta Event (0x3e) plen 45                                                                                                                                                       #1 [hci0] 0.916203
      LE Extended Advertising Report (0x0d)
        Num reports: 1
        Entry 0
          Event type: 0x0013
            Props: 0x0013
              Connectable
              Scannable
              Use legacy advertising PDUs
            Data status: Complete
          Legacy PDU Type: ADV_IND (0x0013)
          Address type: Public (0x00)
          Address: CC:86:EC:61:9F:B6 (OUI CC-86-EC)
          Primary PHY: LE 1M
          Secondary PHY: No packets
          SID: no ADI field (0xff)
          TX power: 127 dBm
          RSSI: -73 dBm (0xb7)
          Periodic advertising invteral: 0.00 msec (0x0000)
          Direct address type: Public (0x00)
          Direct address: 00:00:00:00:00:00 (OUI 00-00-00)
          Data length: 0x13
        02 01 06 03 03 12 18 0b 09 48 48 54 20 52 65 6d  .........HHT Rem
        6f 74 65                                         ote             
< HCI Command: LE Set Extended Scan Enable (0x08|0x0042) plen 6                                                                                                                                 #2 [hci0] 0.916245
        Extended scan: Disabled (0x00)
        Filter duplicates: Disabled (0x00)
        Duration: 0 msec (0x0000)
        Period: 0.00 sec (0x0000)
> HCI Event: Command Complete (0x0e) plen 4                                                                                                                                                     #3 [hci0] 0.920173
      LE Set Extended Scan Enable (0x08|0x0042) ncmd 2
        Status: Success (0x00)
< HCI Command: LE Extended Create Connection (0x08|0x0043) plen 26                                                                                                                              #4 [hci0] 0.920207
        Filter policy: White list is not used (0x00)
        Own address type: Public (0x00)
        Peer address type: Public (0x00)
        Peer address: CC:86:EC:61:9F:B6 (OUI CC-86-EC)
        Initiating PHYs: 0x01
        Entry 0: LE 1M
          Scan interval: 60.000 msec (0x0060)
          Scan window: 60.000 msec (0x0060)
          Min connection interval: 30.00 msec (0x0018)
          Max connection interval: 50.00 msec (0x0028)
          Connection latency: 0 (0x0000)
          Supervision timeout: 30000 msec (0x0bb8)
          Min connection length: 0.000 msec (0x0000)
          Max connection length: 0.000 msec (0x0000)
> HCI Event: Command Status (0x0f) plen 4                                                                                                                                                       #5 [hci0] 0.921162
      LE Extended Create Connection (0x08|0x0043) ncmd 2
        Status: Success (0x00)
> HCI Event: LE Meta Event (0x3e) plen 31                                                                                                                                                       #6 [hci0] 1.920199
      LE Enhanced Connection Complete (0x0a)
        Status: Success (0x00)
        Handle: 2048
        Role: Master (0x00)
        Peer address type: Public (0x00)
        Peer address: CC:86:EC:61:9F:B6 (OUI CC-86-EC)
        Local resolvable private address: 00:00:00:00:00:00 (Non-Resolvable)
        Peer resolvable private address: 00:00:00:00:00:00 (Non-Resolvable)
        Connection interval: 45.00 msec (0x0024)
        Connection latency: 0 (0x0000)
        Supervision timeout: 30000 msec (0x0bb8)
[... 933 lines cut ...]
        Count: 1
> ACL Data RX: Handle 2048 flags 0x02 dlen 5                                                                                                                                                 #184 [hci0] 44.369148
      ATT: Write Response (0x13) len 0
> ACL Data RX: Handle 2048 flags 0x02 dlen 8                                                                                                                                                 #185 [hci0] 44.369151
      ATT: Handle Value Notification (0x1b) len 3
[... cut ...]
From the (shorter) btmon log attached above, it seems that LE Start Encryption occurs in # 40 (2.293188), followed by success in # 42 (2.294177). However, after the supervision timeout of 30s, a disconnect is sent # 86 (32.619238). 

In the second pair attempt, LE Start Encryption occurs in # 142 (37.979552), followed by an Encryption Change event in # 143 (38.157560). Following that, a new long term key event is notified, and pairing succeeds.

Seems as though the Encryption Change event (0x08) only occurs on the second attempt.

Vudentz, 2025-03-19:
> From the (shorter) btmon log attached below, it seems that LE Start Encryption occurs in # 40 (2.293188), followed by success in # 42 (2.294177). However, after the supervision timeout of 30s, a disconnect is sent # 86 (32.619238).
> 
> In the second pair attempt, LE Start Encryption occurs in # 142 (37.979552), followed by an Encryption Change event in # 143 (38.157560). Following that, a new long term key event is notified, and pairing succeeds.
> 
> Seems as though the Encryption Change event (0x08) only occurs on the second attempt.

Pretty much, so to debug this further we would need access to the Link-Layer, which you probably don't have since that would require an air sniffer, in any case this is not host stack (BlueZ's) fault as the same sequence sometimes works.

millerH112, 2025-03-20:
Air sniffer failed pair attempt attached (recommend filter btle.length != 0 in wireshark to filter out empty PDUs)

Vudentz, 2025-03-20:
> [air_sniffer_failed_pair_attempt.log](https://github.com/user-attachments/files/19373613/air_sniffer_failed_pair_attempt.log) Air sniffer failed pair attempt attached (recommend filter btle.length != 0 in wireshark to filter out empty PDUs).
> 
> [air_sniffer_successful_pair_(pair_attempt_2).log](https://github.com/user-attachments/files/19373640/air_sniffer_successful_pair_.pair_attempt_2.log) Air sniffer successful pair attempt (second Pair() attempt via Dbus API).
> 
> From what I can see, Master (computer with AX211) does not send LL_ENC_REQ after Slave receives Pairing DHKey Check when pairing fails (failed pair attempt No. 182 and onward).
> 
> When pairing succeeds, Master sends LL_ENC_REQ after Slave receives Pairing DHKey Check (see successful pair attempt No. 67 and No. 70).
> 
> From further testing, I have verified that this issue _does_ in fact occur using bluetoothctl. May not be BlueZ's fault in this case but a Kernel/Driver issue. Would you like me to "Close as not planned"?

It doesn't seem these are btsnoop files, what air sniffer did you use? In any case this will likely need to forwarded to the manufacturer since we have no control over LL_ENC_REQ.

millerH112, 2025-03-20:
Github does not allow uploading .snoop files. You should be able to open the .log file in wireshark (or rename the filetype to .pcapng).

Air sniffer used was nRF52840 Dongle.

Vudentz, 2025-03-21:
> Github does not allow uploading .pcapng files. You should be able to open the .log file in wireshark (or rename the filetype to .pcapng).
> 
> Air sniffer used was nRF52840 Dongle.

Got, was able to decode and indeed there is no LL_ENC_REQ which shall be the result of LE Start Encryption, looks like the controller is just not doing anything with it, we could perhaps attempt to retry but there doesn't seem to be anything we can use as time out.

Vudentz, 2025-03-21:
> After further testing, issue does occur utilizing bluetoothctl as well. On the first Pair attempt, org.bluez.Error.AuthenticationCanceled occurs.
> 
> On the second attempt, pairing is successful.
> 
> This issue can be duplicated by removing the /var/lib/bluetooth/// folder and restarting the bluetooth.service (systemctl restart bluetooth).
> 
> Seems that without the cached attributes/info, this issue arises. However, if the BLE peripheral attributes/info is stored in /var/lib/bluetooth/// folder, the issue does not occur.

Well this might just be because there will be a lot more traffic since GATT discovery will take place, that said this should make the controller to just ignore the likes of LE Start Encryption.

millerH112, 2025-03-21:
[air_sniffer_with_intel_ac9462.log](https://github.com/user-attachments/files/19391386/air_sniffer_with_intel_ac9462.log)
Here's the air sniffer log with the computer (Master) using Intel AC9462. 

Seems to be _something_ with the GATT discovery on the Intel AX211 specifically - as the ac9462 performs the GATT secondary service group request after the LL_ENC_REQ/LL_START_ENC_RSP while the AX211 seems to start that secondary GATT discovery before LL_ENC_REQ (which never gets sent to the slave).

Vudentz, 2025-03-21:
> [air_sniffer_with_intel_ac9462.log](https://github.com/user-attachments/files/19391386/air_sniffer_with_intel_ac9462.log) Here's the air sniffer log with the computer (Master) using Intel AC9462. [...]

Without having the HCI trace alongside the LL message you can't really tell when LE Start Encryption was sent, and btw it is actually recommended to use Device.Connect method rather than Device.Pair for LE since you don't necessarily need to start with encryption right off the bat.

Another remark is that we don't enforce any ordering for different channels so SMP and ATT traffic can happen in any sequence really, what happens is that if ATT receives a error response that indicates it needs encryption/authentication then it will increase the security level of the ATT socket which will block any ongoing traffic until the security level change is considered complete, however like I explained above Device.Pair does start the pairing procedure in parallel to GATT discovery.

We could perhaps intentionally block the ATT channel if Device.Pair method is used so it will act similarly to Device.Connect with respect to not sending any other traffic while pairing in pending.

Vudentz, 2025-03-24:
@millerH112 btw what is the firmware version (you can see that in dmesg).

millerH112, 2025-03-24:
uname -r
6.13.7-arch1-1

linux-firmware version 20250311
(arch linux version 20250311.b69d4b74-3)

Bluetooth: hci0: Firmware timestamp 2024.48 buildtype 1 build 81864
Bluetooth: hci0: Found device firmware: intel/ibt-0040-0041.sfi
Bluetooth: hci0: Boot Address: 0x100800
Bluetooth: hci0: Firmware Version: 200-48.24

No success with changes attached. Bluetoothctl receives org.bluez.Error.AuthenticationCanceled. 

Pairing operation still works on second attempt when GATT services are in cache.

Vudentz, 2025-03-24:
> uname -r 6.13.7-arch1-1
>
> linux-firmware version 20250311 (arch linux version 20250311.b69d4b74-3)
>
> Bluetooth: hci0: Firmware timestamp 2024.48 buildtype 1 build 81864 Bluetooth: hci0: Found device [...]

Do you have a script or something you are running? Or perhaps that is reproducible with the likes of bluetoothctl?

millerH112, 2025-03-24:
In this case, no script. The only script utilized to remove the bluetooth cache is as follows:
```
rm -r /var/lib/bluetooth/ && sleep 1
systemctl restart bluetooth
```

Utilizing bluetoothctl, here are the commands used
Connecting (bluetoothctl pairs after accept pairing prompt):
```
bluetoothctl
scan le
scan off
devices
connect [ble address]
... wait for [agent] Accept pairing (yes/no): yes
```

Pairing
```
bluetoothctl
scan le
scan off
devices
pair [ble address]
```

This issue arises with both scenarios on AX211 when the bluetooth cache is removed. 

Some further testing has shown that pointing the BLE peripheral's module away from the Master utilizing AX211 results in successful pairing. Module used is BGM220-EK4314A.

While this is just a theory, it seems the speed of communication affects the pairing process success/failure with AX211 when it needs to perform GATT discovery. When the module is very close to the computer (within ~1 foot or within ~3 foot but pointed directly at the computer), the first pairing process fails.

Vudentz, 2025-03-24:
> In this case, no script. The only script utilized to remove the bluetooth cache is as follows:
>
> ```
> rm -r /var/lib/bluetooth/ && sleep 1
> systemctl restart bluetooth
> ```
>
> Utilizing bluetoothctl, here [...]

Not following, what are there 2 separate steps for connecting an pairing? I though the problem was connect followed by pair would fail, but you are saying that connect is already triggering pairing so why is there another step for pair?

millerH112, 2025-03-24:
Apologies, I meant that I have tried both ways, not that I perform 2 separate steps for connecting and pairing.

The issue occurs whether I use "connect [device]" in bluetoothctl _or_ "pair [device]" in bluetoothctl.

millerH112, 2025-03-24:
BlueZ does indicate an error if there is a Pair operation ongoing (so that is working correctly).

With this issue specifically, it's more so the Pair operation waits/hangs until the org.bluez.Error.AuthenticationCanceled occurs. The second attempt then works.

Again, I'm not 100% sure this may be a BlueZ issue after all since this works with Intel AC 9462 but the issue pops up with the Intel AX211.

Vudentz, 2025-03-24:
@millerH112 ok, try with the changes above, that should in theory make Connect/Pair combination detect if one of them is already taken place.

millerH112, 2025-03-26:
[air_sniffer_with_patch.log](https://github.com/user-attachments/files/19468224/air_sniffer_with_patch.log)

Issue still persists when peripheral is close to master computer or ble module pointed directly at computer within ~3ft.

Attached air sniffer contains 3 successful attempts followed by 2 unsuccessful attempts to connect and pair. Note that the 2 unsuccessful attempts occurred after moving the module within 1ft of the computer (have not clarified before, but it is a RNUC14MNK150000U).

Attempt 1 success: Packet # 167 (time 12.830)
Attempt 2 success: Packet # 1232 (time 52.658)
Attempt 3 success: Packet # 1828 (time 87.767)

Attempt 4 failure: Packet # 2416 (time 121.697) - Slave receives DHKey Check but no LL_ENC_REQ.
Attempt 5 failure: Packet # 3954 (time 181.124) - Slave receives DHKey Check but no LL_ENC_REQ.

Vudentz, 2025-03-26:
> [air_sniffer_with_patch.log](https://github.com/user-attachments/files/19468224/air_sniffer_with_patch.log)
>
> Issue still persists when peripheral is close to master computer or ble module pointed [...]

Ok, so I guess the changes from https://github.com/tedd-an/bluez-upstream-test/commit/3756466f97a44387bfc67d64576a4b3a251fd901 didn't make any difference, or didn't apply them?

Also perhaps in case of Device.Pair specifically the problem is that it doesn't elevate the security on GATT/ATT channel in order to block its traffic while bonding is taking place.

millerH112, 2025-03-26:
Changes did not make any difference from tedd-an/bluez-upstream-test@3756466. Patch was applied to Bluez 5.80.

Vudentz, 2025-03-26:
@millerH112 how about: https://github.com/BluezTestBot/bluez/commit/d4f0adf8e68d4f1d14e6edd347c34432c6618a4a

Vudentz, 2025-03-26:
> Further testing required, but when using bluetoothctl's pair [device] method I am seeing
> 
> ```
> gatt_client_init() Elevating security level since bonding is in progress
> ```
> 
> followed by
> 
> ```
> new_auth() Requesting agent authentication for [device MAC address]
> ```
> 
> This patch improved the pair [device] method success quite a bit (I'll have to test how often this succeeds versus before the patch), but in the instances where pairing does not succeed, it seems the authorization does not succeed(?) as
> 
> ```
> device_cancel_authentication() Canceling authentication request for [device MAC address]
> ```
> 
> occurs some time after the
> 
> ```
> new_auth() Requesting agent authentication for [device MAC address]
> ```
> 
> With bluetoothctl's connect [device] method, same logs occur, but
> 
> ```
> gatt_client_init() Elevating security level since bonding is in progress
> ```
> 
> is not seen in the logs.
> 
> I will follow-up with further logs/testing when possible here.

Well the pairing requires can timeout (30 seconds) if the agent doesn't respond it will then cancel the request.

Vudentz, 2025-03-27:
@millerH112 btw, you can use bluetoothctl -a auto in order to get pairing automatically accepted for testing multiple rounds.

millerH112, 2025-03-27:
I have been using bluetooth -a auto and can confirm src/adapter.c btd_adapter_confirm_reply() occurs with success 1. From my previous comment, I was watching only src/device.c so missed the the btd_adapter_confirm_reply().

After testing further, unfortunately I'm still having the issue when the peripheral is within ~1ft of the computer. Does not seem to matter whether connect [device] or pair [device] is used. 

When the device does not pair/bond correctly and pair [device] is used, org.bluez.Error.AuthenticationCanceled message is notified in bluetoothctl. 

When the device does not pair/bond correctly and connect [device] is used, no bluez Error message is notified in bluetoothctl. Properties of the peripheral just change to ServicesResolved: no and Connected: no. 

I've attempted multiple versions of linux-firmware with the same result. At this time, the workaround to re-attempt pairing works. 

When I can find the time to do so, I'll see if this issue occurs on Windows as well. If so, I'll have to chalk it up to the firmware for the AX211.

Vudentz, 2025-03-27:
> I have been using bluetooth -a auto and can confirm src/adapter.c btd_adapter_confirm_reply() occurs with success 1. From my previous comment, I was watching only src/device.c so missed the the [...]

Ok, but are you sure then it is the same issue and not the controller never establishing a connection to begin with? I would consider that to be different than missing LL_ENC_REQ, which is what we are attempt to fix here, if there is a problem with the connection itself then Im afraid there is nothing we can do at host level.

millerH112, 2025-03-27:
The device always connects successfully (bluez always sends out the notification Connected: yes).

It's the pairing itself that is the issue. 

Utilizing the nRF Connect app on a Samsung Galaxy Note 9, I cannot recreate the issue (though there's no way to change the IO capability of the mobile app to NoInputNoOutput or DisplayOnly - unsure if that would affect it).

Vudentz, 2025-03-27:
> The device always connects successfully (bluez always sends out the notification Connected: yes).
>
> It's the pairing itself that is the issue.
>
> Utilizing the nRF Connect app on a Samsung Galaxy Note [...]

Ok, I guess we are running out of options then, but if this improves the situation Id considering closing merging these changes and work with the controller vendor directly.

closed 2025-03-27 by BluezTestBot

ValeZAA, 2025-03-29:
Looking into: https://wikidevi.wi-cat.ru/Intel_Wi-Fi_6_AX_series

So, it is CNVio2 (that means the bug here may be inside the CPU/PCH even). It is Garfield Peak 2.

There is a new firmware for it from 12 days ago https://web.git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/commit/?id=47bc8a2407426274d099607d5af419cb616d9209

ValeZAA, 2025-05-08:
Can you retest on newer bluez, there were a lot of changes, 
>After testing further, unfortunately I'm still having the issue when the peripheral is within ~1ft of the computer. Does not seem to matter whether connect [device] or pair [device] is used.

also this is a duplicate:

 https://bugzilla.kernel.org/show_bug.cgi?id=220100 looks related SMP_IDENTITY_INFORMATION_COMMAND is send before HCI_ENCRYPTION_CHANGE_EVENT, while it should be after.