bluez/bluez #1127: br-connection-profile-unavailable

MatejKovacic, 2025-03-10:
I have a Bluetooth device I can not connect to with Linux (Ubuntu 24.04). I say
`bluetoothctl trust 00:12:F3:45:5F:D8` and got `Changing 00:12:F3:45:5F:D8 trust succeeded`.

Then I say: `bluetoothctl connect 00:12:F3:45:5F:D8` and got this error:
```
Attempting to connect to 00:12:F3:45:5F:D8
[CHG] Device 5D:CB:84:E8:17:01 RSSI: 0xffffffd0 (-48)
[CHG] Device 00:12:F3:45:5F:D8 Connected: yes
[CHG] Device F4:6A:DD:31:E9:B2 RSSI: 0xffffffa0 (-96)
[CHG] Device 00:12:F3:45:5F:D8 ServicesResolved: yes
Failed to connect: org.bluez.Error.NotAvailable br-connection-profile-unavailable
```

I also tried with `bt-device -c 00:12:F3:45:5F:D8` and also got error:
```
Connecting to: 00:12:F3:45:5F:D8
Error: GDBus.Error:org.bluez.Error.AlreadyExists: Already Exists
```

Sorry for my newbie question, can you at least point me to what could be a problem?

fanuware, 2025-03-14:
I am facing the same issues, which in my case are related to the profiles (Discussions #1115).

For instance, having a GNOME Desktop on Ubuntu usually comes with audio profiles, and it connects correctly. You can then isolate down to server mode (`systemctl isolate multi-user.target`), which does not provide audio, resulting in the `br-connection-profile-unavailable`.

woodard, 2025-04-05:
I am also having this problem with a Kenwood TH-D75 Ham radio. I filed a bug about this on Fedora but I misattributed it to first wireplumber and then pipewire before finding this bug. https://bugzilla.redhat.com/show_bug.cgi?id=2356565 

The TH-D75 presents two Bluetooth profiles the HSP and the SPP profile. I am pretty sure that this problem has to do with wiring up the HSP because I am able to connect to the SPP using the deprecated rfcomm utility. What I think is happening is that the the HSP fails and then neither the HSP nor the SPP gets connected. To be able to use the radio to its fullest capability, both the HSP and the SPP need to be wired up. The HSP protocol is used to send audio back and forth to the radio and SPP is variously used for data to and from the built-in modem, controlling the radio like virtually pressing the PTT push to talk button or setting its state.

woodard, 2025-04-05:
Here is some debugging datat:
```
$ bluetoothctl 
[WH-CH720N]# hci0 new_settings: powered bondable ssp br/edr le secure-conn wide-band-speech cis-central cis-peripheral 
[WH-CH720N]# Agent registered
[WH-CH720N]# [CHG] Controller F4:A4:75:07:EE:95 Pairable: yes
[WH-CH720N]# devices
Device 10:CA:BF:54:9C:B2 TH-D75
Device 00:A4:1C:90:C1:19 WH-CH720N
Device D5:BB:A6:53:61:3C MX Keys Mini
Device D1:1C:17:DF:01:7E MX Anywhere 3
[WH-CH720N]# show
Controller F4:A4:75:07:EE:95 (public)
	Manufacturer: 0x0002 (2)
	Version: 0x0c (12)
	Name: pun
	Alias: pun
	Class: 0x007c010c (8126732)
	Powered: yes
	PowerState: on
	Discoverable: no
	DiscoverableTimeout: 0x000000b4 (180)
	Pairable: yes
	UUID: Message Notification Se.. (00001133-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (03b80e5a-ede8-4b33-a751-6ce34ec4c700)
	UUID: OBEX Object Push          (00001105-0000-1000-8000-00805f9b34fb)
	UUID: Message Access Server     (00001132-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	UUID: IrMC Sync                 (00001104-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (00005005-0000-1000-8000-0002ee000001)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Phonebook Access Server   (0000112f-0000-1000-8000-00805f9b34fb)
	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree Audio Gateway   (0000111f-0000-1000-8000-00805f9b34fb)
	UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
	UUID: OBEX File Transfer        (00001106-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
	Modalias: usb:v1D6Bp0246d054F
	Discovering: no
	Roles: central
	Roles: peripheral
Advertising Features:
	ActiveInstances: 0x00 (0)
	SupportedInstances: 0x0c (12)
	SupportedIncludes: tx-power
	SupportedIncludes: appearance
	SupportedIncludes: local-name
	SupportedSecondaryChannels: 1M
	SupportedSecondaryChannels: 2M
	SupportedSecondaryChannels: Coded
	SupportedCapabilities.MinTxPower: 0xffffffde (-34)
	SupportedCapabilities.MaxTxPower: 0x000a (10)
	SupportedCapabilities.MaxAdvLen: 0xfb (251)
	SupportedCapabilities.MaxScnRspLen: 0xfb (251)
	SupportedFeatures: CanSetTxPower
	SupportedFeatures: HardwareOffload
[WH-CH720N]# info 10:CA:BF:54:9C:B2 
Device 10:CA:BF:54:9C:B2 (public)
	Name: TH-D75
	Alias: TH-D75
	Class: 0x00620204 (6423044)
	Icon: phone
	Paired: yes
	Bonded: yes
	Trusted: yes
	Blocked: no
	Connected: no
	LegacyPairing: no
	UUID: Serial Port               (00001101-0000-1000-8000-00805f9b34fb)
	UUID: Headset AG                (00001112-0000-1000-8000-00805f9b34fb)
	UUID: Generic Audio             (00001203-0000-1000-8000-00805f9b34fb)
[WH-CH720N]# connect 10:CA:BF:54:9C:B2 
Attempting to connect to 10:CA:BF:54:9C:B2
[WH-CH720N]# [CHG] Device 10:CA:BF:54:9C:B2 Connected: yes
[WH-CH720N]# [CHG] Device 10:CA:BF:54:9C:B2 ServicesResolved: yes
[WH-CH720N]# Failed to connect: org.bluez.Error.NotAvailable br-connection-profile-unavailable
[WH-CH720N]# [CHG] Device 10:CA:BF:54:9C:B2 ServicesResolved: no
[WH-CH720N]# [CHG] Device 10:CA:BF:54:9C:B2 Connected: no
```

woodard, 2025-04-05:
This is the output of btmon during a connection attempt:
[btmon.txt](https://github.com/user-attachments/files/19618341/btmon.txt)

woodard, 2025-04-06:
Note I threw my bluetoothd into debug mode and did a `journalctl -u bluetooth -f` while trying to connect and bluetoothd didn't print anything during the connection attempt.

woodard, 2025-04-06:
reading the source code:
I have yet to figure out where: 112 and "Host is down" are defined.
However, i did look at src/device.c and search_cb is a static function is only called from one place: https://github.com/bluez/bluez/blob/master/src/device.c#L5662 in browse_cb() 
browse_cb() is also a static function and it seems have something to do with processing the data that was provided as the host and the client exchanged data with the SDP, Service Discovery Protocol. This leads me to the hypothesis that "Host is down" may be referring to the Linux host and that it cannot host a HSP.

nullr0ute, 2025-04-06:
> reading the source code: I have yet to figure out where: 112 and "Host is down" are defined.

It's a generic Linux kernel error:
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/include/uapi/asm-generic/errno.h#n95

pv, 2025-04-06:
If you don't see HSP UUIDs in `bluetoothctl show` (UUID: Headset and UUID: Headset AG), it is not enabled.

If you use Pipewire, as explained in documentation HSP is not enabled by default, as several devices don't fail to work properly with both HFP and HSP enabled. If you want it, enable it:
```
# ~/.config/wireplumber/wireplumber.conf.d/enable-hsp.conf`
monitor.bluez.properties = { bluez5.roles = [ a2dp_sink a2dp_source bap_sink bap_source bap_bcast_sink bap_bcast_source hfp_hf hfp_ag hsp_hs hsp_ag ] }
```

woodard, 2025-04-10:
That might not be the complete solution:

I created the file, rebooted the system (because just creating the file and logging out and logging back in didn't seem to make the change stick)
and now I have the HSP and the HSP AG:

```
> show
Controller F4:A4:75:07:EE:95 (public)
	Manufacturer: 0x0002 (2)
	Version: 0x0c (12)
<snip>
	UUID: Headset AG                (00001112-0000-1000-8000-00805f9b34fb)
	UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
<snip>
```

but still:

```
> devices 
Device 10:CA:BF:54:9C:B2 TH-D75
Device 00:A4:1C:90:C1:19 WH-CH720N
Device D5:BB:A6:53:61:3C MX Keys Mini
Device D1:1C:17:DF:01:7E MX Anywhere 3
[MX Keys Mini]> connect 10:CA:BF:54:9C:B2
Attempting to connect to 10:CA:BF:54:9C:B2
[CHG] Device 10:CA:BF:54:9C:B2 Connected: yes
[CHG] Device 10:CA:BF:54:9C:B2 ServicesResolved: yes
Failed to connect: org.bluez.Error.NotAvailable br-connection-profile-unavailable
[CHG] Device 10:CA:BF:54:9C:B2 ServicesResolved: no
[CHG] Device 10:CA:BF:54:9C:B2 Connected: no
```

Could it be being blocked by this logic in /usr/share/wireplumber/scripts/monitors/bluez.lua
```
  -- Ignore devices that don't support both A2DP sink and HSP/HFP profiles
  local has_a2dpsink_profile = false
  local has_headset_profile = false
  for p in dev:iterate_params("EnumProfile") do
    local profile = cutils.parseParam (p, "EnumProfile")
    if profile.name:find ("a2dp") and profile.name:find ("sink") then
      has_a2dpsink_profile = true
    elseif profile.name:find ("headset") then
      has_headset_profile = true
    end
  end
  if not has_a2dpsink_profile or not has_headset_profile then
    return
  end
```

The failure to wire up the audio device also seems to continue to block subsequent connection to the SPP profile which should provide a virtual serial port which could be used for what we call "rig control". It is a serial protocol which allows you to set parameters on the radio like the frequency. 

I would also like to find out a way to enable HSP for either just this one device or to modify the logic so that if the device supports HFP and HSP it won't wire up HSP unless there is no support for HFP.  That way it won't break my Sony headset.  Where would I find documentation for how to limit it that way? Of course after your comment I found: https://pipewire.pages.freedesktop.org/wireplumber/daemon/configuration/bluetooth.html but looking through the rest of the documentation I have been unable to find anything that tells me how to limit the connections that way. Given all the problems with HSP and HFP, this seems like a reasonable request.

woodard, 2025-04-10:
Just noting that commenting out that entire block of code from bluez.lua and rebooting still doesn't allow the radio to connect.  

In the btmon output I did see:
 ```
<snip>
@ MGMT Event: Device Connected (0x000b) plen 21        {0x0001} [hci0] 9.860539
        BR/EDR Address: 10:CA:BF:54:9C:B2 (Texas Instruments)
        Flags: 0x00000008
          Connection Locally Initiated
        Data length: 8
        Name (complete): TH-D75
<snip>
> HCI Event: Number of Completed Packets (0x13) plen 5    #349 [hci0] 12.200559
        Num handles: 1
        Handle: 256 Address: 10:CA:BF:54:9C:B2 (Texas Instruments)
        Count: 1
        #346: len 12 (12 Kb/s)
        Latency: 8 msec (2-22 msec ~6 msec)
> ACL Data RX: Handle 256 flags 0x02 dlen 12              #350 [hci0] 12.203518
      L2CAP: Disconnection Response (0x07) ident 5 len 4
        Destination CID: 64
        Source CID: 64
<snip>
< HCI Command: Disconnect (0x01|0x0006) plen 3            #369 [hci0] 14.254686
        Handle: 256 Address: 10:CA:BF:54:9C:B2 (Texas Instruments)
        Reason: Remote User Terminated Connection (0x13)
> HCI Event: Command Status (0x0f) plen 4                 #370 [hci0] 14.255604
      Disconnect (0x01|0x0006) ncmd 1
        Status: Success (0x00)
> HCI Event: Disconnect Complete (0x05) plen 4            #371 [hci0] 14.340653
        Status: Success (0x00)
        Handle: 256 Address: 10:CA:BF:54:9C:B2 (Texas Instruments)
        Reason: Connection Terminated By Local Host (0x16)
@ MGMT Event: Device Disconnected (0x000c) plen 8     {0x0001} [hci0] 14.340766
        BR/EDR Address: 10:CA:BF:54:9C:B2 (Texas Instruments)
        Reason: Connection terminated by local host (0x02)
```

I don't know why the radio would choose to disconnect and so I'm confused by this. I would readily accept that I'm misunderstanding the rfmon output.

woodard, 2025-04-10:
[rfmon.txt.gz](https://github.com/user-attachments/files/19677468/rfmon.txt.gz)
Here is the full rfmon output.

closed 2025-09-08 by github-actions[bot]

quantumvoid0, 2025-10-19:
have any of u found any fix? , i saw a post on a forum saying changing btusb.c alt value from 3 to 1 fixed it but nothing seems to solve my issue

quantumvoid0, 2025-10-19:
i found the solution , im on gentoo and i just had to compile pipewire with bluetooth , for pulseadio users get the pulseaudio-bluetooth package (on arch, on other distros idk the exact package name)