bluez/bluez #1129: not able to configure Tx MTU and Rx MTU using obex-client-tool

maheshtalewad, 2025-03-12:
Hi,

We are trying to configure Tx MTU and Rx MTU in obex using obex-client-tool. 
Looks like there is a way to configure Tx MTU and Rx MTU in obex-client-tool.
But when we give following command we are always getting connection refused error and not able to connect to the remote phone.

root@imx_vk:~# obex-client-tool -b -s EC:2E:98:F6:FD:12 -c 13 -d 10:EC:81:61:71:67 -i 512 -o 512
connect to 10:EC:81:61:71:67: Connection refused (111)

# obex-client-tool -b -s <LOCAL_BT_MAC> -c 13 -d <SERVER_BT_MAC> -i MTU -o MTU

steps we followed:
1. start bluetoothd and btmon
2. start bluetoothctl
3. connect to remote phone using bluetoothctl.
4. give below command on another terminal
	obex-client-tool -b -s EC:2E:98:F6:FD:12 -c 13 -d 10:EC:81:61:71:67 -i 512 -o 512
	
Kindly let us know on how to configure Tx MTU and Rx MTU for obex.

Thanks and regards,
Mahesh Vithal Talewad

Vudentz, 2025-03-12:
> Hi,
>
> We are trying to configure Tx MTU and Rx MTU in obex using obex-client-tool. Looks like there is a way to configure Tx MTU and Rx MTU in obex-client-tool. But when we give following command we [...]

We probably need the logs, usually we would recommend using obexctl since obex-client-tool is more of a lower layer testing tool.

maheshtalewad, 2025-03-13:
Hi Luiz,

Kindly checkout the attached logs.zip [bluetoothd, btmon, bluetoothctl, and obex-client-tool logs].
Is it possible to configure Tx MTU and Rx MTU using obexctl? Kindly let us know.

Thanks and regards,
Mahesh Vithal Talewad

[logs.zip](https://github.com/user-attachments/files/19222591/logs.zip)

Vudentz, 2025-03-17:
> Hi Luiz,
> 
> Kindly checkout the attached logs.zip [bluetoothd, btmon, bluetoothctl, and obex-client-tool logs]. Is it possible to configure Tx MTU and Rx MTU using obexctl? Kindly let us know.

No, obexctl doesn't allow changing the MTU.

maheshtalewad, 2025-04-28:
could you pls update?

closed 2025-09-20 by github-actions[bot]