bluez/bluez #1133: [Bluez 5.79] bluetoothctl prints out invalid handles for Characteristic Descriptors

XenoKovah, 2025-03-13:
We've created a teaching tool Peripheral based on Zephyr RTOS running on a nRF52840-DK. When we connect to it with bluetoothctl from a manually-installed BlueZ 5.79 release on Ubuntu 24.04, the handles printed only for Characteristic Descriptors (i.e. 0x2902 or 0x2904) are always invalid. E.g. 0xaa9e or 0xaa9d as shown below. Bluetoothctl (or dbus) knows the correct handles, which is why it assigned it names like `desc003e` for a thing which has a handle of 0x003e. But I suspect this is something like printing out an uninitialized variable or the like. I've also attached a pcap of the connection if you want to confirm actual handle values as enumerated behind the scenes.

[pcap_for_git_issue.pcap.tar.gz](https://github.com/user-attachments/files/19236340/pcap_for_git_issue.pcap.tar.gz)

```
[UVP01]# list-attributes 
Primary Service (Handle 0x002c)
	/org/bluez/hci0/dev_CA_FE_13_37_00_01/service002c
	00001825-0000-1000-8000-00805f9b34fb
	Object Transfer
Characteristic (Handle 0x003c)
	/org/bluez/hci0/dev_CA_FE_13_37_00_01/service002c/char003c
	00002ac6-0000-1000-8000-00805f9b34fb
	Object List Control Point
Descriptor (Handle 0xaa9e)
	/org/bluez/hci0/dev_CA_FE_13_37_00_01/service002c/char003c/desc003e
	00002902-0000-1000-8000-00805f9b34fb
	Client Characteristic Configuration
Characteristic (Handle 0x0039)
	/org/bluez/hci0/dev_CA_FE_13_37_00_01/service002c/char0039
	00002ac5-0000-1000-8000-00805f9b34fb
	Object Action Control Point
Descriptor (Handle 0xaa9e)
	/org/bluez/hci0/dev_CA_FE_13_37_00_01/service002c/char0039/desc003b
	00002902-0000-1000-8000-00805f9b34fb
	Client Characteristic Configuration
Characteristic (Handle 0x0037)
	/org/bluez/hci0/dev_CA_FE_13_37_00_01/service002c/char0037
	00002ac4-0000-1000-8000-00805f9b34fb
	Object Properties
Characteristic (Handle 0x0035)
	/org/bluez/hci0/dev_CA_FE_13_37_00_01/service002c/char0035
	00002ac3-0000-1000-8000-00805f9b34fb
	Object ID
Characteristic (Handle 0x0033)
	/org/bluez/hci0/dev_CA_FE_13_37_00_01/service002c/char0033
	00002ac0-0000-1000-8000-00805f9b34fb
	Object Size
Characteristic (Handle 0x0031)
	/org/bluez/hci0/dev_CA_FE_13_37_00_01/service002c/char0031
	00002abf-0000-1000-8000-00805f9b34fb
	Object Type
Characteristic (Handle 0x002f)
	/org/bluez/hci0/dev_CA_FE_13_37_00_01/service002c/char002f
	00002abe-0000-1000-8000-00805f9b34fb
	Object Name
Characteristic (Handle 0x002d)
	/org/bluez/hci0/dev_CA_FE_13_37_00_01/service002c/char002d
	00002abd-0000-1000-8000-00805f9b34fb
	OTS Feature
Primary Service (Handle 0x001a)
	/org/bluez/hci0/dev_CA_FE_13_37_00_01/service001a
	444d0000-4865-6d6c-6f63-6b21f09fa4aa
	Vendor specific
Characteristic (Handle 0x002a)
	/org/bluez/hci0/dev_CA_FE_13_37_00_01/service001a/char002a
	444d0008-4865-6d6c-6f63-6b21f09fa4aa
	Vendor specific
Characteristic (Handle 0x0028)
	/org/bluez/hci0/dev_CA_FE_13_37_00_01/service001a/char0028
	444d0007-4865-6d6c-6f63-6b21f09fa4aa
	Vendor specific
Characteristic (Handle 0x0026)
	/org/bluez/hci0/dev_CA_FE_13_37_00_01/service001a/char0026
	444d0006-4865-6d6c-6f63-6b21f09fa4aa
	Vendor specific
Characteristic (Handle 0x0024)
	/org/bluez/hci0/dev_CA_FE_13_37_00_01/service001a/char0024
	444d0005-4865-6d6c-6f63-6b21f09fa4aa
	Vendor specific
Characteristic (Handle 0x0022)
	/org/bluez/hci0/dev_CA_FE_13_37_00_01/service001a/char0022
	444d0004-4865-6d6c-6f63-6b21f09fa4aa
	Vendor specific
Characteristic (Handle 0x001f)
	/org/bluez/hci0/dev_CA_FE_13_37_00_01/service001a/char001f
	444d0003-4865-6d6c-6f63-6b21f09fa4aa
	Vendor specific
Descriptor (Handle 0xaa9e)
	/org/bluez/hci0/dev_CA_FE_13_37_00_01/service001a/char001f/desc0021
	00002902-0000-1000-8000-00805f9b34fb
	Client Characteristic Configuration
Characteristic (Handle 0x001d)
	/org/bluez/hci0/dev_CA_FE_13_37_00_01/service001a/char001d
	444d0002-4865-6d6c-6f63-6b21f09fa4aa
	Vendor specific
Characteristic (Handle 0x001b)
	/org/bluez/hci0/dev_CA_FE_13_37_00_01/service001a/char001b
	444d0001-4865-6d6c-6f63-6b21f09fa4aa
	Vendor specific
Primary Service (Handle 0x0015)
	/org/bluez/hci0/dev_CA_FE_13_37_00_01/service0015
	0000180a-0000-1000-8000-00805f9b34fb
	Device Information
Characteristic (Handle 0x0018)
	/org/bluez/hci0/dev_CA_FE_13_37_00_01/service0015/char0018
	00002a29-0000-1000-8000-00805f9b34fb
	Manufacturer Name String
Characteristic (Handle 0x0016)
	/org/bluez/hci0/dev_CA_FE_13_37_00_01/service0015/char0016
	00002a24-0000-1000-8000-00805f9b34fb
	Model Number String
Primary Service (Handle 0x0010)
	/org/bluez/hci0/dev_CA_FE_13_37_00_01/service0010
	0000180f-0000-1000-8000-00805f9b34fb
	Battery Service
Characteristic (Handle 0x0011)
	/org/bluez/hci0/dev_CA_FE_13_37_00_01/service0010/char0011
	00002a19-0000-1000-8000-00805f9b34fb
	Battery Level
Descriptor (Handle 0xaa9d)
	/org/bluez/hci0/dev_CA_FE_13_37_00_01/service0010/char0011/desc0014
	00002904-0000-1000-8000-00805f9b34fb
	Characteristic Format
Descriptor (Handle 0xaa9d)
	/org/bluez/hci0/dev_CA_FE_13_37_00_01/service0010/char0011/desc0013
	00002902-0000-1000-8000-00805f9b34fb
	Client Characteristic Configuration
Primary Service (Handle 0x0001)
	/org/bluez/hci0/dev_CA_FE_13_37_00_01/service0001
	00001801-0000-1000-8000-00805f9b34fb
	Generic Attribute Profile
Characteristic (Handle 0x0007)
	/org/bluez/hci0/dev_CA_FE_13_37_00_01/service0001/char0007
	00002b2a-0000-1000-8000-00805f9b34fb
	Database Hash
Characteristic (Handle 0x0005)
	/org/bluez/hci0/dev_CA_FE_13_37_00_01/service0001/char0005
	00002b29-0000-1000-8000-00805f9b34fb
	Client Supported Features
Characteristic (Handle 0x0002)
	/org/bluez/hci0/dev_CA_FE_13_37_00_01/service0001/char0002
	00002a05-0000-1000-8000-00805f9b34fb
	Service Changed
Descriptor (Handle 0xaa9d)
	/org/bluez/hci0/dev_CA_FE_13_37_00_01/service0001/char0002/desc0004
	00002902-0000-1000-8000-00805f9b34fb
	Client Characteristic Configuration
```

Vudentz, 2025-03-13:
I think this was fixed by 720e8ec9760b8d8bfb565e535bd311bbc8273a76 but perhaps something else is at player, or the Handle property really contains invalid values.

XenoKovah, 2025-03-13:
The description and diffs make it seem like that's very likely the issue. I'm going to be OOO for the next week but I'll try to confirm when I get back.

closed 2025-03-14 by BluezTestBot