bluez/bluez #1134: Blocked state not being respected

ArtemSmaznov, 2025-03-15:
I use a script to block/unblock my Bluetooth headset using bluetoothctl to control when it would connect to my PC (audio input/output switches automatically between my speakers and headset when the headset is connected as I keep the headset as default sink). I keep the headset in the blocked state most of the time and only unblock it when I want to switch audio to a headset.

About a month or two ago I did a system update and now the blocked status is not being respected. I turn my PC on, and when I put on the headset it automatically connects and disables my speakers output even though the headset is still blocked.

```
➜ bluetoothctl info 80:C3:BA:49:FA:92
Device 80:C3:BA:49:FA:92 (public)
	Name: MOMENTUM 4
	Alias: momentum-4
	Class: 0x00240404 (2360324)
	Icon: audio-headset
	Paired: yes
	Bonded: yes
	Trusted: yes
	Blocked: yes
	Connected: yes
	LegacyPairing: no
	UUID: Vendor specific           (00000000-deca-fade-deca-deafdecacaff)
	UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: Advanced Audio Distribu.. (0000110d-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
	UUID: Battery Service           (0000180f-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (0000eb10-d102-11e1-9b23-00025b00a5a5)
	UUID: Qualcomm Technologies I.. (0000fd92-0000-1000-8000-00805f9b34fb)
	UUID: SENNHEISER electronic G.. (0000fdce-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (a2129ff3-081b-4c45-8afe-469d9c4842ec)
	Battery Percentage: 0x3c (60)
```

My setup info:
```
blueman: 2.4.4-1
bluez: 5.79-1
OS: Arch Linux x86_64
Kernel: Linux 6.13.6-zen1-1-zen
```

Repro steps:
```
*Given* I have my headset set as the default sink
*And* I have my headset blocked via bluetoothctl or blueman-manager
*And* I power on my PC
*And* I have music playing from my speakers
*When* I turn on my headset
*Then* nothing on my PC should change
```
Actual result
* the headset connects to the PC even though it is blocked - this doesn't always happen immediately and can take a minute or two, also the behavior seems different first time after boot as compared to after I interact with the device via bluetoothctl
* PC stops outputting to the speakers
* Sometimes the audio starts playing via the headset and something there is no audio at all (I am not sure about this behavior and don't particularly care about it as this would be a different issue with the sinks that would be out of scope of this report)

In order to preserve my workflow I now need to disable the whole Bluetooth adapter instead of just 1 device. Not sure if there was a regression in bluez in a recent update or that is something else on my system interfering with how it works. I would appreciate any troubleshooting info.

I am not sure how widespread the issue is as I don't expect many people to utilize the blocking functionality. If this is not easily reproducible please let me know what other info I can provide to help narrow down this issue

ArtemSmaznov, 2025-03-15:
This is what printed in the syslog when I turned the headset on
```
Mar 15 13:05:45 izumi systemd[5016]: Reached target Bluetooth.
Mar 15 13:05:46 izumi blueman-tray[6485]: gtk_widget_get_scale_factor: assertion 'GTK_IS_WIDGET (widget)' failed
Mar 15 13:05:51 izumi bluetoothd[858]: src/profile.c:ext_connect() Hands-Free Voice gateway failed connect to 80:C3:BA:49:FA:92: Connection refused (111)
Mar 15 13:05:51 izumi wireplumber[5032]: spa.bluez5.native: RFCOMM receive command but modem not available: AT+BTRH?
Mar 15 13:05:52 izumi kernel: input: MOMENTUM 4 (AVRCP) as /devices/virtual/input/input25
Mar 15 13:05:52 izumi systemd-logind[865]: Watching system buttons on /dev/input/event15 (MOMENTUM 4 (AVRCP))
Mar 15 13:05:53 izumi bluetoothd[858]: profiles/audio/avdtp.c:avdtp_setconf_cmd() Unable to get a audio sink object
Mar 15 13:05:53 izumi bluetoothd[858]: profiles/audio/avdtp.c:avdtp_setconf_cmd() Unable to get a audio sink object
Mar 15 13:05:53 izumi bluetoothd[858]: profiles/audio/avdtp.c:avdtp_setconf_cmd() Unable to get a audio sink object
Mar 15 13:05:53 izumi bluetoothd[858]: profiles/audio/avdtp.c:avdtp_setconf_cmd() Unable to get a audio sink object
Mar 15 13:05:53 izumi bluetoothd[858]: profiles/audio/avdtp.c:avdtp_setconf_cmd() Unable to get a audio sink object
Mar 15 13:05:53 izumi bluetoothd[858]: profiles/audio/avdtp.c:avdtp_setconf_cmd() Unable to get a audio sink object
Mar 15 13:05:53 izumi bluetoothd[858]: profiles/audio/avdtp.c:avdtp_setconf_cmd() Unable to get a audio sink object
Mar 15 13:05:53 izumi bluetoothd[858]: profiles/audio/avdtp.c:avdtp_setconf_cmd() Unable to get a audio sink object
Mar 15 13:05:53 izumi bluetoothd[858]: profiles/audio/avdtp.c:avdtp_setconf_cmd() Unable to get a audio sink object
Mar 15 13:05:53 izumi bluetoothd[858]: profiles/audio/avdtp.c:avdtp_setconf_cmd() Unable to get a audio sink object
Mar 15 13:05:53 izumi bluetoothd[858]: profiles/audio/avdtp.c:avdtp_setconf_cmd() Unable to get a audio sink object
Mar 15 13:05:54 izumi bluetoothd[858]: profiles/audio/avdtp.c:avdtp_setconf_cmd() Unable to get a audio sink object
Mar 15 13:05:54 izumi bluetoothd[858]: profiles/audio/avdtp.c:avdtp_setconf_cmd() Unable to get a audio sink object
```

KeyofBlueS, 2025-04-26:
I'm experiencing the same issue on Debian Sid, running BlueZ 5.82 and kernel 6.12.25. I've tried downgrading BlueZ to versions 5.79 and 5.66, booted into earlier kernels, 6.12.21, 6.12.20, 6.12.19, and 6.12.17, and even rolled back some firmware packages, all setups where Bluetooth used to work fine, but none of that helped.

I have two Bluetooth controllers, one integrated and one USB, from different vendors, and both are affected by the same issue. That makes me inclined to believe it's not a hardware or device-specific problem.

I'm not sure where the problem lies. The dmesg and journalctl logs don't show anything out of the ordinary, aside from the controllers repeatedly trying to connect to paired audio devices. At this point, I'm not sure where else to look.

That said, I did find a workaround: in Blueman, while a device is connected, right-click it and uncheck:
Auto-connect > Audio and input profiles.

This fixed the issue for me, although it's worth noting that this step wasn't necessary before.