bluez/bluez #1138: A Logitech M240 bluetooth mouse turns unusable with the transition from 5.79 to 5.80

friedrichromstedt, 2025-03-18:
This is on an Arch Linux system which is patched every couple of days. Today, with the upgrade of the Arch Linux packages `bluez`, `bluez-libs` and `bluez-utils` from v5.79 to v5.80, my Logitech M240 mouse is no longer recognised with 5.80. Patching back to 5.79 is a usable work-around.

I am carefully watching my System Journal. With v5.79 I obtain a log like:
```
Mar 18 15:31:54 T470p systemd[748]: Reached target Bluetooth.
Mar 18 15:31:55 T470p kernel: input: LOGI M240 Mouse as /devices/virtual/misc/uhid/0005:046D:B03A.0001/input/input18
Mar 18 15:31:55 T470p kernel: hid-generic 0005:046D:B03A.0001: input,hidraw0: BLUETOOTH HID v0.09 Mouse [LOGI M240] on f8:59:71:8d:e0:81
```
when moving the mouse around a little after startup of X, while with v5.80 there is no such output at all.

Feel free to let me know what kind of logging can be useful to troubleshoot this.

This is not an immediate problem for me, as I can flawlessly revert to v5.79. However, I felt it would be useful to report immediately.

*Patching back* means here to downgrade the abovementioned Arch Linux packages and to subsequently reboot the machine.

If you prefer, I can turn to the Arch guys first, in case this is a configuration problem.

Best,
Friedrich

Heyian, 2025-03-18:
I got the same exact issue with the MX Vertical. Reverting to 5.79 solved the issue.

Vudentz, 2025-03-18:
git diff 5.79..5.80 profiles/input/:
```
diff --git a/profiles/input/device.c b/profiles/input/device.c
index 8cf8e5ea78e1..3627573e79c8 100644
--- a/profiles/input/device.c
+++ b/profiles/input/device.c
@@ -1527,9 +1527,9 @@ int input_device_register(struct btd_service *service)
        if (uhid_state) {
                idev->uhid = bt_uhid_new_default();
                if (!idev->uhid) {
-                       error("bt_uhid_new_default: failed");
-                       input_device_free(idev);
-                       return -EIO;
+                       DBG("bt_uhid_new_default failed, switching to kernel "
+                           "mode");
+                       uhid_state = UHID_DISABLED;
                }
        }
```

Vudentz, 2025-03-18:
> git diff 5.79..5.80 profiles/input/:
>
> ```
> diff --git a/profiles/input/device.c b/profiles/input/device.c
> index 8cf8e5ea78e1..3627573e79c8 100644
> --- a/profiles/input/device.c
> +++ [...]

There not so much changes with respect to input devices from 5.79 to 5.80, or perhaps there are not reconnecting at all?

Vudentz, 2025-03-18:
[342302.589164] input: Logitech MX Anywhere 3 as /devices/virtual/misc/uhid/0005:046D:B025.0010/input/input47
[342302.590230] logitech-hidpp-device 0005:046D:B025.0010: input,hidraw10: BLUETOOTH HID v0.14 Mouse [Logitech MX Anywhere 3] on xx:xx:xx:xx:xx:xx
[342302.671257] logitech-hidpp-device 0005:046D:B025.0010: HID++ 4.5 device connected.

Vudentz, 2025-03-18:
@friedrichromstedt perhaps the problem is that it doesn't even connect?

pashynskykh, 2025-03-19:
I have exactly the same problem on Arch Linux: after upgrading `bluez` and `bluez-libs` from v5.79 to v5.80 - both my keyboard and mouse (MX Keys and MX Master 3) stopped working. 

I tried to repair the mouse, but after successfully pairing it - it refused to connect again.

After downgrading to v5.79 and restarting the Bluetooth service, everything started working again.

ee-boi, 2025-03-19:
Same problem here with a Logitech MX Master 2S. running the following commands reconnects the mouse but does not survive a reboot:

`sudo systemctl stop bluetooth`
`sudo rm -rf /var/lib/bluetooth/*`
`sudo systemctl start bluetooth`

and then using `bluetoothctl` : 
```bluetoothctl
agent on
default-agent
power on
scan on
pair XX:XX:XX:XX:XX:XX
connect XX:XX:XX:XX:XX:XX
trust XX:XX:XX:XX:XX:XX```

ArkadyBuryakov, 2025-03-19:
Same thing with Logitech MX Mechanical Keyboard and MX Master 3 mouse

Vudentz, 2025-03-19:
```
< HCI Command: LE Set Extended Scan Parameters (0x08|0x0041) plen 13                                              #1080 [hci0] 15:32:33.270637
        Own address type: Public (0x00)
        Filter policy: Ignore not in accept list (0x01)
        PHYs: 0x05
        Entry 0: LE 1M
          Type: Passive (0x00)
          Interval: 60.000 msec (0x0060)
          Window: 30.000 msec (0x0030)
        Entry 1: LE Coded
          Type: Passive (0x00)
          Interval: 180.000 msec (0x0120)
          Window: 90.000 msec (0x0090)
> HCI Event: Command Complete (0x0e) plen 4                                                                       #1081 [hci0] 15:32:33.271613
      LE Set Extended Scan Parameters (0x08|0x0041) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Extended Scan Enable (0x08|0x0042) plen 6                                                   #1082 [hci0] 15:32:33.271637
        Extended scan: Enabled (0x01)
        Filter duplicates: Enabled (0x01)
        Duration: 0 msec (0x0000)
        Period: 0.00 sec (0x0000)
```

The it reconnects via Directed Advertisement:

```
> HCI Event: LE Meta Event (0x3e) plen 26                                                                         #1084 [hci0] 15:33:08.756291
      LE Extended Advertising Report (0x0d)
        Num reports: 1
        Entry 0
          Event type: 0x0015
            Props: 0x0015
              Connectable
              Directed
              Use legacy advertising PDUs
            Data status: Complete
          Legacy PDU Type: ADV_DIRECT_IND (0x0015)
          Address type: Random (0x01)
          Address: XX:XX:XX:XX:XX:XX (Static)
          Primary PHY: LE 1M
          Secondary PHY: No packets
          SID: no ADI field (0xff)
          TX power: 127 dBm
          RSSI: -74 dBm (0xb6)
          Periodic advertising interval: 0.00 msec (0x0000)
          Direct address type: Public (0x00)
          Direct address: XX:XX:XX:XX:XX:XX (Intel Corporate)
          Data length: 0x00
```

SamuelGilbert42, 2025-03-19:
```
Bluetooth monitor ver 5.80
btmon[1542]: = Note: Linux version 6.13.7-arch1-1 (x86_64)                                                                     0.462853
btmon[1542]: = Note: Bluetooth subsystem version 2.22                                                                          0.462854
= New Index: 00:28:F8:19:8C:68 (Primary,USB,hci0)                                                                       [hci0] 0.462855
= Open Index: 00:28:F8:19:8C:68                                                                                         [hci0] 0.462856
= Index Info: 00:28:F8:19:8C:68 (Intel Corp.)                                                                           [hci0] 0.462856
bluetoothd[496]: @ MGMT Open: bluetoothd (privileged) version 1.23                                                    {0x0001} 0.462857
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2                                                              #1 [hci0] 23.235327
        Scanning: Disabled (0x00)
        Filter duplicates: Disabled (0x00)
> HCI Event: Command Complete (0x0e) plen 4                                                                         #2 [hci0] 23.345269
      LE Set Scan Enable (0x08|0x000c) ncmd 2
        Status: Success (0x00)
< HCI Command: LE Set Address Resolution Enable (0x08|0x002d) plen 1                                                #3 [hci0] 23.345513
        Address resolution: Disabled (0x00)
> HCI Event: Command Complete (0x0e) plen 4                                                                         #4 [hci0] 23.346118
      LE Set Address Resolution Enable (0x08|0x002d) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Address Resolution Enable (0x08|0x002d) plen 1                                                #5 [hci0] 23.346266
        Address resolution: Enabled (0x01)
> HCI Event: Command Complete (0x0e) plen 4                                                                         #6 [hci0] 23.347109
      LE Set Address Resolution Enable (0x08|0x002d) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7                                                          #7 [hci0] 23.347229
        Type: Passive (0x00)
        Interval: 60.000 msec (0x0060)
        Window: 60.000 msec (0x0060)
        Own address type: Public (0x00)
        Filter policy: Ignore not in accept list (0x01)
> HCI Event: Command Complete (0x0e) plen 4                                                                         #8 [hci0] 23.348103
      LE Set Scan Parameters (0x08|0x000b) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2                                                              #9 [hci0] 23.348225
        Scanning: Enabled (0x01)
        Filter duplicates: Enabled (0x01)
> HCI Event: Command Complete (0x0e) plen 4                                                                        #10 [hci0] 23.349102
      LE Set Scan Enable (0x08|0x000c) ncmd 2
        Status: Success (0x00)
> HCI Event: LE Meta Event (0x3e) plen 18                                                                          #11 [hci0] 41.863680
      LE Direct Advertising Report (0x0b)
        Num reports: 1
        Event type: Connectable directed - ADV_DIRECT_IND (0x01)
        Address type: Random (0x01)
        Address: CC:6D:D5:87:86:2D (Static)
        Direct address type: Random (0x01)
        Direct address: 00:28:F8:19:8C:68 (Non-Resolvable)
        RSSI: -48 dBm (0xd0)
@ MGMT Event: Connect Failed (0x000d) plen 8                                                                  {0x0001} [hci0] 65.480460
        LE Address: CC:6D:D5:87:86:2D (Static)
        Status: Disconnected (0x0e)
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2                                                             #12 [hci0] 65.503239
        Scanning: Disabled (0x00)
        Filter duplicates: Disabled (0x00)
> HCI Event: Command Complete (0x0e) plen 4                                                                        #13 [hci0] 65.513076
      LE Set Scan Enable (0x08|0x000c) ncmd 2
        Status: Success (0x00)
< HCI Command: LE Set Address Resolution Enable (0x08|0x002d) plen 1                                               #14 [hci0] 65.513166
        Address resolution: Disabled (0x00)
> HCI Event: Command Complete (0x0e) plen 4                                                                        #15 [hci0] 65.514038
      LE Set Address Resolution Enable (0x08|0x002d) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Address Resolution Enable (0x08|0x002d) plen 1                                               #16 [hci0] 65.514142
        Address resolution: Enabled (0x01)
> HCI Event: Command Complete (0x0e) plen 4                                                                        #17 [hci0] 65.514871
      LE Set Address Resolution Enable (0x08|0x002d) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7                                                         #18 [hci0] 65.514983
        Type: Passive (0x00)
        Interval: 60.000 msec (0x0060)
        Window: 30.000 msec (0x0030)
        Own address type: Public (0x00)
        Filter policy: Ignore not in accept list (0x01)
> HCI Event: Command Complete (0x0e) plen 4                                                                        #19 [hci0] 65.515913
[... cut ...]
> HCI Event: LE Meta Event (0x3e) plen 18                                                                          #22 [hci0] 83.984214
      LE Direct Advertising Report (0x0b)
        Num reports: 1
        Event type: Connectable directed - ADV_DIRECT_IND (0x01)
        Address type: Random (0x01)
        Address: CC:6D:D5:87:86:2D (Static)
        Direct address type: Random (0x01)
        Direct address: 00:28:F8:19:8C:68 (Non-Resolvable)
        RSSI: -54 dBm (0xca)
```
Nothing much happened when I launched `btmon` so I used the KDE widget and clicked on connect for my keyboard. I hope this is useful!

Vudentz, 2025-03-19:
> ```
>       LE Direct Advertising Report (0x0b)
>         Num reports: 1
>         Event type: Connectable directed - ADV_DIRECT_IND (0x01)
>         Address type: Random (0x01)
>         Address: [...]

So it doesn't connect even after getting the above event? Can you try reloading bluetoothd to see if the address above is programmed in the controller auto-connect list?

SamuelGilbert42, 2025-03-19:
Kernel version: 6.13.7-arch1-1
Gimme a sec for the reload test, I have reverted to 5.79 to be able to use my system...

SamuelGilbert42, 2025-03-19:
The btmon log while executing `sudo systemctl restart bluetooth` is pretty long (>1000 lines).
Should I paste it here nevertheless?

Vudentz, 2025-03-19:
> The btmon log while executing `sudo systemctl restart bluetooth` is pretty long (>1000 lines). Should I paste it here nevertheless?

You can save it in a binary format with -w <filename>, then just attach it in a new comment.

Vudentz, 2025-03-19:
```
> HCI Event: LE Meta Event (0x3e) plen 18                                                                          #22 [hci0] 83.984214
      LE Direct Advertising Report (0x0b)
        Num reports: 1
        Event type: Connectable directed - ADV_DIRECT_IND (0x01)
        Address type: Random (0x01)
        Address: CC:6D:D5:87:86:2D (Static)
        Direct address type: Random (0x01)
        Direct address: 00:28:F8:19:8C:68 (Non-Resolvable)
        RSSI: -54 dBm (0xca)
```

Not sure how I failed to realize this but the Direct address type is incorrect, your address type is public, I actually have no idea how this could work previously but perhaps the remote side don't use Directed Advertisements with 5.79 for some reason, that said the kernel seems to require privacy to handle such events:

```
	/* If the direct address is present, then this report is from
	 * a LE Direct Advertising Report event. In that case it is
	 * important to see if the address is matching the local
	 * controller address.
	 */
	if (!hci_dev_test_flag(hdev, HCI_MESH) && direct_addr) {
		direct_addr_type = ev_bdaddr_type(hdev, direct_addr_type,
						  &bdaddr_resolved);

		/* Only resolvable random addresses are valid for these
		 * kind of reports and others can be ignored.
		 */
		if (!hci_bdaddr_is_rpa(direct_addr, direct_addr_type))
			return;

		/* If the controller is not using resolvable random
		 * addresses, then this report can be ignored.
		 */
		if (!hci_dev_test_flag(hdev, HCI_PRIVACY))
			return;

		/* If the local IRK of the controller does not match
		 * with the resolvable random address provided, then
		 * this report can be ignored.
		 */
		if (!smp_irk_matches(hdev, hdev->irk, direct_addr))
			return;
	}
```

The extended version completely bypass these checks so Im pretty sure this is wrong.

Vudentz, 2025-03-20:
> ```
>       LE Direct Advertising Report (0x0b)
>         Num reports: 1
>         Event type: Connectable directed - ADV_DIRECT_IND (0x01)
>         Address type: Random (0x01)
>         Address: [...]

Actually this is correct, the Direct address type can only be 0x01:

[image]

It is the very event that is wrong since it shall only be generated if:

[image]

But the filter_type is 0x01 so the code bellow is actually doing the correct thing which is to assume local privacy shall be enabled to result in filter_type 0x02 or 0x03.

> ```
> 	/* If the direct address is present, then this report is from
> 	 * a LE Direct Advertising Report event. In that case it is
> 	 * important to see if the address is matching the local
> 	 * [...]

So this is correct but perhaps the controller is confused by the filter_type and generates the Directed Report rather than the regular Advertising Report which prevents the code to perform the regular auto-connect logic.

cwrau, 2025-03-20:
I'm running into a similar issue using bluez 5.80 compared to 5.79; my (stadia) controller aren't connecting. Downgrading fixes it.

What kind of logs can I send to help?

SamuelGilbert42, 2025-03-20:
@Vudentz Yes, I'll patch my kernel.
It's been a while since I built a kernel; it brings me back to my Linux From Scratch days ;-)

Vudentz, 2025-03-20:
@cwrau @SamuelGilbert42 another option is perhaps disable CentralAddressResolution in main.conf and see if that helps:

https://github.com/bluez/bluez/blob/master/src/main.conf#L185

It is just a work around though since we might have to fix this in the kernel anyway.

SamuelGilbert42, 2025-03-20:
I'm currently compiling the kernel, so I'll try that first.
It will probably take a while on my old Core i5-7300U laptop.

SamuelGilbert42, 2025-03-21:
Everything works as expected with the custom kernel. +1
Tell me if you need me to run other tests, and I will gladly do so.

cwrau, 2025-03-21:
> [@cwrau](https://github.com/cwrau?rgh-link-date=2025-03-20T19%3A42%3A02.000Z) [@SamuelGilbert42](https://github.com/SamuelGilbert42?rgh-link-date=2025-03-20T19%3A42%3A02.000Z) another option is perhaps disable CentralAddressResolution in main.conf and see if that helps:
> 
> [`master`/src/main.conf#L185](https://github.com/bluez/bluez/blob/master/src/main.conf?rgh-link-date=2025-03-20T19%3A42%3A02.000Z#L185)
> 
> It is just a work around though since we might have to fix this in the kernel anyway.

That does indeed work! I will stick to the older version though as I don't really like editing files that are [provided by a package](https://gitlab.archlinux.org/archlinux/packaging/packages/bluez/-/blob/main/PKGBUILD?ref_type=heads#L76) 😅

Off-topic question, is there a way to set a configuration without touching the `main.conf` maybe something like `main.conf.d/*.conf`? I couldn't find anything about that.

Vudentz, 2025-03-21:
@cwrau can you try with the changes above, this should be similar to disabling in main.conf but it uses a new flag that is supposed to convey the controller support for the feature (LL Privacy), if that works we might release 5.81 to get this fixed faster than waiting for the kernel fix to land on the distros.

ArchangeGabriel, 2025-03-22:
> That does indeed work! I will stick to the older version though as I don't really like editing files that are [provided by a [...]

It is in the backup array (https://gitlab.archlinux.org/archlinux/packaging/packages/bluez/-/blob/main/PKGBUILD?ref_type=heads#L74), which means it is expected that you may modify it.

ArchangeGabriel, 2025-03-22:
> [@cwrau](https://github.com/cwrau) can you try with the changes above, this should be similar to disabling in main.conf but it uses a new flag that is supposed to convey the controller support for [...]

You mean this commit: https://github.com/tedd-an/bluez-upstream-test/commit/978851b729c81e6f092927a6682dee751821b039?

I tried that, it did not work.

Vudentz, 2025-03-24:
> > [@cwrau](https://github.com/cwrau) can you try with the changes above, this should be similar to disabling in main.conf but it uses a new flag that is supposed to convey the controller support for the feature (LL Privacy), if that works we might release 5.81 to get this fixed faster than waiting for the kernel fix to land on the distros.
> 
> You mean this commit: [tedd-an/bluez-upstream-test@978851b](https://github.com/tedd-an/bluez-upstream-test/commit/978851b729c81e6f092927a6682dee751821b039)?
> 
> I tried that, it did not work.

You might have a different problem then, please attach the btmon traces.

Vudentz, 2025-03-24:
> [@cwrau](https://github.com/cwrau) [@SamuelGilbert42](https://github.com/SamuelGilbert42) another option is perhaps disable CentralAddressResolution in main.conf and see if that helps: [...]

https://github.com/bluez/bluez/commit/a0d6d3c78b28e667e53524b5e5961711e60ff13e should have made CentralAddressResolution conditional to LL Privacy so if you guys could please check with current git master, if that works we can probably go ahead and release 5.81.

Vudentz, 2025-03-24:
> Here is my btmon trace with the new version of this patch (which still does not work):
> 
> ```
> < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2                                                                                                                                                                                                      #1 [hci0] 24.021233
>         Scanning: Disabled (0x00)
>         Filter duplicates: Disabled (0x00)
> > HCI Event: Command Complete (0x0e) plen 4                                                                                                                                                                                                                 #2 [hci0] 24.118556
>       LE Set Scan Enable (0x08|0x000c) ncmd 2
>         Status: Success (0x00)
> < HCI Command: LE Set Address Resolution Enable (0x08|0x002d) plen 1                                                                                                                                                                                        #3 [hci0] 24.118718
>         Address resolution: Disabled (0x00)
> > HCI Event: Command Complete (0x0e) plen 4                                                                                                                                                                                                                 #4 [hci0] 24.119532
>       LE Set Address Resolution Enable (0x08|0x002d) ncmd 1
>         Status: Success (0x00)
> < HCI Command: LE Set Address Resolution Enable (0x08|0x002d) plen 1                                                                                                                                                                                        #5 [hci0] 24.119695
>         Address resolution: Enabled (0x01)
> > HCI Event: Command Complete (0x0e) plen 4                                                                                                                                                                                                                 #6 [hci0] 24.120568
>       LE Set Address Resolution Enable (0x08|0x002d) ncmd 1
>         Status: Success (0x00)
> < HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7                                                                                                                                                                                                  #7 [hci0] 24.120601
>         Type: Passive (0x00)
>         Interval: 60.000 msec (0x0060)
>         Window: 60.000 msec (0x0060)
>         Own address type: Public (0x00)
>         Filter policy: Ignore not in accept list (0x01)
> > HCI Event: Command Complete (0x0e) plen 4                                                                                                                                                                                                                 #8 [hci0] 24.121502
>       LE Set Scan Parameters (0x08|0x000b) ncmd 1
>         Status: Success (0x00)
> < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2                                                                                                                                                                                                      #9 [hci0] 24.121526
>         Scanning: Enabled (0x01)
>         Filter duplicates: Enabled (0x01)
> > HCI Event: Command Complete (0x0e) plen 4                                                                                                                                                                                                                #10 [hci0] 24.122500
>       LE Set Scan Enable (0x08|0x000c) ncmd 2
>         Status: Success (0x00)
> > HCI Event: LE Meta Event (0x3e) plen 18                                                                                                                                                                                                                  #11 [hci0] 24.123510
>       LE Direct Advertising Report (0x0b)
>         Num reports: 1
>         Event type: Connectable directed - ADV_DIRECT_IND (0x01)
>         Address type: Random (0x01)
>         Address: D6:71:4F:D7:17:C6 (Static)
>         Direct address type: Random (0x01)
>         Direct address: D0:AB:D5:00:08:BF (Static)
>         RSSI: -47 dBm (0xd1)
> ```
> 
> Disabling `CentralAddressResolution` in `main.conf` still works.

Weird, if you haven't patched your kernel then it shall not have the new LL Privacy flag so CentralAddressResolution shall never be enabled, so either you are not running with the git master or something else is going on.

ArchangeGabriel, 2025-03-24:
I am not running with git master (that is not trivial to do from the current PKGBUILD using our ArchLinux build system, though if that is really needed I can do it, that is just a bit more work vs adding a patch), but with https://github.com/bluez/bluez/commit/a0d6d3c78b28e667e53524b5e5961711e60ff13e applied on top of 5.80. And I made sure I’m compiling with this patch added, I checked the source code after patching. If there is another commit I should cherry-pick, please let me know. So yes, I am replacing the system bluetoothd, and I made sure to restart the bluetooth.service between each test, because indeed if I did not it would keep the old version running (e.g. I could have 5.79 running, install 5.80 and everything would be fine until I restarted the service).

I have not patched my kernel, the source of the one I’m using can be found here: https://github.com/anthraxx/linux-hardened/blob/6.13/net/bluetooth/hci_event.c#L6047

ArchangeGabriel, 2025-03-25:
Here are the logs from the CLI:
[5.79.log](https://github.com/user-attachments/files/19442788/5.79.log)
[5.80.log](https://github.com/user-attachments/files/19442789/5.80.log)
Here are the logs from `-w`:
[5.79.btmon.log](https://github.com/user-attachments/files/19442807/5.79.btmon.log)
[5.80.btmon.log](https://github.com/user-attachments/files/19442808/5.80.btmon.log)

cwrau, 2025-03-25:
The git version (installed via https://aur.archlinux.org/packages/bluez-git, resulting in `912e2f54f`) also doesn't work for me;

[btmon-working.log](https://github.com/user-attachments/files/19448360/btmon-working.log)
[btmon-not-working.log](https://github.com/user-attachments/files/19448362/btmon-not-working.log)

I did no changes in the config. Using kernel 6.12.19-1-lts

Vudentz, 2025-03-25:
Ok so this is due to the following code it seems:
```

	/* For backward compatibility set LL Privacy as enabled if
	 * CentralAddressResolution has been set so old kernel LL Privacy is
	 * enabled.
	 */
	if (!strcmp(uuid, "15c0a148-c273-11ea-b3de-0242ac130004") &&
			btd_opts.defaults.le.addr_resolution)
		return true
```

So this will enable the old experimental flag for LL Privacy:

```
@ MGMT Command: Read Experimental Features Information (0x0049) plen 0                                                {0x0001} [hci0] 6.394912
@ MGMT Event: Command Complete (0x0001) plen 105                                                                      {0x0001} [hci0] 6.394931
      Read Experimental Features Information (0x0049) plen 102
        Status: Success (0x00)
        Features: 5
        UUID: BlueZ Experimental Simultaneous Central and Peripheral
        Flags: 0x00000000
        UUID: BlueZ Experimental LL privacy
        Flags: 0x00000002
          Settings change
        UUID: BlueZ Experimental Bluetooth Quality Report
        Flags: 0x00000000
        UUID: BlueZ Experimental ISO Socket
        Flags: 0x00000000
        UUID: Vendor specific
        Flags: 0x00000000
@ MGMT Command: Set Experimental Feature (0x004a) plen 17                                                             {0x0001} [hci0] 6.394979
        UUID: BlueZ Experimental LL privacy
        Action: Enabled (0x01)
```

Now I can remove that code so 5.80 would behave exactly as 5.79 for older kernel, but as soon as distros move to newer kernels it would just resurface causing:

```
> HCI Event: LE Meta Event (0x3e) plen 18                                                                                #151 [hci0] 10.388155
      LE Direct Advertising Report (0x0b)
        Num reports: 1
        Event type: Connectable directed - ADV_DIRECT_IND (0x01)
        Address type: Random (0x01)
        Address: D6:71:4F:D7:17:C6 (Static)
        Direct address type: Random (0x01)
        Direct address: D0:AB:D5:00:08:BF (Static)
        RSSI: -51 dBm (0xcd)
```

So setting CentralAddressResolution=0 is the only possible the best solution until https://github.com/bluez/bluetooth-next/commit/a37df2815c1b51468a2fe71758b81cb2084433eb lands.

Vudentz, 2025-03-25:
@cwrau @ArchangeGabriel and other with still see the problem even with the fixes in master, please check the following changes:

https://github.com/tedd-an/bluez-upstream-test/commit/0084331bdc7843e1c85965d519324fcb7849ec23 

Ive already sent the pull request that include the fixes for the kernel so hopefully when it land in your distros it shall be alright.

AlphaElwedritsch, 2025-03-26:
Does this happen only for mouses or any other Bluetooth devices?

cwrau, 2025-03-26:
> Does this happen only for mouses or any other Bluetooth devices?

For me it just happens on my (stadia) controllers, my (a couple of years old) MX Master works without any problem.

Vudentz, 2025-03-26:
> Does this happen only for mouses or any other Bluetooth devices?

Any device using directed advertising can probably trigger this problem, not though that it requires a certain combination of controllers not supporting extended advertising + LL Privacy.

As explained in https://github.com/bluez/bluez/issues/1138#issuecomment-2752630320 there is a kernel fix/workaround for these controllers, but we will likely want to release a 5.81 for older systems where these fixes will probably not be deployed so userspace will need to keep working without them, so hopefully the following fixes it for good:

https://patchwork.kernel.org/project/bluetooth/patch/20250325213230.2513331-1-luiz.dentz@gmail.com/

closed 2025-03-27 by BluezTestBot

cwrau, 2025-03-28:
Nice, bluez-git now works for me! Currently running commit 5df9521ce and my controller successfully connects!

pashynskykh, 2025-04-09:
Today on Arch Linux we got Linux kernel v6.14.1.arch1-1, on which both versions of Bluez - 5.82 and 5.79 - do not work. I'm guessing this is related to this bug.

pashynskykh, 2025-04-09:
@Vudentz it must be, unfortunately I can't check it right now. But bluez `5.82` fixed the problem for me and worked fine until today, when we got `linux-6.14.1.arch1-1`. 

Reverting the bluez back to `5.79`, which worked fine before, didn't solve the problem. 

I had to revert the kernel back to `linux-6.13.8.arch1-1`, which works fine with both versions of bluez - `5.79` and `5.82`.

pashynskykh, 2025-04-09:
@theclash8 as a temporary workaround, you can revert to the previous kernel version:
```sh
sudo pacman -U https://archive.archlinux.org/packages/l/linux/linux-6.13.8.arch1-1-x86_64.pkg.tar.zst
```

To skip the kernel from being upgraded, add it to the `/etc/pacman.conf`:
```
IgnorePkg=linux
```
But don't forget about this. Avoiding kernel updates for a long time can leave your system vulnerable or just break at some point.

theclash8, 2025-04-10:
> [@theclash8](https://github.com/theclash8) as a temporary workaround, you can revert to the previous kernel version:
> 

I confirm it's working 

linux-6.13.8.arch1-1 with bluez-git.

Thanks

friedrichromstedt, 2025-05-24:
This is just to confirm that `bluez-5.82`, in its Arch Linux revision `5.82-1` (built upstream 2 April 2025), **works well** for me with `linux-6.14.6` (Arch revision `6.14.6-arch1-1`, built upstream 9 May 2025).