bluez/bluez #1139: Using BIS Assistant to set the Source of two BISes fails on BlueZ 5.80

chinli, 2025-03-21:
DUT environment：BlueZ 5.80+pipewire 1.3.83+kernel 6.12
Assistant：PTS dongle
BIS Source：1BIG 2BISes

When transport inquiry bis2, it will prompt **_bt_io_set: setsockopt(BT_ISO_QOS): Invalid argument (22)_**, resulting in only one channel having data, and the kernel will prompt **_hci0: ISO packet for unknown connection handle 257_**
In addition, there is another message, using **_bluetoothctl->scan on_** and then manually setting **_transport.select xxx_** can work properly, that is, both channels have sound.

[btmon.log](https://github.com/user-attachments/files/19387625/btmon.log)
[bluetoothd.log](https://github.com/user-attachments/files/19387626/bluetoothd.log)

Vudentz, 2025-03-21:
@iulia-tanasescu can you take a look?

iulia-tanasescu, 2025-03-25:
@chinli @Vudentz I submitted a patch with the fix: https://patchwork.kernel.org/project/bluetooth/cover/20250325155211.18111-1-iulia.tanasescu@nxp.com/

closed 2025-03-26 by BluezTestBot