bluez/bluez #1140: Querying LE OOB local data in RPA mode

SFrohn, 2025-03-22:
We are trying to query the LE OOB local data in RPA (privacy) mode.

This works as expected:

btmgmt --index hci0 power off
btmgmt --index hci0 privacy off
btmgmt --index hci0 power on
btmgmt --index hci0 le-oob

This return random and confirm value.
However:

btmgmt --index hci0 power off
btmgmt --index hci0 privacy **on**
btmgmt --index hci0 power on
btmgmt --index hci0 le-oob

Returns "Rejected" and no random and confirm value.

Digging around in the sources, we found this code / comment: https://github.com/bluez/bluetooth-next/blob/a37df2815c1b51468a2fe71758b81cb2084433eb/net/bluetooth/mgmt.c#L8241

This indicates that querying the OOB local is intentionally turned off, when in RPA mode.

Is there any _other_ way to query the OOB local data when in RPA mode? We also tried to turn on RPA mode after querying the OOB data. However, this requires powering the adapter off and on again, invalidating the OOB data.

Vudentz, 2025-03-24:
@SFrohn we will need to actually return the RPA address (hdev->rpa), that said the RPA address is supposed to change every so often so you will need to know when it changes which is the tricky part that the comment is talking about as the kernel will update the hdev->rpa on demand (when needed), so we might need an event telling when that happens, that said I don't think btmgmt will be the way go anyway since you can't really monitor events with that, we could perhaps monitor it in bluetoothd and then signal when the RPA changes.

SFrohn, 2025-03-30:
@Vudentz Thank you very much for the explanation. I understand that retrieving the RPA is tricky without having some kind of notification. In the description of the Mgmt API I found the "Local Out Of Band Extended Data Updated Event" (https://github.com/bluez/bluez/blob/d9430c0635bc36177d521d62a51f0d2a4e8c0fbb/doc/mgmt-api.txt#L4817) which seems to address this problem. In my tests, I was not able to observe this event using btmon, but according to the documentation it is only sent to selected sockets.

My main interest is not the actual RPA but the current LE OOB local data (random and confirm value). Maybe there is any other way to retrieve these values when running in RPA mode?

closed 2025-09-07 by github-actions[bot]