bluez/bluez #1142: No audio after Bluetooth profile switch from A2DP to HSF/HFP and back. Only reconnect fixes it.

rsniezek, 2025-03-25:
Hi all,

First of all, I don't know if this is the right place to ask but I think it's the most reasonable one. Please forgive me if not.

I have a Bluetooth headset with built in microphone, Audiotechnica ATX20xBT. Most of the time I use A2DP/AAC profile but I sometimes need to attend some MS Teams meetings in which case I (auto)switch the profile to HSP/HFP(mSBC) so that the built in microphone is usable.

From few system updates, I cannot get back to the A2DP and have working audio. When the meeting ends, my system automatically switches back but the audio is not playing on the headphones. If I manually switch to HSP/HFP, I do get audio (with low quality as expected) but I cannot get A2DP to work until a reconnect. 

System indicates that there is a playback happening, the device is not muted, muting/unmuting doesn't change anything, volume change as well. switching the application to different output and back to headphones also has no impact.  Tested on two different devices, my main Audiotechnica and some fnr earbuds with the same profiles, both devices act identically.

What can I do to debug this further? Is bluez the right place to look?
System info:
 bluetoothd -v
`5.79`
uname -a
`Linux nixos 6.13.4 #1-NixOS SMP PREEMPT_DYNAMIC Fri Feb 21 13:11:21 UTC 2025 x86_64 GNU/Linux`

Thanks in advance for any suggestions.

ValeZAA, 2025-03-28:
Yep, just got that issue as well. My earbuds are LE audio, maybe that is an issue. I was testing classic A2DP though.

msizanoen1, 2025-03-28:
You likely got hit by a kernel bug (patch: https://lore.kernel.org/all/24279b4ca58e92b96fbe8f8214bf95d485ab73f5.1740691671.git.pav@iki.fi/)

rsniezek, 2025-03-28:
Yup, seems to be the case. update from 6.13.4 to 6.14.1 fixes this for me.

ValeZAA, 2025-03-28:
Yes, tried 2 headphones!!

Upgrading! Interesting that on some headphones it switches to HFP (my Sony XM5), on others it does not at all, Buds3 pro. So even this kernel bug has different bugs. Upgrading!!

root@kali:~# uname -a
Linux kali 6.13-rt-amd64 #1 SMP PREEMPT_RT Debian 6.13.7-1~exp1 (2025-03-14) x86_64 GNU/Linux
root@kali:~# apt-get install -t experimental linux-image-amd64

ValeZAA, 2025-03-28:
There is still a bug that with my headphones when I start recording in recorder of Gnome there is latency and first seconds are not recorded. LMAO, same bug as on Windows 11.

Crazy.

ValeZAA, 2025-04-28:
Pleade close this, it was a kernel issue.

rsniezek, 2025-04-28:
Kernel issue