bluez/bluez #1143: Regression on 5.80: Bluetooth Mouse doesn't reconnect after reboot or mouse shutdown

plrigaux, 2025-03-25:
Hello,

The issue happen after my laptop hibernate , reboot or I shutdown (and open) my mouse. My device (Logitech m720) does't reconnect automatically.

### Workaround 1
My workaround was to pair it again through **Gnome Bluetooth settings** or `bluetoothctl`.  Basically, I have to reconnect manually using a different MAC address. 

### Workaround 2
I downgrade to bluez 5.79 and the issue disappeared. (I'm on Arch)

Thank you 

PS
No issue with my Bluetooth keyboard

Vudentz, 2025-03-25:
Sound like a duplicate of https://github.com/bluez/bluez/issues/1138

plrigaux, 2025-03-25:
I guess it is, so I close

closed 2025-03-25 by plrigaux

bravhek, 2026-09-07:
I am seeing the same general M720 automatic-reconnect symptom again on a much newer BlueZ release, so I wanted to add current diagnostic information to this historical report.

Current environment:

    Fedora 44 x86_64
    Kernel: 7.1.13-200.fc44.x86_64
    BlueZ: 5.87-6.fc44
    Logitech M720 Triathlon BLE
    Device address: E6:E7:A5:70:C4:51

Like the original report, the M720 remains paired/bonded and can be used after reconnecting it manually.

However, after reboot the automatic reconnect is unreliable/delayed.

There is an important difference from simply saying that the mouse is "slow":

    time bluetoothctl connect E6:E7:A5:70:C4:51

completes in approximately 0.034 seconds.

I also captured the connection with btmon. The BLE transport itself appears healthy:

    LE Create Connection
    -> LE Connection Complete: Success
    -> encryption succeeds
    -> GATT/HID setup proceeds

So the M720 can establish the BLE connection immediately when BlueZ explicitly initiates it.

The abnormal behavior occurs later in the HOG/UHID path. During the reconnect problem I repeatedly see:

    bt_uhid_input: Invalid argument (22)

followed by:

    No matching connection for device

from profiles/input/hog-lib.c.

The kernel also repeatedly creates new M720 HID/UHID instances over time.

The M720 is using a static random address in the trace, so I do not believe the current problem is an RPA/address-resolution issue.

This suggests that the recurring M720 problem may involve BlueZ connection/profile state management rather than the radio link itself.

The fact that there is an old M720-specific reconnect report on BlueZ 5.80 and the problem is still reproducible on 5.87 makes me think this device is a useful regression test case for the current reconnect/HOG code.

I am not claiming that the implementation failure is identical to the 5.80 regression, only that the user-visible behavior is similar and the current trace gives some additional information about where the failure occurs.