bluez/bluez #1145: After BIG sync is successful, BIS Assistant cannot remove BIS source on BlueZ 5.80

chinli, 2025-03-27:
DUT environment：BlueZ 5.80+pipewire 1.3.83+kernel 6.12
Assistant：Redmi K70 phone
BIS Source：1BIG 2BISes

When the assistant sends a Modify operation, no command to stop the PA is sent.
[image]

[btmon_5.80.log](https://github.com/user-attachments/files/19482874/btmon_5.80.log)
[bluetoothd_5.80.log](https://github.com/user-attachments/files/19482876/bluetoothd_5.80.log)

Pls take a look~ Thanks!

Vudentz, 2025-03-27:
> DUT environment：BlueZ 5.80+pipewire 1.3.83+kernel 6.12 Assistant：Redmi K70 phone BIS Source：1BIG 2BISes
>
> When the assistant sends a Modify operation, no command to stop the PA is sent. [image] [...]

Afaik assistant roles doesn't establish a BIS/BIG, it only scan and does a sort lived PA sync to enumerate the available broadcasters, or the phone is actually the broadcast source and you want to synchronize to it? Anyway, you may need to update your kernel since 6.12 is quite old.

@iulia-tanasescu do you have any specific feedback for this sort of setup?

chinli, 2025-03-28:
Hi @Vudentz 
The BIS Source device is not a K70 phone, but another Android device. I compared the latest version of the kernel and found no necessary changes in the BIS.

closed 2025-04-24 by BluezTestBot