bluez/bluez #1146: BL earbuds AAC codec recognition issue after the 5.80 update (regression)

glanduin, 2025-03-27:
I have multiple pairs of earbuds which support both AAC and SBC codecs. Before the 5.80 update, i was able to switch between these profiles easily. After the update, with the initial connection after a fresh reboot, only one codec gets recognised (SBC). I can see that when i check the endpoint info using bluetoothctl (device seems like it only supports SBC). If i disconnect and reconnect the device, both codecs get recognised and i can switch in between. However, the initial connection only recognises one codec (SBC). It doesn't matter if i clear the bluetooth cache and re-pair the devices. It happens again after a reboot. Only a disconnect-reconnect fixes it.

In addition, i shall mention that, when switching is possible, sometimes, sound comes from only one earbud after a codec switch. Again, switching back and forth fixes it.

In my opinion, maybe, some waiting times were truncated in the code for the handshake process and maybe devices which aren't fast to respond to the bluetooth stack are having such issues now. This is how i surmise it.

Vudentz, 2025-03-27:
@glanduin we should probably start by collecting HCI trace (using btmon).

@fdanis-oss perhaps this has something to do with the qualification fixes for AVDTP?

glanduin, 2025-03-27:
> [@glanduin](https://github.com/glanduin) we should probably start by collecting HCI trace (using btmon).
> 
> [@fdanis-oss](https://github.com/fdanis-oss) perhaps this has something to do with the qualification fixes for AVDTP?

Btmon output is extremely large even for a few seconds. Can you give me the specific items necessary, can i narrow it down for a more compact output?

Vudentz, 2025-03-27:
> Btmon output is extremely large even for a few seconds. Can you give me the specific items necessary, can i narrow it down for a more compact output?

We just need the AVDTP messages though, the bluetoothd traces should be logged alongside so please past them as well.

glanduin, 2025-03-27:
Here is an output. I connected and then disconnected a pair (i also switched in between profiles). In this occasion, both profiles (aac and sbc) were recognised. So if you can't see the problem here, then i should try to get one with a single profile connection.

[btmon.txt](https://github.com/user-attachments/files/19491066/btmon.txt)

Vudentz, 2025-03-27:
```
> ACL Data RX: Handle 11 flags 0x02 dlen 22                                                                                                           #83 [hci0] 26.783220
      Channel: 65 len 18 [PSM 25 mode Basic (0x00)] {chan 1}
      AVDTP: Set Configuration (0x03) Command (0x00) type 0x00 label 15 nosp 0
        ACP SEID: 7
        INT SEID: 7
        Service Category: Media Transport (0x01)
        Service Category: Media Codec (0x07)
          Media Type: Audio (0x00)
          Media Codec: MPEG-2,4 AAC (0x02)
            Object Type: MPEG-2 AAC LC (0x80)
            Frequency: 48000 (0x80)
            Channels: 2 (0x04)
            Bitrate: 264630bps
            VBR: Yes
        Service Category: Delay Reporting (0x08)
bluetoothd[507]: < ACL Data TX: Handle 11 flags 0x00 dlen 6                                                                                           #84 [hci0] 26.787032
      Channel: 66 len 2 [PSM 25 mode Basic (0x00)] {chan 1}
      AVDTP: Set Configuration (0x03) Response Accept (0x02) type 0x00 label 15 nosp 0
```

So this only show when it working which doesn't really help.

glanduin, 2025-03-27:
Ok i caught it. Here is the output. On this occasion, no aac was recognised. Couldn't switch to it.

[btmon (no aac).txt](https://github.com/user-attachments/files/19491682/btmon.no.aac.txt)

pv, 2025-03-27:
Reproduce:

1. rm -rf /var/lib/bluetooth/*
2. pair & connect & trust device
3. power headset off
4. systemctl restart bluetooth
5. power headset on
6. Headset connects avdtp, a2dp.c:endpoint_setconf_ind Set_Configuration_Ind in bluetoothd logs
7. endpoint.show
8. Only SBC visible

git bisect: ea3f4047e452b081e34d017cce316aba9d623301 is the first bad commit

In the other connect flow where we do sink_connect -> Set_Configuration_Cfm we always discover before config. In the Set_Configuration_Ind there should be reverse discovery to find them, but apparently it doesn't work properly with that commit removing caches.

Vudentz, 2025-03-27:
> Reproduce:
> 
>     1. rm -rf /var/lib/bluetooth/*
> 
>     2. pair & connect & trust device
> 
>     3. power headset off
> 
>     4. systemctl restart bluetooth
> 
>     5. power headset on
> 
>     6. Headset connects avdtp, a2dp.c:endpoint_setconf_ind Set_Configuration_Ind in bluetoothd logs
> 
>     7. endpoint.show
> 
>     8. Only SBC visible
> 
> 
> git bisect: [ea3f404](https://github.com/bluez/bluez/commit/ea3f4047e452b081e34d017cce316aba9d623301) is the first bad commit
> 
> In the other connect flow where we do sink_connect -> Set_Configuration_Cfm we always discover before config. In the Set_Configuration_Ind there should be reverse discovery to find them, but apparently it doesn't work properly with that commit removing caches.
> 
> `device_remove` is maybe not the right place to do what the commit intended to do, as the callback seems more related to removing device from the service (and is called on bluetoothd stop).

Right, the device_remove is called when the reloading the daemon, etc, but that doesn't mean it shall remove the stored data.

@fdanis-oss we should perhaps do the Endpoint cleanup in device_remove_stored rather then doing it in the plugin, or you tried doing that?

ValeZAA, 2025-03-28:
In the commit it says >This may prevent future connection, after a new pairing, to use the
best codec available.

So this is like some hack where new firmware adds a new A2DP codec or adds a new frequency, like 88.2 khz? Why not just revert it.

Vudentz, 2025-03-28:
@glanduin could you please test with the following change: https://github.com/tedd-an/bluez-upstream-test/commit/544042210c7165f6171fd6e8f0b3cbe6d6314b2e

glanduin, 2025-03-28:
Well i think that means you want me to build and install bluez from source. I can do that, no problem, but what i did was to modify the etc/bluetooth/main.conf and add:

"ReverseServiceDiscovery = true"

And it is possible that this fixed the issue for me. I need to test more and i will let you know.

Vudentz, 2025-03-28:
> Well i think that means you want me to build and install bluez from source. I can do that, no problem, but what i did was to modify the etc/bluetooth/main.conf and add:
>
> "ReverseServiceDiscovery = [...]

That is the default though, did you had as false before?

glanduin, 2025-03-28:
> That is the default though, did you had as false before?

No. It was # (commented out). So it was in its default state.

closed 2025-03-28 by BluezTestBot

ValeZAA, 2025-03-28:
Thinking about it more, it may make sense it now only shows SBC, since there is I am pretty sure no support of Samsung codec SSC (even worse its XQ version). Am I wrong?

Do you have any mechanism where the not supported codecs are removed from pactl list cards? At the very least the headphones do not work when i force 
pactl send-message /card/bluez_card.C4_77_64_XX_XX_XX/bluez switch-codec 2 (which is indeed SSC-HQ)

I will test aac thing, my sony supports AAC... BTW, just realised I cannot see any aac at all, LMAO!

>pactl send-message /card/bluez_card.88_C9_E8_XX_XX_XX/bluez list-codecs
[{"name":"8","description":"LDAC"},{"name":"1","description":"SBC"},{"name":"2","description":"SBC-XQ"}]

There was supposed to be AAC here right? WH-1000XM5 supports AAC and works in Windows in A2DP alternative driver hackery... BTW, what is up with LDAC, do you support all 24 bit encoding of LDAC and dynamic decrease of bitrate?

ValeZAA, 2025-03-28:
Used this instruction and now I have aac

https://tookmund.com/2024/02/aac-and-debian

root@kali:~# pactl send-message /card/bluez_card.88_C9_E8_XX_XX_XX/bluez list-codecs
[{"name":"8","description":"LDAC"},{"name":"4","description":"AAC"},{"name":"1","description":"SBC"},{"name":"2","description":"SBC-XQ"}]

glanduin, 2025-03-29:
> > > > Well i think that means you want me to build and install bluez from source. I can do that, no problem, but what i did was to modify the etc/bluetooth/main.conf and add:
> > > > "ReverseServiceDiscovery = true"
> > > > And it is possible that this fixed the issue for me. I need to test more and i will let you know.
> > > 
> > > 
> > > That is the default though, did you had as false before?
> > 
> > 
> > No. It was # (commented out). So it was in its default state.
> 
> Yeah, I think the problem is that sometimes there is a race where the AVDTP_Discovery is taking place and then in the meantime the remote side setup a stream, which probably aborts the discovery (maybe another bug).

The issue still persists with "ReverseServiceDiscovery = true". So yeah. I will rebuild with your fix and see if it fixes it (i am not skilled at these things, hopefuly i will manage).

glanduin, 2025-03-29:
Yeah i installed the "aur" version of bluez (perks of arch linux) and yeah, the latest commits are included in this, and it works. The issue is no longer there.