bluez/bluez #1149: Galaxy Buds3 Pro: LE audio causes a crash inside dmesg with Intel BE200

ValeZAA, 2025-03-28:
bluetooth/main.conf has

Experimental = true
KernelExperimental = 6fbaf188-05e0-496a-9885-d6ddfdb4e03e
Autoconnecttimeout=16380
ControllerMode = le

```
# btmgmt info
Index list with 1 item
hci0:	Primary controller
	addr E8:BF:B8:XX:XX:XX version 13 manufacturer 2 class 0x6c0104
	supported settings: powered connectable fast-connectable discoverable bondable link-security ssp br/edr le advertising secure-conn debug-keys privacy configuration static-addr phy-configuration **wide-band-speech cis-central cis-peripheral** 
	current settings: powered bondable ssp br/edr le secure-conn wide-band-speech cis-central cis-peripheral 
	name kali
	short name 
hci0:	Configuration options
	supported options: public-address 
	missing options: 

```

```
# bluetoothctl show
Controller E8:BF:B8:XX:XX:XX (public)
	Manufacturer: 0x0002 (2)
	Version: 0x0d (13)
	Name: kali
	Alias: kali
	Class: 0x006c0104 (7078148)
	Powered: yes
	PowerState: on
	Discoverable: no
	DiscoverableTimeout: 0x000000b4 (180)
	Pairable: yes
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (03b80e5a-ede8-4b33-a751-6ce34ec4c700)
	UUID: Audio Stream Control      (0000184e-0000-1000-8000-00805f9b34fb)
	UUID: Volume Offset Control     (00001845-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	UUID: Volume Control            (00001844-0000-1000-8000-00805f9b34fb)
	UUID: Microphone Control        (0000184d-0000-1000-8000-00805f9b34fb)
	UUID: Audio Input Control       (00001843-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree Audio Gateway   (0000111f-0000-1000-8000-00805f9b34fb)
	UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
	UUID: Published Audio Capabil.. (00001850-0000-1000-8000-00805f9b34fb)
	UUID: Broadcast Audio Scan      (0000184f-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
	Modalias: usb:v1D6Bp0246d0550
	Discovering: no
	Roles: central
	Roles: peripheral
	ExperimentalFeatures: BlueZ Experimental ISO... (6fbaf188-05e0-496a-9885-d6ddfdb4e03e)
Advertising Features:
	ActiveInstances: 0x00 (0)
	SupportedInstances: 0x0c (12)
	SupportedIncludes: tx-power
	SupportedIncludes: appearance
	SupportedIncludes: local-name
	SupportedSecondaryChannels: 1M
	SupportedSecondaryChannels: 2M
	SupportedSecondaryChannels: Coded
	SupportedCapabilities.MinTxPower: 0xffffffc0 (-64)
	SupportedCapabilities.MaxTxPower: 0x0042 (66)
	SupportedCapabilities.MaxAdvLen: 0xfb (251)
	SupportedCapabilities.MaxScnRspLen: 0xfb (251)
	SupportedFeatures: CanSetTxPower
	SupportedFeatures: HardwareOffload
Advertisement Monitor Features:
	SupportedMonitorTypes: or_patterns
```

bluetoothctl pair C4:77:64:XX:XX:XX

bluetoothctl connect C4:77:64:6C:84:8E

Causes this crash:

[  644.081855] ------------[ cut here ]------------
[  644.081864] WARNING: CPU: 13 PID: 2654 at net/bluetooth/hci_conn.c:567 hci_conn_timeout+0x61/0xc0 [bluetooth]
[  644.082058] Modules linked in: ccm uinput snd_seq_dummy snd_hrtimer rfcomm snd_seq snd_seq_device qrtr uhid cmac algif_hash algif_skcipher af_alg bnep binfmt_misc nls_ascii nls_cp437 vfat fat snd_sof_pci_intel_mtl snd_sof_intel_hda_generic soundwire_intel soundwire_generic_allocation iwlmvm soundwire_cadence snd_sof_intel_hda_common snd_soc_hdac_hda snd_sof_intel_hda_mlink snd_sof_intel_hda snd_sof_pci mac80211 snd_sof_xtensa_dsp intel_uncore_frequency intel_uncore_frequency_common snd_sof intel_pmc_core snd_sof_utils x86_pkg_temp_thermal snd_hda_ext_core intel_powerclamp snd_soc_acpi_intel_match coretemp snd_soc_acpi_intel_sdca_quirks snd_soc_acpi libarc4 snd_hda_codec_realtek kvm_intel snd_soc_core snd_hda_codec_generic snd_compress iwlwifi snd_hda_scodec_component snd_pcm_dmaengine kvm soundwire_bus snd_hda_codec_hdmi pmt_telemetry mei_gsc_proxy btusb pmt_class snd_soc_sdca btrtl intel_rapl_msr battery btintel btbcm snd_hda_intel btmtk rapl cfg80211 snd_intel_dspcfg intel_cstate snd_intel_sdw_acpi bluetooth
[  644.082170]  snd_hda_codec iTCO_wdt intel_pmc_bxt iTCO_vendor_support intel_uncore wmi_bmof spd5118 watchdog snd_hda_core joydev pcspkr snd_hwdep snd_pcm rfkill processor_thermal_device_pci processor_thermal_device snd_timer mei_me processor_thermal_wt_hint snd processor_thermal_rfim mei intel_vpu processor_thermal_rapl soundcore intel_rapl_common processor_thermal_wt_req processor_thermal_power_floor intel_vsec processor_thermal_mbox drm_shmem_helper int3403_thermal sg int340x_thermal_zone acpi_tad int3400_thermal acpi_thermal_rel acpi_pad evdev vmwgfx msr i2c_dev dm_mod parport_pc ppdev lp parport efi_pstore configfs nfnetlink efivarfs ip_tables x_tables autofs4 ext4 crc16 mbcache jbd2 crc32c_generic hid_logitech_hidpp xe drm_suballoc_helper hid_logitech_dj hid_generic sd_mod usbhid hid nouveau drm_gpuvm i915 drm_exec gpu_sched drm_ttm_helper drm_buddy ttm ahci i2c_algo_bit libahci drm_client_lib drm_display_helper cec crct10dif_pclmul rc_core crc32_pclmul xhci_pci crc32c_intel nvme drm_kms_helper
[  644.082303]  ghash_clmulni_intel libata xhci_hcd sha512_ssse3 nvme_core mxm_wmi usbcore scsi_mod drm i2c_i801 sha256_ssse3 nvme_auth igc thunderbolt intel_lpss_pci sha1_ssse3 scsi_common i2c_smbus usb_common intel_lpss idma64 video wmi fan button aesni_intel gf128mul crypto_simd cryptd
[  644.082352] CPU: 13 UID: 0 PID: 2654 Comm: kworker/u81:1 Tainted: G        W          6.13-rt-amd64 #1 PREEMPT_RT  Debian 6.13.8-1~exp1
[  644.082369] Tainted: [W]=WARN
[  644.082372] Hardware name: Micro-Star International Co., Ltd. MS-7E17/MPG Z890 CARBON WIFI (MS-7E17), BIOS 1.A54 03/11/2025
[  644.082377] Workqueue: hci0 hci_conn_timeout [bluetooth]
[  644.082540] RIP: 0010:hci_conn_timeout+0x61/0xc0 [bluetooth]
[  644.082694] Code: 48 c7 c1 9c 8b bb c1 83 e8 01 83 f8 08 76 3d 48 89 ea 48 c7 c6 aa 8b bb c1 48 c7 c7 c0 31 b6 c1 e8 84 1c 88 cb 45 85 e4 79 c4 <0f> 0b 0f b6 83 75 f7 ff ff 3c 01 74 1f be 13 00 00 00 3c 80 74 16
[  644.082701] RSP: 0018:ffffb89226fafe58 EFLAGS: 00010286
[  644.082709] RAX: ffffffffc1afe6c0 RBX: ffff8e16490498c8 RCX: ffff8e16714b0c28
[  644.082714] RDX: 0000000000000000 RSI: 000000000a200000 RDI: ffff8e16490498c8
[  644.082719] RBP: ffff8e1649049000 R08: ffff8e165e1810c0 R09: 0000000000000014
[  644.082724] R10: 0000000000000002 R11: 0000000000000000 R12: 00000000ffffffff
[  644.082729] R13: ffff8e1669730000 R14: ffff8e16490498c8 R15: 0000000000000000
[  644.082733] FS:  0000000000000000(0000) GS:ffff8e1da0080000(0000) knlGS:0000000000000000
[  644.082740] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  644.082745] CR2: 000055ae19a6f0a8 CR3: 00000002ba42c001 CR4: 0000000000f70ef0
[  644.082751] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
[  644.082754] DR3: 0000000000000000 DR6: 00000000ffff07f0 DR7: 0000000000000400
[  644.082758] PKRU: 55555554
[  644.082761] Call Trace:
[  644.082768]  <TASK>
[  644.082772]  ? hci_conn_timeout+0x61/0xc0 [bluetooth]
[  644.082922]  ? __warn.cold+0x93/0xf6
[  644.082938]  ? hci_conn_timeout+0x61/0xc0 [bluetooth]
[  644.083090]  ? report_bug+0xff/0x140
[  644.083101]  ? handle_bug+0x58/0x90
[  644.083107]  ? exc_invalid_op+0x17/0x70
[  644.083114]  ? asm_exc_invalid_op+0x1a/0x20
[  644.083130]  ? __pfx_hci_conn_timeout+0x10/0x10 [bluetooth]
[  644.083280]  ? hci_conn_timeout+0x61/0xc0 [bluetooth]
[  644.083427]  process_one_work+0x174/0x330
[  644.083440]  worker_thread+0x251/0x390
[  644.083449]  ? __pfx_worker_thread+0x10/0x10
[  644.083457]  kthread+0xcf/0x100
[  644.083469]  ? __pfx_kthread+0x10/0x10
[  644.083480]  ret_from_fork+0x31/0x50
[  644.083495]  ? __pfx_kthread+0x10/0x10
[  644.083505]  ret_from_fork_asm+0x1a/0x30
[  644.083522]  </TASK>
[  644.083524] ---[ end trace 0000000000000000 ]---

Vudentz, 2025-03-28:
What are the kernel and bluetoothd versions?

Vudentz, 2025-03-28:
> That was on commit [5df9521](https://github.com/bluez/bluez/commit/5df9521ce4d50ffa48153503bc5156c11e6ed26b)
> 
> Kernel 6.13.8 is mentioned in the stack trace

Any other trace, btmon, etc? Btw, you don't need to do pair + connect, in fact these are mutually exclusive with: https://github.com/bluez/bluez/commit/1c2ec29900e48c785f7faa8fbe88f3085aaf806b

ValeZAA, 2025-03-28:
Also caused [ 3513.020065] bluetoothd[3468]: segfault at fffffffffffffff9 ip 00007f1026ce24aa sp 00007ffeb90a6a60 error 5 in libc.so.6[a24aa,7f1026c68000+165000] likely on CPU 5 (core 28, socket 0)
[ 3513.020075] Code: 0f 6b 10 00 48 8d 3d 95 c8 10 00 e8 e0 5e f9 ff 48 85 ff 0f 84 bf 00 00 00 55 48 8d 77 f0 53 48 83 ec 18 48 8b 1d 56 49 14 00 <48> 8b 47 f8 64 8b 2b a8 02 75 5b 48 8b 15 e4 48 14 00 64 48 83 3a

Vudentz, 2025-03-28:
This is quite weird if you got HFP over LE, since you have LE only mode in main.conf, or that is not in effect? Anyway these traces are quite useless without the btmon logs to know what is the traffic that are causing the crash.

ValeZAA, 2025-03-28:
>This is quite weird if you got HFP over LE, since you have LE only mode in main.conf

Yes, I switched back to dual, cause it does not pair in le mode anyway.

Vudentz, 2025-03-28:
> Do you have any idea why LE scan shows my buds3 pro have like 10 different MAC addresses from different OUIs (3 first bytes) not even from Samsung?? It is using some cloaking??

Perhaps it is using RPA/Privacy, which can change over time, but changing they shouldn't be advertising this many, otherwise it is quite hard to know what is the actual address to connect to.

> It also has two addresses from actual Samsung though. One LE audio MAC and another from Bluetooth classic. info or connect does not work on those MACs.

Are they both considered visible? bluetoothctl will gray out the ones no supposed to be visible. Also if you have LE only enable they obviously it should see the classic, but perhaps what is going on is that you see 2 address because they are a coordinate set, in that case both are LE but one will be the primary address, normally discoverable, while the secondary is non-discoverable then upon connecting to the primary address it will detect that it is a set and connect the second automatically, well at least that how it should work, but I think the likes of gnome-setting don't know how to distinguish discoverable and non-discoverable devices yet.

ValeZAA, 2025-03-28:
>I think the likes of gnome-setting don't know how to distinguish discoverable and non-discoverable devices yet.

Yes, on windows 11 it works just fine (I had to install a special driver Intel Smart Sound for BLE) and then it pairs, after pairing the first it right there asks to pair the second with cool animation. But there is no sound after that... On Linux bluetoothctl is supposed to ask to pair with the second earbud. Even that does not happen.

Anyway, here is this thing called Resolvable Set Identifier:

```
@ MGMT Event: Device Found (0x0012) plen 160                                                                                                                         {0x0001} [hci0] 16.465894
        LE Address: C4:77:64:45:AD:24 (Samsung Electronics Co.,Ltd)
        RSSI: -51 dBm (0xcd)
        Flags: 0x00000000
        Data length: 146
        Flags: 0x04
          BR/EDR Not Supported
        Appearance: Undefined (0x0941)
        Service Data: Telephony and Media Audio Profile Role (0x2b51)
          Data[2]: 2a00
        Resolvable Set Identifier: 6E-4A-38-AC-28-B3
          Hash: 0xac28b3
          Random: 0x6e4a38
        Service Data: Common Audio (0x1853)
          Data[1]: 00
        Service Data: Audio Stream Control (0x184e)
          Data[6]: 006f006b0000
        16-bit Service UUIDs (partial): 5 entries
          Audio Stream Control (0x184e)
          Broadcast Audio Scan (0x184f)
          Published Audio Capabilities (0x1850)
          Volume Control (0x1844)
          Microphone Control (0x184d)
        Company: Samsung Electronics Co. Ltd. (117)
          Data[43]: 020901000000150300000000000000000000000000000000000000030154015b4865616470686f6e655d20
        Service Data UUID 128: Vendor specific 
          Data[5]: 01548e8409
        Name (complete): xxxxx's Buds3 Pro
```

And the other earbud is this one:

```
@ MGMT Event: Device Found (0x0012) plen 160                                                                                                                          {0x0001} [hci0] 9.720765
        LE Address: 7F:CE:39:23:F8:EF (Resolvable)
        RSSI: -69 dBm (0xbb)
        Flags: 0x00000000
        Data length: 146
        Flags: 0x04
          BR/EDR Not Supported
        Appearance: Undefined (0x0941)
        Service Data: Telephony and Media Audio Profile Role (0x2b51)
          Data[2]: 2a00
        Resolvable Set Identifier: 6E-4A-38-AC-28-B3
          Hash: 0xac28b3
          Random: 0x6e4a38
        Service Data: Common Audio (0x1853)
          Data[1]: 00
        Service Data: Audio Stream Control (0x184e)
          Data[6]: 006f006b0000
        16-bit Service UUIDs (partial): 5 entries
          Audio Stream Control (0x184e)
          Broadcast Audio Scan (0x184f)
          Published Audio Capabilities (0x1850)
          Volume Control (0x1844)
          Microphone Control (0x184d)
        Company: Samsung Electronics Co. Ltd. (117)
          Data[43]: 020901000000150300000000000000000000000000000000000000030154015b4865616470686f6e655d20
        Service Data UUID 128: Vendor specific 
          Data[5]: 01548e840d
        Name (complete): xxxxx's Buds3 Pro
```

Also there is some all zeros address:

```
> HCI Event: LE Meta Event (0x3e) plen 172                                                                                                                                #43 [hci0] 16.465723
      LE Extended Advertising Report (0x0d)
        Num reports: 1
        Entry 0
          Event type: 0x0001
            Props: 0x0001
              Connectable
            Data status: Complete
          Address type: Public (0x00)
          Address: C4:77:64:45:AD:24 (Samsung Electronics Co.,Ltd)
          Primary PHY: LE 1M
          Secondary PHY: LE 2M
          SID: 0x03
          TX power: 127 dBm
          RSSI: -51 dBm (0xcd)
          Periodic advertising interval: 0.00 msec (0x0000)
          Direct address type: Public (0x00)
          Direct address: 00:00:00:00:00:00 (OUI 00-00-00)
          Data length: 0x92
        02 01 04 03 19 41 09 05 16 51 2b 2a 00 07 2e b3  .....A...Q+*....
        28 ac 38 4a 6e 04 16 53 18 00 09 16 4e 18 00 6f  (.8Jn..S....N..o
        00 6b 00 00 0b 02 4e 18 4f 18 50 18 44 18 4d 18  .k....N.O.P.D.M.
        2e ff 75 00 02 09 01 00 00 00 15 03 00 00 00 00  ..u.............
        00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 03  ................
        01 54 01 5b 48 65 61 64 70 68 6f 6e 65 5d 20 16  .T.[Headphone] .
        21 3a 04 8f 2b a9 7e a6 ae 1b 49 c6 19 e9 73 a4  !:..+.~...I...s.
        a7 01 54 8e 84 09 1b 09 d0 92 d0 b0 d0 bb d0 b5  ..T.............
        xx xx d0 xx d0 xx xx xx xx 42 75 64 73 33 20 50  ......'s Buds3 P
        72 6f                                            ro              
        Flags: 0x04
          BR/EDR Not Supported
        Appearance: Undefined (0x0941)
        Service Data: Telephony and Media Audio Profile Role (0x2b51)
          Data[2]: 2a00
        Resolvable Set Identifier: 6E-4A-38-AC-28-B3
          Hash: 0xac28b3
          Random: 0x6e4a38
        Service Data: Common Audio (0x1853)
          Data[1]: 00
        Service Data: Audio Stream Control (0x184e)
          Data[6]: 006f006b0000
        16-bit Service UUIDs (partial): 5 entries
          Audio Stream Control (0x184e)
          Broadcast Audio Scan (0x184f)
          Published Audio Capabilities (0x1850)
          Volume Control (0x1844)
          Microphone Control (0x184d)
        Company: Samsung Electronics Co. Ltd. (117)
          Data[43]: 020901000000150300000000000000000000000000000000000000030154015b4865616470686f6e655d20
        Service Data UUID 128: Vendor specific 
          Data[5]: 01548e8409
        Name (complete): xxxxxx's Buds3 Pro
```

ValeZAA, 2025-03-29:
On Android 14 of Galaxy S24 Ultra different mac addresses are shown 

[image]

[image]

Here is GATT, BTW @Vudentz how do you see that data?

[image]

Vudentz, 2025-03-31:
> Yes, on windows 11 it works just fine (I had to install a special driver Intel Smart Sound for BLE) and then it pairs, after pairing the first it right there asks to pair the second with cool [...]

It is an RPA, thus why the address changes.

> Also there is some all zeros address:
>
> ```
>       LE Extended Advertising Report (0x0d)
>         Num reports: 1
>         Entry 0
>           Event type: 0x0001
>             Props: 0x0001 [...]

Direct address field is only used for directed advertisement report, which is not the case above. Can you paste the full btmon traces when you attempt to pair?

Vudentz, 2025-03-31:
> Which MAC do I need to pair to? Probably the one of the left earbud: 7F:CE:39:23:F8:EF cause it says (Resolvable)?

Yep.

> How can I tell it is using RPA or there is no way to tell, besides weird MACs? Do I need to enable privacy first on Intel adapter?
> 
> btmgmt --index hci0 power off btmgmt --index hci0 privacy on btmgmt --index hci0 power on

No, privacy command is for local privacy, you can't really control remote privacy.

> > Direct address field is only used for directed advertisement report, which is not the case above.
> 
> That is where you get all 00 address in issue [#1103](https://github.com/bluez/bluez/issues/1103) maybe that one needs to be fixed first?

Don't think so, we only used to directed address to compare with the local address, if you see all 00 then there must be someone advertising it over the air, are you perhaps in a crowed area with a lot advertisements?

ValeZAA, 2025-03-31:
>you perhaps in a crowed area with a lot advertisements?

No, I am home...

Direct address type: Public (0x00)
          Direct address: 00:00:00:00:00:00 (OUI 00-00-00)
          Data length: 0x92

is spam from Buds3 pro.

ValeZAA, 2025-04-27:
This bug is just caused by weird gnome issue that it does not init (scan bredr) stuff. And that causes a timeout crash. BTW, some progress in connection: pactl list cards now shows 

bap-duplex !!! LETS GO! BE200 is a good card??

closed 2025-09-07 by github-actions[bot]