bluez/bluez #1150: Problem reconnecting

iteralind, 2025-03-29:
Hi,
after upgrading to 5.80-1 (I'm on Manjaro Linux) I am experiencing trouble in reconnecting a bluetooth mouse (Dell) and Keyboard (Logitech Pebble) after either disconnecting manually or putting the computer into sleep mode. Pairing the devices anew works and the devices stay connected. I have checked journalctl and since shortly after the update I see the following error messages: "src/device.c:set_wake_allowed_complete() Set device flags return status: Invalid Parameters". Oddly enough disconnecting and reconnecting works fine for my Headset (Jabra). Ist this a know issue and if so is there a solution for this?

alekho, 2025-03-30:
I have a similar issue. The MX Master 3S mouse does not connect or pair via the graphical interface. I managed to connect it only through bluetoothctl and only when putting the mouse into pairing mode. However, reconnection after a reboot is impossible. I have to delete the device and repeat the steps. I downgraded the bluez package to version 5.79-1, and now everything works.

iteralind, 2025-03-30:
I should add that I can also only re-pair using bluetoothctl. Pairing via the UI does not work for me either. I also have to delete the device first.

andrewib, 2025-04-02:
I experience the exact same problems with a Logitech MX Master mouse. Also on Manjaro.

Eduruiz, 2025-04-02:
Something similar is happening here, I'm on arch linux and my aula f75 keyboard can't reconnect after system reboot, or keyboard "idle" state where I think it kinda turns off (this happens a lot if you don't type for a while).  
I can reconnect using bluetoothctl, but I see that the keyboard's MAC address always changes, so I get about 8 different devices for the same keyboard.  

I left btmon running and waited for the disconnect. when waking up the keyboard again, I got:
```
HCI Event: LE Meta Event (0x3e) plen 18                                                                                                        #2367 [hci0] 9968.586385
      LE Direct Advertising Report (0x0b)
        Num reports: 1
        Event type: Connectable directed - ADV_DIRECT_IND (0x01)
        Address type: Random (0x01)
        Address: D3:49:60:2D:6E:B9 (Static)
        Direct address type: Random (0x01)
        Direct address: 04:D3:B0:C5:0B:0B (Non-Resolvable)
        RSSI: -50 dBm (0xce)
```

not sure if it helps, but if someone wants some tests and can give me instructions,  I can run it.
after downgrading to bluez version `5.79-1`, the problem no longer happens.

andrewib, 2025-04-09:
Disconnection seems to happen after a while even if the system has not been idled or rebooted. I also notice the Logitech MX Master mouse cycles through different MAC adresses over time, although it does not seem to be very many MAC addresses (perhaps 10 different ones, always the last part og the MAC is changing)

andrewib, 2025-04-13:
Ah, it seems to have been fixed in version 5.82-1! 

At least it is for me.

bdux, 2025-05-04:
Still happens for me with a DualShock 4 controller in version 5.82-1.

Vishal8523, 2025-07-11:
Any updates on this ?

mkurz, 2025-11-03:
not stale