bluez/bluez #1152: Bluetooth HeadPhone failed to connect, or connected but sound choppy, or connected but disconnect.

eval-exec, 2025-03-29:
Hello,

Thank for your great work on this project, I have no idea why my Sony WF1000XM5 not work well on my laptop:
The related config is:
```bash
❯ cat /etc/bluetooth/main.conf
[General]
ControllerMode=dual
Experimental=true
FastConnectable=true

[Policy]
AutoEnable=true
```
2. version
```bash
bluetoothd version: 5.78
```

What information should I provide to help you investigate this issue?

```
Mar 30 00:41:19 Mufasa systemd[1]: Starting Bluetooth service...
Mar 30 00:41:19 Mufasa (uetoothd)[87536]: bluetooth.service: ConfigurationDirectory 'bluetooth' already exists but the mode is different. (File system: 755 ConfigurationDirectoryMode: 555)
Mar 30 00:41:19 Mufasa bluetoothd[87536]: Bluetooth daemon 5.78
Mar 30 00:41:19 Mufasa systemd[1]: Started Bluetooth service.
Mar 30 00:41:19 Mufasa bluetoothd[87536]: Starting SDP server
Mar 30 00:41:19 Mufasa bluetoothd[87536]: src/plugin.c:init_plugin() System does not support ccp plugin
Mar 30 00:41:19 Mufasa bluetoothd[87536]: Bluetooth management interface 1.22 initialized
Mar 30 00:41:19 Mufasa bluetoothd[87536]: Battery Provider Manager created
Mar 30 00:41:19 Mufasa bluetoothd[87536]: profiles/audio/bap.c:bap_adapter_probe() BAP requires ISO Socket which is not enabled
Mar 30 00:41:19 Mufasa bluetoothd[87536]: bap: Operation not supported (95)
Mar 30 00:41:19 Mufasa bluetoothd[87536]: Adv Monitor Manager created with supported features:0x00000000, enabled features:0x00000000, max number of supported monitors:32, max number of supported patterns:16
Mar 30 00:41:19 Mufasa bluetoothd[87536]: Endpoint registered: sender=:1.73 path=/MediaEndpoint/A2DPSource/ldac
Mar 30 00:41:19 Mufasa bluetoothd[87536]: Endpoint registered: sender=:1.73 path=/MediaEndpoint/A2DPSink/aptx_hd
Mar 30 00:41:19 Mufasa bluetoothd[87536]: Endpoint registered: sender=:1.73 path=/MediaEndpoint/A2DPSource/aptx_hd
Mar 30 00:41:19 Mufasa bluetoothd[87536]: Endpoint registered: sender=:1.73 path=/MediaEndpoint/A2DPSink/aptx
Mar 30 00:41:19 Mufasa bluetoothd[87536]: Endpoint registered: sender=:1.73 path=/MediaEndpoint/A2DPSource/aptx
Mar 30 00:41:19 Mufasa bluetoothd[87536]: Endpoint registered: sender=:1.73 path=/MediaEndpoint/A2DPSink/aac
Mar 30 00:41:19 Mufasa bluetoothd[87536]: Endpoint registered: sender=:1.73 path=/MediaEndpoint/A2DPSource/aac
Mar 30 00:41:19 Mufasa bluetoothd[87536]: Endpoint registered: sender=:1.73 path=/MediaEndpoint/A2DPSink/opus_g
Mar 30 00:41:19 Mufasa bluetoothd[87536]: Endpoint registered: sender=:1.73 path=/MediaEndpoint/A2DPSource/opus_g
Mar 30 00:41:19 Mufasa bluetoothd[87536]: Endpoint registered: sender=:1.73 path=/MediaEndpoint/A2DPSink/sbc
Mar 30 00:41:19 Mufasa bluetoothd[87536]: Endpoint registered: sender=:1.73 path=/MediaEndpoint/A2DPSource/sbc
Mar 30 00:41:19 Mufasa bluetoothd[87536]: Endpoint registered: sender=:1.73 path=/MediaEndpoint/A2DPSource/aptx_ll_1
Mar 30 00:41:19 Mufasa bluetoothd[87536]: Endpoint registered: sender=:1.73 path=/MediaEndpoint/A2DPSource/aptx_ll_0
Mar 30 00:41:19 Mufasa bluetoothd[87536]: Endpoint registered: sender=:1.73 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_1
Mar 30 00:41:19 Mufasa bluetoothd[87536]: Endpoint registered: sender=:1.73 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_0
Mar 30 00:41:19 Mufasa bluetoothd[87536]: Endpoint registered: sender=:1.73 path=/MediaEndpoint/A2DPSource/faststream
Mar 30 00:41:19 Mufasa bluetoothd[87536]: Endpoint registered: sender=:1.73 path=/MediaEndpoint/A2DPSource/faststream_duplex
Mar 30 00:41:19 Mufasa bluetoothd[87536]: Endpoint registered: sender=:1.73 path=/MediaEndpoint/A2DPSink/opus_05
Mar 30 00:41:19 Mufasa bluetoothd[87536]: Endpoint registered: sender=:1.73 path=/MediaEndpoint/A2DPSource/opus_05
Mar 30 00:41:19 Mufasa bluetoothd[87536]: Endpoint registered: sender=:1.73 path=/MediaEndpoint/A2DPSink/opus_05_duplex
Mar 30 00:41:19 Mufasa bluetoothd[87536]: Endpoint registered: sender=:1.73 path=/MediaEndpoint/A2DPSource/opus_05_duplex
Mar 30 00:41:34 Mufasa bluetoothd[87536]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down
Mar 30 00:42:03 Mufasa bluetoothd[87536]: profiles/audio/avdtp.c:cancel_request() Start: Connection timed out (110)
Mar 30 00:42:05 Mufasa bluetoothd[87536]: profiles/audio/avdtp.c:cancel_request() Abort: Connection timed out (110)
Mar 30 00:42:38 Mufasa bluetoothd[87536]: profiles/input/device.c:control_connect_cb() connect to AC:80:0A:46:F7:9D: Connection refused (111)
Mar 30 00:42:44 Mufasa bluetoothd[87536]: src/device.c:search_cb() AC:80:0A:46:F7:9D: error updating services: Connection timed out (110)
Mar 30 00:42:44 Mufasa bluetoothd[87536]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
Mar 30 00:42:54 Mufasa bluetoothd[87536]: src/profile.c:ext_connect() Hands-Free Voice gateway failed connect to AC:80:0A:46:F7:9D: Connection refused (111)
Mar 30 00:42:54 Mufasa bluetoothd[87536]: profiles/audio/avdtp.c:try_send() send: Transport endpoint is not connected (107)
Mar 30 00:42:54 Mufasa bluetoothd[87536]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for AC:80:0A:46:F7:9D: Input/output error
Mar 30 00:43:11 Mufasa bluetoothd[87536]: profiles/audio/avdtp.c:cancel_request() Open: Connection timed out (110)
Mar 30 00:43:13 Mufasa bluetoothd[87536]: profiles/audio/avdtp.c:cancel_request() Abort: Connection timed out (110)
Mar 30 00:47:47 Mufasa bluetoothd[87536]: No matching connection for device
Mar 30 00:47:54 Mufasa bluetoothd[87536]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down
Mar 30 00:48:08 Mufasa bluetoothd[87536]: /org/bluez/hci0/dev_AC_80_0A_46_F7_9D/sep2/fd2: fd(43) ready
Mar 30 00:48:27 Mufasa bluetoothd[87536]: /org/bluez/hci0/dev_AC_80_0A_46_F7_9D/sep3/fd3: fd(43) ready
Mar 30 00:48:49 Mufasa bluetoothd[87536]: /org/bluez/hci0/dev_AC_80_0A_46_F7_9D/sep2/fd4: fd(43) ready
Mar 30 00:52:59 Mufasa bluetoothd[87536]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
Mar 30 00:53:36 Mufasa bluetoothd[87536]: src/profile.c:ext_connect() Hands-Free Voice gateway failed connect to AC:80:0A:46:F7:9D: Connection refused (111)
Mar 30 00:53:39 Mufasa bluetoothd[87536]: /org/bluez/hci0/dev_AC_80_0A_46_F7_9D/sep2/fd5: fd(43) ready
Mar 30 00:54:14 Mufasa bluetoothd[87536]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
Mar 30 00:54:21 Mufasa bluetoothd[87536]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down
Mar 30 00:54:28 Mufasa bluetoothd[87536]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down
Mar 30 00:54:37 Mufasa bluetoothd[87536]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down
Mar 30 00:54:51 Mufasa bluetoothd[87536]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down
Mar 30 00:55:19 Mufasa bluetoothd[87536]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Function not implemented
Mar 30 00:55:25 Mufasa bluetoothd[87536]: profiles/audio/avdtp.c:avdtp_connect_cb() connect to AC:80:0A:46:F7:9D: Function not implemented (38)
Mar 30 01:00:14 Mufasa bluetoothd[87536]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Connection timed out
Mar 30 01:00:16 Mufasa bluetoothd[87536]: plugins/policy.c:reconnect_timeout() Reconnecting services failed: Device or resource busy (16)
Mar 30 01:00:20 Mufasa bluetoothd[87536]: profiles/audio/avdtp.c:avdtp_connect_cb() connect to AC:80:0A:46:F7:9D: Host is down (112)

```

ValeZAA, 2025-03-31:
Sony WF1000XM5 is LE audio device, btw. It may have all kinds of problems and weird quirks.

>Experimental=true

What for? That is not enough to connect in LE audio mode.

eval-exec, 2025-03-31:
> What for? That is not enough to connect in LE audio mode.

To show battery percentage.

ValeZAA, 2025-03-31:
>What I need to connect in LE audio mode?

ControllerMode=le

Experimental=true

KernelExperimental = 6fbaf188-05e0-496a-9885-d6ddfdb4e03e

eval-exec, 2025-04-02:
> > What I need to connect in LE audio mode?
> 
> ControllerMode=le
> 
> Experimental=true
> 
> KernelExperimental = 6fbaf188-05e0-496a-9885-d6ddfdb4e03e

Thank you, but Unlucky It not fixed my issue, :-(

ValeZAA, 2025-04-03:
I have the same issue with my Galaxy Buds3 pro. It appears that the bug is that in those settings in main.conf break audio. It does connect but not audio.

Now, you MUST reboot after change in config. 

service bluetooth restart is not enough for whatever strange reason.

closed 2025-09-07 by github-actions[bot]