bluez/bluez #1158: [Bug Report] NULL pointer dereference in BlueZ CLI tool when configuring endpoint without preset

littlehuanua-1127, 2025-04-02:
When configuring a local endpoint without specifying a preset and manually entering codec configuration bytes, the CLI tool crashes due to a NULL pointer dereference in util_iov_append().

1. Start the CLI tool:
gdb ./bluetoothctl
2. Register a local SBC Source endpoint:
register 0000110A-0000-1000-8000-00805F9B34FB 0x00 20 15 80 20 03 01 53
3. Run configuration 
config /org/bluez/hci0/dev_90_98_38_F9_56_5A/sep1 /local/endpoint/ep0
4. When prompted:
[/local/endpoint/ep0] Enter configuration:Enter this SBC configuration:
20 15 2A 35
5. The CLI crashes immediately with:
Program received signal SIGSEGV, Segmentation fault.
in util_iov_append (iov=0x0, ...) at src/shared/util.c

[image]

🧠 Root cause:
The cfg->caps field is not initialized unless a preset is used. However, when the user enters configuration manually, cfg->caps remains NULL. Later, it is passed to util_iov_append() in endpoint_config() without a NULL check, causing a crash.

🔎 Environment:
BlueZ version: 5.80
System: Ubuntu 24.04

ValeZAA, 2025-04-02:
Does it do undefined behaviour if you just run

bluetoothctl -e

Vudentz, 2025-04-02:
@littlehuanua-1127 looks like a bug indeed, it should also not request fields that are specific to LE Audio, note though that in theory the likes of pipewire are what we expect to use normally, or you are trying to test something else? bluetoothctl is support for endpoints is more of a testing tool to check if D-Bus interfaces, etc, are working as intended.

ValeZAA, 2025-04-03:
I noticed that if you change the config midway from dual to le bluetoothctl -e now prints both lc3 and a2dp stuff.

Looks like service bluetooth restart is not enough. Reboot is needed.

littlehuanua-1127, 2025-04-07:
@@Vudentz  To clarify my use case: I’m currently working on Bluetooth protocol testing and vulnerability validation, particularly focusing on how endpoint configuration and transport acquisition are handled at the BlueZ layer.

closed 2025-09-06 by github-actions[bot]