bluez/bluez #1162: Soundcore Sport X20 Earbuds won't stay connected on my Gigabyte B650 Elite AX v2 Motherboard

polluxau, 2025-04-06:
Im trying to use these earbuds on my main pc and have zero issues using it on my 2015 Acer Travelmate P-236-M laptop on Pop!_os

But on my Arch PC its constantly fighting to stay connected or play audio through them which then it gives up and disconnects

```
Apr 06 12:04:21 cachyos-x8664 bluetoothd[707]: /org/bluez/hci0/dev_B0_38_E2_65_87_1A/fd9: fd(31) ready
Apr 06 12:06:11 cachyos-x8664 bluetoothd[707]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
Apr 06 12:06:18 cachyos-x8664 bluetoothd[707]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down
Apr 06 12:06:25 cachyos-x8664 bluetoothd[707]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down
Apr 06 12:06:34 cachyos-x8664 bluetoothd[707]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down
Apr 06 12:06:47 cachyos-x8664 bluetoothd[707]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down
```

eval-exec, 2025-04-16:
I have same issue too for `Sony WF-1000XM5`:
```
Apr 16 19:19:07 Mufasa bluetoothd[1112]: Endpoint registered: sender=:1.103 path=/MediaEndpoint/A2DPSource/opus_05_duplex
Apr 16 19:19:08 Mufasa bluetoothd[1112]: plugins/policy.c:reconnect_timeout() Reconnecting services failed: Operation already in progress (114)
Apr 16 19:19:39 Mufasa bluetoothd[1112]: /org/bluez/hci0/dev_AC_80_0A_46_F7_9D/sep2/fd2: fd(45) ready
Apr 16 19:31:53 Mufasa bluetoothd[1112]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
Apr 16 19:31:59 Mufasa bluetoothd[1112]: profiles/audio/avdtp.c:avdtp_connect_cb() connect to AC:80:0A:46:F7:9D: Host is down (112)
Apr 16 19:32:06 Mufasa bluetoothd[1112]: profiles/audio/avdtp.c:avdtp_connect_cb() connect to AC:80:0A:46:F7:9D: Host is down (112)
Apr 16 19:32:16 Mufasa bluetoothd[1112]: profiles/audio/avdtp.c:avdtp_connect_cb() connect to AC:80:0A:46:F7:9D: Host is down (112)
Apr 16 19:32:30 Mufasa bluetoothd[1112]: profiles/audio/avdtp.c:avdtp_connect_cb() connect to AC:80:0A:46:F7:9D: Host is down (112)
Apr 16 19:32:52 Mufasa bluetoothd[1112]: profiles/audio/avdtp.c:avdtp_connect_cb() connect to AC:80:0A:46:F7:9D: Host is down (112)
Apr 16 19:34:23 Mufasa bluetoothd[1112]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
Apr 16 20:13:42 Mufasa bluetoothd[1112]: /org/bluez/hci0/dev_AC_80_0A_46_F7_9D/sep2/fd4: fd(43) ready
Apr 16 20:25:41 Mufasa bluetoothd[1112]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
Apr 16 20:33:15 Mufasa bluetoothd[1112]: /org/bluez/hci0/dev_AC_80_0A_46_F7_9D/sep2/fd5: fd(43) ready
Apr 16 20:40:43 Mufasa bluetoothd[1112]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
Apr 16 20:40:51 Mufasa bluetoothd[1112]: /org/bluez/hci0/dev_AC_80_0A_46_F7_9D/sep2/fd6: fd(43) ready
```
```bash
❯ cat /etc/bluetooth/main.conf
[General]
ControllerMode=dual
Experimental=true
FastConnectable=true

[Policy]
AutoEnable=true
```
```bash
❯ /nix/store/c2imbiwhhnhynfb74r0s0qd2s6kkyqqb-bluez-5.78/libexec/bluetooth/bluetoothd --version
5.78
```

AlkyoneAoide, 2025-06-03:
Having what I believe is the same issue, WH-1000XM5 and a Soundcore Select 4 Go on Arch Linux on a Framework Laptop 13.  After connecting, audio cuts out randomly until it eventually disconnects.  I first noticed the issue early May.

Bluetoothd 5.83, Linux kernel 6.14.9-arch1-1, and using an Intel AX210 wifi/bluetooth card.

```
Jun 03 11:04:52 AngryCat bluetoothd[8836]: Endpoint registered: sender=:1.40 path=/MediaEndpoint/A2DPSource/opus_05_duplex
Jun 03 11:04:58 AngryCat bluetoothd[8836]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down
Jun 03 11:05:03 AngryCat bluetoothd[8836]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down
Jun 03 11:05:45 AngryCat bluetoothd[8836]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down
Jun 03 11:06:26 AngryCat bluetoothd[8836]: /org/bluez/hci0/dev_88_C9_E8_F0_2B_65/fd0: fd(31) ready
Jun 03 11:06:45 AngryCat bluetoothd[8836]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down
Jun 03 11:07:42 AngryCat bluetoothd[8836]: profiles/audio/avdtp.c:handle_unanswered_req() No reply to Suspend request
Jun 03 11:07:42 AngryCat bluetoothd[8836]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
Jun 03 11:07:45 AngryCat bluetoothd[8836]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down
Jun 03 11:07:50 AngryCat bluetoothd[8836]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down
```

pollux78, 2025-06-05:
SOLVED: i needed the wifi antenna which supposedly helps with Bluetooth which it does, no issues with connecting my earbuds anymore :)

[image]

pollux78, 2025-07-10:
If anyone is having a issue similar to this I would highly recommend finding a antenna for your mobo or buying a Bluetooth dongle as they can be rather cheap

If your still experiencing the issue I would open a separate report about said issue with your audio device

closed 2025-07-10 by pollux78