bluez/bluez #1165: Supporting the DualShock 3 without disabling the ClassicBondedOnly option

RyuzakiKK, 2025-04-09:
Currently to use the DualShock 3 controller (sixaxis) you need to set the option `ClassicBondedOnly=false`, which exposes you to the CVE-2023-45866.

We are trying to come up with an alternative solution to support the DualShock 3, while limiting the attach surface as much as possible.

The high level design is to:
- Keep the `ClassicBondedOnly` option enabled
- During the custom USB pairing process of the DualShock 3, we record its MAC in a special local config storage
- When BlueZ receives a new device connection, if it is a known Sixaxis MAC from that local config, we accept HID reports from that device as if we had `ClassicBondedOnly=false`
- To reduce the possible attacks, we can limit the allowed input events (bpf-hid/pre-uhid filtering) to only what a real DualShock 3 is capable of, e.g. don't allow it to send arbitrary keyboard events
- Expose a new property to notify clients that a possible insecure device is currently connected

An attacker could still sniff the sixaxis MAC address and impersonate it, but this should narrow the scope considerably compared to just using `ClassicBondedOnly=false`.

One problem we are currently facing with that approach, is that by default the server is started with a `bt_io_listen()` that uses the `BT_IO_SEC_MEDIUM` option. That security option let the kernel enable encryption on all bonded devices, which breaks the DualShock 3 communications (eventually asking for a PIN that the device doesn't expect).

Before continuing with this approach, @Vudentz I'd really like to know your opinion, in case you have any pointers or suggestions for a different design.

Vudentz, 2025-04-10:
@RyuzakiKK ClassicBondedOnly is actually not to bad consider we do the following:

```
	/* Encryption is mandatory for keyboards */
	/* Some platforms may choose to require encryption for all devices */
	/* Note that this only matters for pre 2.1 devices as otherwise the */
	/* device is encrypted by default by the lower layers */
	if (classic_bonded_only || idev->type == BT_UHID_KEYBOARD) {
```

So keyboards still enforces encryption, that said perhaps you are saying that we don't enforce keyboard events to be limited to only keyboards type.

Btw, how other system handle this? Or the likes of playstation do have some means to encrypt the link in a non-standard way?

legluondunet, 2025-04-11:
DualShock 4 controller needs to set the option ClassicBondedOnly=false too.

RyuzakiKK, 2025-04-11:
We started to look into this one because with ClassicBondedOnly it seems like an attacker could simply sniff a paired device MAC address and clone it as a pre 2.1 device.

E.g. in theory if you have any paired gamepad controller, anyone with a flipper zero could fake a new device similar to the sixaxis with the same paired MAC address. And in that instance the host will happily let that faked device connect without any encryption.

So even if there will be a way to limit this unsecure mode to only the gamepad class, it would still allow anyone that knows a paired MAC address to initiate a connection, which is not great.

That's why we wanted something more strict, like an explicit list of MACs that you are willing to accept in that unsecure mode.
This would limit the attack vector to only users that consciously paired a DualShock 3 device in the first place. While leaving all the others still guarded.

With a possible end result being:
- If you only paired non-sixaxis devices, you should be under the same security level as `ClassicBondedOnly=true`
- If you paired a sixaxis device, only that MAC address could expose you to external threats
- Mark that paired device as "special"
  - In this way clients could expose a notification/icon when that device is in use
- Enforce a strict hid filtering to only the inputs that the DualShock 3 can do 

> So keyboards still enforces encryption, that said perhaps you are saying that we don't enforce keyboard events to be limited to only keyboards type.

Correct. As far as I'm aware the Bluetooth class is only used to identify the type of device without any real enforcement. E.g. I think a device could declare itself as an HID card reader (avoiding the `idev->type == BT_UHID_KEYBOARD` check) but still being able to inject keystrokes.

RyuzakiKK, 2025-04-11:
> DualShock 4 controller needs to set the option ClassicBondedOnly=false too.

I'm assuming this is not an original Sony controller? Because the DualShock 4 should work correctly without changing the ClassicBondedOnly option.

legluondunet, 2025-04-11:
"I'm assuming this is not an original Sony controller?"
It's possible, I bought mine on Ebay, looks like a true DualShock 4 controller, but could be a clone...

https://www.ebay.fr/itm/365456812155

legluondunet, 2025-04-11:
more info about DualShock 4 v2:

```
$dmesg:
[32235.030832] playstation 0005:054C:05C4.0009: hidraw3: BLUETOOTH HID v1.00 Gamepad [Wireless Controller] on 10:6f:d9:74:f6:e6
[32235.257507] input: Wireless Controller as /devices/virtual/misc/uhid/0005:054C:05C4.0009/input/input36
[32235.257976] input: Wireless Controller Motion Sensors as /devices/virtual/misc/uhid/0005:054C:05C4.0009/input/input37
[32235.258267] input: Wireless Controller Touchpad as /devices/virtual/misc/uhid/0005:054C:05C4.0009/input/input38
[32235.258752] playstation 0005:054C:05C4.0009: Registered DualShock4 controller hw_version=0x00006404 fw_version=0x00007008
```

RyuzakiKK, 2025-04-14:
One solution that doesn't involve plumbing additional changes to the kernel side could be:

- Add a new property to BlueZ `btd_device` like `sixaxis_trusted`
- During the sixaxis custom USB pairing procedure, set `sixaxis_trusted` to true
- Start the input listening server with `BT_IO_SEC_LOW`, regardless of the ClassicBondedOnly option
- In `hidp_add_connection()` bump the device connection security level to `*_MEDIUM`, i.e. require encryption, for all devices but the once with `sixaxis_trusted`
- Limit the allowed input events (bpf-hid/pre-uhid filtering) to only what a real gamepad is capable of
- Expose the `sixaxis_trusted` property so that clients could act upon that info (e.g. show a warning when said device is connected)

Or a slightly alternative solution could be to start the input listening server with `BT_IO_SEC_LOW` only if we already have a paired device with `sixaxis_trusted`, otherwise sticking to `BT_IO_SEC_MEDIUM` as usual.

@Vudentz do you know if there are possible meaningful differences between starting the listening input server with `BT_IO_SEC_MEDIUM`, compared to starting it with `BT_IO_SEC_LOW` and then bumping the sec level in BlueZ `hidp_add_connection()`? Does this bring any security concerns?

RyuzakiKK, 2025-04-25:
This has been addressed with https://marc.info/?l=linux-bluetooth&m=174551203108913&w=2. Thanks @Vudentz for reviewing it.

@legluondunet your issue is slightly different and is probably better to continue the discussion on the existing https://github.com/bluez/bluez/issues/702.

closed 2025-04-25 by RyuzakiKK