bluez/bluez #1167: WriteValue not triggered by bluez

pavankn, 2025-04-14:
I am trying to establish a connection with a digital wallet on my Android phone using BLE

```
BLE Peripheral is a Linux Box (NXP Imx8mm running yocto mickledore)
BLE Central Client is an Android phone hosting a Digital Wallet
```
I am using dbus-java, bluez-dbus along with a native unix socket.

Data is Sent Successfully from my GattServer to client .. This is OK ,
The client is sending bursts of data ("Digital Identity" information) but my server doesnt receive them 
The spec mandates a "write-without-response" characteristic which my server has registered .

This is what i receive on GattCharacteristic1 with "write-without-response" attribute, i dont see all the fragments, 
i just see a last fragment that had data "00 75 73 14"

hcidump :  Can see all data

```
From BTMON: (Btmon Sees the Packets , The Last Data is 00757314 which is received by me )  .. 
Wireshark is able to reassemble this dump

> ACL Data RX: Handle 3585 flags 0x02 dlen 27                                                                #236 [hci0] 35.621960
> ACL Data RX: Handle 3585 flags 0x01 dlen 27                                                                #237 [hci0] 35.622618
> ACL Data RX: Handle 3585 flags 0x01 dlen 27                                                                #238 [hci0] 35.623279
> ACL Data RX: Handle 3585 flags 0x01 dlen 27                                                                #239 [hci0] 35.623949
> ACL Data RX: Handle 3585 flags 0x01 dlen 27                                                                #240 [hci0] 35.624712
> ACL Data RX: Handle 3585 flags 0x01 dlen 27                                                                #241 [hci0] 35.625339
> ACL Data RX: Handle 3585 flags 0x01 dlen 27                                                                #242 [hci0] 35.625974
> ACL Data RX: Handle 3585 flags 0x01 dlen 27                                                                #243 [hci0] 35.626730
> ACL Data RX: Handle 3585 flags 0x01 dlen 27                                                                #244 [hci0] 35.627355
> ACL Data RX: Handle 3585 flags 0x01 dlen 27                                                                #245 [hci0] 35.632511
> ACL Data RX: Handle 3585 flags 0x01 dlen 27                                                                #246 [hci0] 35.633049
> ACL Data RX: Handle 3585 flags 0x01 dlen 27                                                                #247 [hci0] 35.633734
> ACL Data RX: Handle 3585 flags 0x01 dlen 27                                                                #248 [hci0] 35.634414
> ACL Data RX: Handle 3585 flags 0x01 dlen 27                                                                #249 [hci0] 35.635083
> ACL Data RX: Handle 3585 flags 0x01 dlen 27                                                                #250 [hci0] 35.635754
> ACL Data RX: Handle 3585 flags 0x01 dlen 27                                                                #251 [hci0] 35.636440
> ACL Data RX: Handle 3585 flags 0x01 dlen 27                                                                #252 [hci0] 35.637134
> ACL Data RX: Handle 3585 flags 0x01 dlen 27                                                                #253 [hci0] 35.637809
> ACL Data RX: Handle 3585 flags 0x01 dlen 27                                                                #254 [hci0] 35.638470
> ACL Data RX: Handle 3585 flags 0x01 dlen 6                                                                 #255 [hci0] 35.638970
      ATT: Write Command (0x52) len 514
        Handle: 0x00c7
          Data: 
> ACL Data RX: Handle 3585 flags 0x02 dlen 27                                                                #256 [hci0] 35.639735
> ACL Data RX: Handle 3585 flags 0x01 dlen 27                                                                #257 [hci0] 35.640344
> ACL Data RX: Handle 3585 flags 0x01 dlen 27                                                                #258 [hci0] 35.641001
> ACL Data RX: Handle 3585 flags 0x01 dlen 27                                                                #259 [hci0] 35.641737
> ACL Data RX: Handle 3585 flags 0x01 dlen 27                                                                #260 [hci0] 35.642366
> ACL Data RX: Handle 3585 flags 0x01 dlen 27                                                                #261 [hci0] 35.643034
> ACL Data RX: Handle 3585 flags 0x01 dlen 27                                                                #262 [hci0] 35.643735
> ACL Data RX: Handle 3585 flags 0x01 dlen 27                                                                #263 [hci0] 35.644394
> ACL Data RX: Handle 3585 flags 0x01 dlen 27                                                                #264 [hci0] 35.645058
> ACL Data RX: Handle 3585 flags 0x01 dlen 27                                                                #265 [hci0] 35.645734
> ACL Data RX: Handle 3585 flags 0x01 dlen 22                                                                #266 [hci0] 35.647456
> ACL Data RX: Handle 3585 flags 0x01 dlen 27                                                                #267 [hci0] 35.648035
> ACL Data RX: Handle 3585 flags 0x01 dlen 27                                                                #268 [hci0] 35.648734
> ACL Data RX: Handle 3585 flags 0x01 dlen 27                                                                #269 [hci0] 35.649370
> ACL Data RX: Handle 3585 flags 0x01 dlen 27                                                                #270 [hci0] 35.650060
> ACL Data RX: Handle 3585 flags 0x01 dlen 27                                                                #271 [hci0] 35.650736
> ACL Data RX: Handle 3585 flags 0x01 dlen 27                                                                #272 [hci0] 35.651415
> ACL Data RX: Handle 3585 flags 0x01 dlen 27                                                                #273 [hci0] 35.652086
> ACL Data RX: Handle 3585 flags 0x01 dlen 27                                                                #274 [hci0] 35.652784
> ACL Data RX: Handle 3585 flags 0x01 dlen 11                                                                #275 [hci0] 35.653307
      ATT: Write Command (0x52) len 514
        Handle: 0x00c7
          Data: 
> ACL Data RX: Handle 3585 flags 0x02 dlen 11                                                                #276 [hci0] 35.653852
      ATT: Write Command (0x52) len 6
        Handle: 0x00c7
[... 16 lines cut ...]
Apr 14 01:20:18 nitrogen8mm bluetoothd[539]: bluetoothd[539]: /usr/src/debug/bluez5/5.66-r0/src/shared/gatt-server.c:write_complete_cb() Write Complete: err 0
Apr 14 01:20:18 nitrogen8mm bluetoothd[539]: bluetoothd[539]: /usr/src/debug/bluez5/5.66-r0/src/shared/att.c:can_read_data() (chan 0xaaaac1dce2b0) ATT PDU received: 0x52
Apr 14 01:20:18 nitrogen8mm bluetoothd[539]: [104B blob data]
Apr 14 01:20:18 nitrogen8mm bluetoothd[539]: bluetoothd[539]: /usr/src/debug/bluez5/5.66-r0/src/shared/gatt-server.c:write_cb() Write Cmd - handle: 0x001f
Apr 14 01:20:18 nitrogen8mm bluetoothd[539]: [126B blob data]
Apr 14 01:20:19 nitrogen8mm bluetoothd[539]: bluetoothd[539]: /usr/src/debug/bluez5/5.66-r0/src/gatt-database.c:send_notification_to_device() GATT server sending notification
Apr 14 01:20:19 nitrogen8mm bluetoothd[539]: bluetoothd[539]: /usr/src/debug/bluez5/5.66-r0/src/gatt-database.c:send_notification_to_device() GATT server sending notification
Apr 14 01:20:19 nitrogen8mm bluetoothd[539]: bluetoothd[539]: /usr/src/debug/bluez5/5.66-r0/src/gatt-database.c:send_notification_to_device() GATT server sending notification
Apr 14 01:20:27 nitrogen8mm bluetoothd[539]: bluetoothd[539]: /usr/src/debug/bluez5/5.66-r0/src/shared/att.c:can_read_data() (chan 0xaaaac1dce2b0) ATT PDU received: 0x52
Apr 14 01:20:27 nitrogen8mm bluetoothd[539]: [104B blob data]
Apr 14 01:20:27 nitrogen8mm bluetoothd[539]: bluetoothd[539]: /usr/src/debug/bluez5/5.66-r0/src/shared/gatt-server.c:write_cb() Write Cmd - handle: 0x001d
Apr 14 01:20:27 nitrogen8mm bluetoothd[539]: bluetoothd[539]: /usr/src/debug/bluez5/5.66-r0/src/shared/att.c:can_read_data() (chan 0xaaaac1dce2b0) ATT PDU received: 0x52
Apr 14 01:20:27 nitrogen8mm bluetoothd[539]: [104B blob data]
Apr 14 01:20:27 nitrogen8mm bluetoothd[539]: bluetoothd[539]: /usr/src/debug/bluez5/5.66-r0/src/shared/gatt-server.c:write_cb() Write Cmd - handle: 0x001d
[... cut ...]
Apr 14 01:20:27 nitrogen8mm bluetoothd[539]: bluetoothd[539]: /usr/src/debug/bluez5/5.66-r0/src/shared/att.c:can_read_data() (chan 0xaaaac1dce2b0) ATT PDU received: 0x52
Apr 14 01:20:27 nitrogen8mm bluetoothd[539]: [104B blob data]
Apr 14 01:20:27 nitrogen8mm bluetoothd[539]: bluetoothd[539]: /usr/src/debug/bluez5/5.66-r0/src/shared/gatt-server.c:write_cb() Write Cmd - handle: 0x001d
Apr 14 01:20:27 nitrogen8mm bluetoothd[539]: bluetoothd[539]: /usr/src/debug/bluez5/5.66-r0/src/gatt-database.c:send_write() Pavankn send_write Triggered, Value: , Len: 115
```

Vudentz, 2025-04-14:
Looks like you are using 5.66, might be a good idea to check with the latest, there were some fixes like 8bf3a4a265bb199e21735639c57f1f652e5b8d45 introduced in the meantime.

pavankn, 2025-04-18:
Thank you very much, really appreciate ,  this patch helped me resolve the issue but i upgraded to 5.82  on my OpenEmbedded yocto distribution

closed 2025-04-18 by pavankn