bluez/bluez #1169: [bug] Long advertisement + advertisement monitor causes crash

allard-potma, 2025-04-15:
I see a bluetoothd crash when using an advertisement monitor. A long advertisement packet is the cause. I've seen crashes with the Galaxy Buds 3 Pro, but it is also reproducible with NRF Connect.

### Versions
Linux kernel version: 6.6.48 (Yocto)
BlueZ version: 5.82 also seen at 5.79

### To reproduce
- Start `bluetoothctl`
- Goto the right menu `menu monitor`
- Add an advertisement monitor `add-or-pattern 0 255 1234`
- In the advertiser section of NRF Connect (Android), create a long advertisement, Adv. Extension enabled.
- Enable the advertisement
- Crash happens:

> *** buffer overflow detected ***: terminated
> Program received signal SIGABRT, Aborted.

### The stack trace

> #0  __libc_do_syscall () at ../sysdeps/unix/sysv/linux/arm/libc-do-syscall.S:47
> #1  0xb6e05c58 in __pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6,
>     no_tid=no_tid@entry=0) at pthread_kill.c:43
> #2  0xb6e05c8c in __pthread_kill_internal (signo=6, threadid=<optimized out>) at pthread_kill.c:78
> #3  0xb6dd63ce in __GI_raise (sig=sig@entry=6)
>     at /usr/src/debug/glibc/2.39+git/sysdeps/posix/raise.c:26
> #4  0xb6dc7f5c in __GI_abort () at abort.c:79
> #5  0xb6dfd608 in __libc_message_impl (fmt=0xb6ea1a50 "*** %s ***: terminated\n")
>     at /usr/src/debug/glibc/2.39+git/sysdeps/posix/libc_fatal.c:134
> #6  0xb6e5a430 in __GI___fortify_fail (msg=<optimized out>) at fortify_fail.c:24
> #7  0xb6e59ffe in __GI___chk_fail () at chk_fail.c:28
> #8  0xb6e5a8a2 in __GI___memcpy_chk (dstpp=dstpp@entry=0xbefff7e6, srcpp=<optimized out>,
>     len=<optimized out>, dstlen=dstlen@entry=29) at memcpy_chk.c:27
> #9  0x004944f4 in memcpy (__len=<optimized out>, __src=<optimized out>, __dest=0xbefff7e6)
>     at /usr/include/bits/string_fortified.h:29
> #10 match_manufacturer (data=<optimized out>, user_data=<optimized out>)
>     at /usr/src/debug/bluez5/5.82/src/shared/ad.c:1355
> #11 0x00490ba6 in queue_find (queue=<optimized out>, function=0x494485 <match_manufacturer>,
>     match_data=match_data@entry=0xbefff848) at /usr/src/debug/bluez5/5.82/src/shared/queue.c:230
> #12 0x0049465c in pattern_match (user_data=0xbefff848, data=<optimized out>)
>     at /usr/src/debug/bluez5/5.82/src/shared/ad.c:1451
> #13 pattern_match (data=<optimized out>, user_data=0xbefff848)
>     at /usr/src/debug/bluez5/5.82/src/shared/ad.c:1430
> #14 0x00490b58 in queue_foreach (queue=queue@entry=0x50c9f8, function=0x49462d <pattern_match>,
>     user_data=user_data@entry=0xbefff848) at /usr/src/debug/bluez5/5.82/src/shared/queue.c:207
> #15 0x004955d2 in bt_ad_pattern_match (ad=0x50d218, patterns=warning: could not convert 'queue' from the host encoding (ANSI_X3.4-1968) to UTF-32.
> This normally should not happen, please file a bug report.
> 0x50c9f8)
>     at /usr/src/debug/bluez5/5.82/src/shared/ad.c:1481
> #16 0x00477b72 in adv_match_per_monitor (user_data=0xbefff8ac, data=0x512260)
>     at /usr/src/debug/bluez5/5.82/src/adv_monitor.c:1865
> #17 adv_match_per_monitor (data=0x512260, user_data=0xbefff8ac)
>     at /usr/src/debug/bluez5/5.82/src/adv_monitor.c:1846
> #18 0x00490b58 in queue_foreach (queue=0x502a90, function=0x477b51 <adv_match_per_monitor>,
>     user_data=0xbefff8ac) at /usr/src/debug/bluez5/5.82/src/shared/queue.c:207
> #19 0x00490b58 in queue_foreach (queue=0x505af0, function=0x477709 <adv_match_per_app>,
>     user_data=user_data@entry=0xbefff8ac) at /usr/src/debug/bluez5/5.82/src/shared/queue.c:207
> #20 0x00479366 in btd_adv_monitor_content_filter (manager=<optimized out>, ad=ad@entry=0x2d)
>     at /usr/src/debug/bluez5/5.82/src/adv_monitor.c:1908
> #21 0x00462966 in btd_adapter_device_found (adapter=adapter@entry=0x5042f0,
>     bdaddr=bdaddr@entry=0x501748, bdaddr_type=<optimized out>, rssi=-67 '\275',
>     flags=flags@entry=4, data=data@entry=0x501756 "*\3774\022cgrrw7'aae7'\030rrsw77'\027srrry\220",
>     data_len=data_len@entry=51 '3', monitoring=monitoring@entry=true)
>     at /usr/src/debug/bluez5/5.82/src/adapter.c:7320
> #22 0x0047740a in adv_monitor_device_found_callback (index=0, length=<optimized out>,
>     param=0x501746, user_data=0x505ac8) at /usr/src/debug/bluez5/5.82/src/adv_monitor.c:1610
> #23 0x00490b58 in queue_foreach (queue=0x5019b8, function=0x491c39 <notify_handler>,
>     user_data=user_data@entry=0xbefffa28) at /usr/src/debug/bluez5/5.82/src/shared/queue.c:207
> #24 0x00492740 in process_notify (param=<optimized out>, length=67, index=<optimized out>,
>     event=<optimized out>, mgmt=0x501620) at /usr/src/debug/bluez5/5.82/src/shared/mgmt.c:349
> #25 can_read_data (io=<optimized out>, user_data=0x501620)
>     at /usr/src/debug/bluez5/5.82/src/shared/mgmt.c:409
> #26 0x004b1fec in watch_callback (channel=<optimized out>, cond=<optimized out>,
>     user_data=<optimized out>) at /usr/src/debug/bluez5/5.82/src/shared/io-glib.c:157
> #27 0xb6f29792 in ?? () from /usr/lib/libglib-2.0.so.0
> Backtrace stopped: previous frame identical to this frame (corrupt stack?)

allard-potma, 2025-04-15:
This seems to work:
`---
 src/shared/ad.c | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/src/shared/ad.c b/src/shared/ad.c
index dac381b..e2b3d4d 100644
--- a/src/shared/ad.c
+++ b/src/shared/ad.c
@@ -1334,7 +1334,7 @@ static bool match_manufacturer(const void *data, const void *user_data)
 	const struct bt_ad_manufacturer_data *manufacturer_data = data;
 	const struct pattern_match_info *info = user_data;
 	const struct bt_ad_pattern *pattern;
-	uint8_t all_data[BT_AD_MAX_DATA_LEN];
+	uint8_t all_data[BT_PA_MAX_DATA_LEN];
 
 	if (!manufacturer_data || !info)
 		return false;
-- 
2.34.1`

Vudentz, 2025-04-18:
> This seems to work:
>
> ```
> ---
>  src/shared/ad.c | 2 +-
>  1 file changed, 1 insertion(+), 1 deletion(-)
>
> diff --git a/src/shared/ad.c b/src/shared/ad.c
> index dac381b..e2b3d4d 100644
> --- [...]

I guess it makes more sense to use BT_EA_MAX_DATA_LEN though, since this is about regular advertisement not Periodic ones, those I don't think the monitor can filter.

allard-potma, 2025-04-21:
Aha that makes sense, please take care of it if you have the time.

closed 2025-04-28 by BluezTestBot