bluez/bluez #1172: Advertising HSP-HS (0x1131)?

ferrieux, 2025-04-20:
Hi,

I'm struggling to advertise as an HSP-HS (0x1131). I want to do this, so that a phone can connect to my Linux system, where a program of mine exposing bound RFCOMM and a SCO sockets will take care of the semantics I want to apply. (No I don't want PulseAudio, Alsa, Gstreamer, nor Pipewire for that, I need to do specific stuff).

So basically, it is a matter of advertising the appropriate device class and UUID, right ?

For the device class:
```shell
# daemon started with --noplugin=hostname
# main.conf contains:
  Class = 0x400404 (telephony ; audio/video ; wearable headset)
  DiscoverableTimeout = 0
# hcidump shows the proper class is taken into account:
< HCI Command: Write Class of Device (0x03|0x0024) plen 3 | class 0x400404
```

But then, I cannot seem to find a way to tell `bluetoothctl` to advertise the UUID `0x1131`.
So, instead, I turn to `hcitool`, which does the job:

```shell
$ hcitool -i hci1 cmd 03 52 00 0A 09 42 6C 75 65 5A 61 6C 65 78 02 0A 08 03 03 31 11 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
# which is properly decoded by hcidump as:
< HCI Command: Write Extended Inquiry Response (0x03|0x0052) plen 241 | fec 0x00 | Complete local name: 'BlueZalex' | TX power level: 8 | Complete service classes: 0x1131
```

Then, I tell my phone to pair with my 'BlueZalex' newly advertised device, which starts a pretty long dialogue (below)... but ends up with an SDP rejection:

```shell
2025-04-20 23:36:07.000234 > ... SDP SSA Req: tid 0x0 len 0x13 | pat uuid-16 0x110b (AudioSink) | max 1008 | aid(s) 0x0001 (SrvClassIDList) 0x0004 (ProtocolDescList) 0x0009 (BTProfileDescList) | cont 00
2025-04-20 23:36:07.000988 < ... SDP SSA Rsp: tid 0x0 len 0x5 | count 2 | cont 00
2025-04-20 23:36:07.012247 > ... L2CAP(s): Disconn req: dcid 0x0040 scid 0x0042
2025-04-20 23:36:07.012378 < ... L2CAP(s): Disconn rsp: dcid 0x0040 scid 0x0042
```

I'm surprised to see the phone requesting SSA with UUID 0x110B while the device has been discovered as 0x1131. 

What am I doing wrong ? 

```shell
2025-04-20 23:36:06.470342 > HCI Event: Connect Request (0x04) plen 10 | bdaddr 38:78:62:D0:81:8A class 0x5a020c type ACL
2025-04-20 23:36:06.470441 < HCI Command: Accept Connection Request (0x01|0x0009) plen 7 | bdaddr 38:78:62:D0:81:8A role 0x00 | Role: Central
2025-04-20 23:36:06.470998 > HCI Event: Command Status (0x0f) plen 4 | Accept Connection Request (0x01|0x0009) status 0x00 ncmd 1
2025-04-20 23:36:06.586014 > HCI Event: Role Change (0x12) plen 8 | status 0x00 bdaddr 38:78:62:D0:81:8A role 0x00 | Role: Central
2025-04-20 23:36:06.778216 > HCI Event: Vendor (0xff) plen 4 | 0000: 26 00 01 01                                       &...
2025-04-20 23:36:06.782044 > HCI Event: Connect Complete (0x03) plen 11 | status 0x00 handle 256 bdaddr 38:78:62:D0:81:8A type ACL encrypt 0x00
2025-04-20 23:36:06.782307 < HCI Command: Read Remote Supported Features (0x01|0x001b) plen 2 | handle 256
2025-04-20 23:36:06.782998 > HCI Event: Command Status (0x0f) plen 4 | Read Remote Supported Features (0x01|0x001b) status 0x00 ncmd 1
2025-04-20 23:36:06.786986 > HCI Event: Page Scan Repetition Mode Change (0x20) plen 7 | bdaddr 38:78:62:D0:81:8A mode 1
2025-04-20 23:36:06.791991 > HCI Event: Max Slots Change (0x1b) plen 3 | handle 256 slots 5
2025-04-20 23:36:06.794001 > HCI Event: Read Remote Supported Features (0x0b) plen 11 | status 0x00 handle 256 | Features: 0xff 0xfe 0x8f 0xfe 0xd8 0x3f 0x5b 0x87
2025-04-20 23:36:06.794149 < HCI Command: Read Remote Extended Features (0x01|0x001c) plen 3 | handle 256 page 1
2025-04-20 23:36:06.795019 > HCI Event: Command Status (0x0f) plen 4 | Read Remote Extended Features (0x01|0x001c) status 0x00 ncmd 1
2025-04-20 23:36:06.800216 > ACL data: handle 256 flags 0x02 dlen 10 | L2CAP(s): Info req: type 2
2025-04-20 23:36:06.801043 > HCI Event: Read Remote Extended Features (0x23) plen 13 | status 0x00 handle 256 page 1 max 2 | Features: 0x0b 0x00 0x00 0x00 0x00 0x00 0x00 0x00
2025-04-20 23:36:06.801265 < HCI Command: Remote Name Request (0x01|0x0019) plen 10 | bdaddr 38:78:62:D0:81:8A mode 2 clkoffset 0x0000
2025-04-20 23:36:06.801303 < ACL data: handle 256 flags 0x00 dlen 10 | L2CAP(s): Info req: type 2
2025-04-20 23:36:06.801329 < ACL data: handle 256 flags 0x00 dlen 16 | L2CAP(s): Info rsp: type 2 result 0 | Extended feature mask 0x02b8 | Enhanced Retransmission mode | Streaming mode | FCS Option | Fixed Channels | Unicast Connectless Data Reception
2025-04-20 23:36:06.801987 > HCI Event: Command Status (0x0f) plen 4 | Remote Name Request (0x01|0x0019) status 0x00 ncmd 1
2025-04-20 23:36:06.822472 > HCI Event: Number of Completed Packets (0x13) plen 5 | handle 256 packets 1
2025-04-20 23:36:06.827047 > HCI Event: Remote Name Req Complete (0x07) plen 255 | status 0x00 bdaddr 38:78:62:D0:81:8A name 'Xperia XZ2 Compact'
2025-04-20 23:36:06.827987 > HCI Event: Number of Completed Packets (0x13) plen 5 | handle 256 packets 1
2025-04-20 23:36:06.832260 > ACL data: handle 256 flags 0x02 dlen 16 | L2CAP(s): Info rsp: type 2 result 0 | Extended feature mask 0x00b8 | Enhanced Retransmission mode | Streaming mode | FCS Option | Fixed Channels
2025-04-20 23:36:06.832404 < ACL data: handle 256 flags 0x00 dlen 10 | L2CAP(s): Info req: type 3
2025-04-20 23:36:06.839041 > HCI Event: Number of Completed Packets (0x13) plen 5 | handle 256 packets 1
2025-04-20 23:36:06.844228 > ACL data: handle 256 flags 0x02 dlen 10 | L2CAP(s): Info req: type 3
2025-04-20 23:36:06.844231 > ACL data: handle 256 flags 0x02 dlen 20 | L2CAP(s): Info rsp: type 3 result 0 | Fixed channel list 0x00000082 | L2CAP Signalling Channel
2025-04-20 23:36:06.844271 < ACL data: handle 256 flags 0x00 dlen 20 | L2CAP(s): Info rsp: type 3 result 0 | Fixed channel list 0x00000086 | L2CAP Signalling Channel | L2CAP Connless
2025-04-20 23:36:06.870337 > HCI Event: Number of Completed Packets (0x13) plen 5 | handle 256 packets 1
2025-04-20 23:36:06.924421 > ACL data: handle 256 flags 0x02 dlen 12 | L2CAP(s): Connect req: psm 1 scid 0x0042
2025-04-20 23:36:06.924542 < ACL data: handle 256 flags 0x00 dlen 16 | L2CAP(s): Connect rsp: dcid 0x0040 scid 0x0042 result 0 status 0 | Connection successful
2025-04-20 23:36:06.924567 < ACL data: handle 256 flags 0x00 dlen 23 | L2CAP(s): Config req: dcid 0x0042 flags 0x00 clen 11 | RFC 0x00 (Basic)
2025-04-20 23:36:06.951335 > HCI Event: Number of Completed Packets (0x13) plen 5 | handle 256 packets 1
2025-04-20 23:36:06.952016 > HCI Event: Number of Completed Packets (0x13) plen 5 | handle 256 packets 1
2025-04-20 23:36:06.960269 > ACL data: handle 256 flags 0x02 dlen 16 | L2CAP(s): Config req: dcid 0x0040 flags 0x00 clen 4 | MTU 1024
2025-04-20 23:36:06.960275 > ACL data: handle 256 flags 0x02 dlen 14 | L2CAP(s): Config rsp: scid 0x0040 flags 0x00 result 0 clen 0 | Success
2025-04-20 23:36:06.960388 < ACL data: handle 256 flags 0x00 dlen 18 | L2CAP(s): Config rsp: scid 0x0042 flags 0x00 result 0 clen 4 | MTU 1024
2025-04-20 23:36:06.988432 > HCI Event: Number of Completed Packets (0x13) plen 5 | handle 256 packets 1
2025-04-20 23:36:07.000234 > ACL data: handle 256 flags 0x02 dlen 28 | L2CAP(d): cid 0x0040 len 24 [psm 1] | SDP SSA Req: tid 0x0 len 0x13 | pat uuid-16 0x110b (AudioSink) | max 1008 | aid(s) 0x0001 (SrvClassIDList) 0x0004 (ProtocolDescList) 0x0009 (BTProfileDescList) | cont 00
2025-04-20 23:36:07.000988 < ACL data: handle 256 flags 0x00 dlen 14 | L2CAP(d): cid 0x0042 len 10 [psm 1] | SDP SSA Rsp: tid 0x0 len 0x5 | count 2 | cont 00
2025-04-20 23:36:07.005091 > HCI Event: Number of Completed Packets (0x13) plen 5 | handle 256 packets 1
2025-04-20 23:36:07.012247 > ACL data: handle 256 flags 0x02 dlen 12 | L2CAP(s): Disconn req: dcid 0x0040 scid 0x0042
2025-04-20 23:36:07.012378 < ACL data: handle 256 flags 0x00 dlen 12 | L2CAP(s): Disconn rsp: dcid 0x0040 scid 0x0042
2025-04-20 23:36:07.018034 > HCI Event: Number of Completed Packets (0x13) plen 5 | handle 256 packets 1
2025-04-20 23:36:11.181698 > HCI Event: Disconn Complete (0x05) plen 4 | status 0x00 handle 256 reason 0x13 | Reason: Remote User Terminated Connection

```

pv, 2025-04-21:
Probably your application needs to do the same what Pipewire/Pulseaudio/etc are doing, RegisterProfile etc.

Vudentz, 2025-04-21:
https://github.com/bluez/bluez/blob/master/doc/org.bluez.ProfileManager.rst, that is the actual API documentation of what pulseaudio and pipewire are using, there is no integration with bluetoothctl yet though, which I assume is why you were throwing a tantrum on https://github.com/bluez/bluez/issues/747#issuecomment-2816890705 when you couldn't register an HSP endpoint.

ferrieux, 2025-04-21:
I understand that bluetoothctl isn't ready for the job, so i'm trying to do it at a lower level.  Another reason is that my final target is a bare-bones linux kernel without D-bus, so I'd really like to be able to do it "the old way".

So, I spawned bluetoothd in --compat mode, and used `sdptool add HS` so now the local SDP instance holds (among others):

```shell
Service Name: Headset
Service RecHandle: 0x10007
Service Class ID List:
  "Headset" (0x1108)
  "Generic Audio" (0x1203)
Protocol Descriptor List:
  "L2CAP" (0x0100)
  "RFCOMM" (0x0003)
    Channel: 5
Profile Descriptor List:
  "Headset" (0x1108)
    Version: 0x0100
```

But then, if I try hooking to this from my phone, I still get the strange SSA request for AudioSink (0x110b), which naturally fails as 0x110b is not in the advertised list.

So I tried "cheating" a bit, by adding it to the service class list via `sdptool setseq 0x10007 0x0001 u0x1108  u0x110b`, and also by removing all the default services that were there. So now the whole `sdptool records local` is:

```shell
Service Name: Headset
Service RecHandle: 0x10007
Service Class ID List:
  "Headset" (0x1108)
  "Audio Sink" (0x110b)
Protocol Descriptor List:
  "L2CAP" (0x0100)
  "RFCOMM" (0x0003)
    Channel: 5
Profile Descriptor List:
  "Headset" (0x1108)
    Version: 0x0100
```

This does get me a bit further: now the SAA response is non-empty,

```shell
2025-04-21 22:50:28.712508 < ACL data: handle 256 flags 0x00 dlen 57 | L2CAP(d): cid 0x004c len 53 [psm 1] | SDP SSA Rsp: tid 0x0 len 0x30 | count 45 | record #0 | aid 0x0001 (SrvClassIDList) | < uuid-16 0x1108 (Headset) uuid-16 0x110b (AudioSink) > | aid 0x0004 (ProtocolDescList) | < < uuid-16 0x0100 (L2CAP) > < | uuid-16 0x0003 (RFCOMM) uint 0x5 > > | aid 0x0009 (BTProfileDescList) | < < uuid-16 0x1108 (Headset) uint 0x100 > > | cont 00
```

then things seem to proceed, with some link key negotiation:

```shell
2025-04-21 22:50:28.739894 > HCI Event: Link Key Request (0x17) plen 6 | bdaddr 38:78:62:D0:81:8A
2025-04-21 22:50:28.739938 < HCI Command: Link Key Request Reply (0x01|0x000b) plen 22 | bdaddr 38:78:62:D0:81:8A key 29336484D6AFC6F04FDA2B4AF37DCE1F
2025-04-21 22:50:28.740685 > HCI Event: Command Complete (0x0e) plen 10 | Link Key Request Reply (0x01|0x000b) ncmd 1 | status 0x00 bdaddr 38:78:62:D0:81:8A
2025-04-21 22:50:28.770702 > HCI Event: Encrypt Change (0x08) plen 4 | status 0x00 handle 256 encrypt 0x02
2025-04-21 22:50:28.770792 < HCI Command: Read Encryption Key Size (0x05|0x0008) plen 2 | 0000: 00 01                                             ..
2025-04-21 22:50:28.771671 > HCI Event: Command Complete (0x0e) plen 7 | Read Encryption Key Size (0x05|0x0008) ncmd 1 | 0000: 00 00 01 10                                       ....
```

However, the show stops abruptly when the phone tries an L2CAP to PSM 25:

```shell
2025-04-21 22:50:28.780262 > ACL data: handle 256 flags 0x02 dlen 12 | L2CAP(s): Connect req: psm 25 scid 0x004d
2025-04-21 22:50:28.780323 < ACL data: handle 256 flags 0x00 dlen 16 | L2CAP(s): Connect rsp: dcid 0x0000 scid 0x004d result 2 status 0 | Connection refused - PSM not supported
```

Indeed I have not prepared an AVDPT endpoint, hence the result.

However, I from what I read of the HSP, I only expected to be required to bind an RFCOMM socket (for the AT commands like "RING") and a SCO for the bidirectional audio. So I assume this is all due to my "lying" about being an AudioSink. But why in the first place isn't the phone satisfied with a vanilla 0x1108 ?

Vudentz, 2025-04-21:
> I understand that bluetoothctl isn't ready for the job, so i'm trying to do it at a lower level. Another reason is that my final target is a bare-bones linux kernel without D-bus, so I'd really like to be able to do it "the old way".

I would have stopped right here, there is no "the old way" when it comes to Bluetooth, all the profiles will require bluetoothd in one way or the other since pairing itself is maintained by bluetoothd, while there is nothing stopping you from using linux-bluetooth subsystem and its sockets but you would have to re-implement a lot of stuff which I don't think really feasible in a short term.

ferrieux, 2025-04-21:
Well, regardless of the tooling, hcidump gives a pretty objective view of what is going on. 
So, even though I did not use a ready-made integration like pa's , may I respectfully ask for a hint as to what went wrong in the logs I shared  ?

closed 2025-09-06 by github-actions[bot]