bluez/bluez #1173: HID/HOS/HCE/BV-01-C, PTS does not detect Virtual Cable Unplug

nao244, 2025-04-21:
Hello.
I couldn't pass the HID/HOS/HCE/BV-01-C PTS test, so I tried various things, but I was able to pass it by making the following changes to the code.
I operated it from bluetoothctl during the test.  
I was trying various methods and it just happened to pass, so I don't know if this is the correct way to do it.
Without this change, after being instructed to disconnect and performing the operation, the PTS would determine that there was no Virtual Cable Unplug and would not give a PASS.
For now, I would like to report that v5.82 did not pass HID/HOS/HCE/BV-01-C as is.

~~~bluez/profiles/input/device.c
static int connection_disconnect(struct input_device *idev, uint32_t flags)
{
	int sock;
	if (!is_connected(idev))
		return -ENOTCONN;

	/* Standard HID disconnect
	 * Intr channel must be disconnected before ctrl channel, so only
	 * disconnect intr here, ctrl is disconnected in intr_watch_cb.
	 */
	if (idev->intr_io) {
		sock = g_io_channel_unix_get_fd(idev->intr_io);
		shutdown(sock, SHUT_WR);
	}

        // original
	// if (flags & (1 << HIDP_VIRTUAL_CABLE_UNPLUG)){
	// 	idev->virtual_cable_unplug = true;
		
	// 	DBG("%s: Virtual Cable Unplug", __func__);
	// }

        // modification
        if ((flags & (1 << HIDP_VIRTUAL_CABLE_UNPLUG)) && idev->ctrl_io) {
            hidp_send_ctrl_message(idev,
                HIDP_TRANS_HID_CONTROL | HIDP_CTRL_VIRTUAL_CABLE_UNPLUG,
                NULL, 0);
            DBG("Sent Virtual Cable Unplug");
        }

	if (idev->uhid)
		return uhid_disconnect(idev, false);
	else
		return ioctl_disconnect(idev, flags);
}
~~~

Vudentz, 2025-04-21:
> Hello. I couldn't pass the HID/HOS/HCE/BV-01-C PTS test, so I tried various things, but I was able to pass it by making the following changes to the code. I operated it from bluetoothctl during the [...]

Looks like we haven't implemented the virtual unplug logic for uhid which was handled by the likes of HIDPCONNDEL in hogp.

Vudentz, 2025-05-02:
@nao244 please check with the changes above.

Vudentz, 2025-05-06:
nao244 any chance to test the above changes?

nao244, 2025-05-07:
Sorry, Japan was on a long holiday until yesterday.
I'll try it now.

nao244, 2025-05-08:
@Vudentz When I tested with your patch, HID/HOS/HCE/BV-01-C passed.
Thank you!