bluez/bluez #1175: Airpods connection fails to play audio and disconnects when the airpods initiate the connection

russnes, 2025-04-21:
I'm able to connect my airpods pro gen 2 to my laptop with bluez, GNOME 3 and pipewire/wireplumber and pipewire-pulse, and it works perfectly. However, once I disconnect them by putting the pods in their case, the next time I take the pods out of their case and try to connect the airpods, they will only appear to connect but won't play any audio. This second time around there is no sound transmitted into the airpods, except for the airpods sound notification indicating a connection to a device. In fact, the airpods will appear to be connected, however if I try to adjust the volume through the graphical audio widget, that's when the airpods will be triggered to disconnect, and so an endless loop of trying to connect will go on.

To make it work again, I'll have to first stop trying to connect them to the computer by disabling computer bluetooth and instead connect them to my android phone. After having done that, I can disconnect them from the phone, and initiate a connection from the starting point of the computer, upon which it will finally work again.

It seems that if I manually disconnect the airpods through the audio widget, and then reinitiate a connection from the computer, it works fine. This suggests that the problem arises when the airpods are the ones initiating the connection rather than when the computer initiates it.

**Package versions:**
Distro: Void Linux (rolling release, glibc)
PipeWire: 1.2.7
WirePlumber: 0.5.6
BlueZ: 5.78
DE: KDE Plasma 6.2.0

**bluetoothctl output**
Here are the different outputs of the scenarios in bluetoothctl:

    successful connection initiated from the computer:
	[bluetooth]# [CHG] Device 2C:18:09:DD:08:F4 Connected: yes
	[AirPods Pro]# [NEW] Endpoint /org/bluez/hci0/dev_AB_CD_EF_00_01_02/sep1 
	[AirPods Pro]# [NEW] Endpoint /org/bluez/hci0/dev_AB_CD_EF_00_01_02/sep2 
	[AirPods Pro]# [NEW] Endpoint /org/bluez/hci0/dev_AB_CD_EF_00_01_02/sep3 
	[AirPods Pro]# [NEW] Transport /org/bluez/hci0/dev_AB_CD_EF_00_01_02/sep2/fd20 
	[AirPods Pro]# [CHG] Transport /org/bluez/hci0/dev_AB_CD_EF_00_01_02/sep2/fd20 Delay: 0x05dc (1500)
	[AirPods Pro]# [CHG] Transport /org/bluez/hci0/dev_AB_CD_EF_00_01_02/sep2/fd20 Volume: 0x0051 (81)
	[AirPods Pro]# [CHG] Device 2C:18:09:DD:08:F4 ServicesResolved: yes

	successful disconnect initiated from the computer:
	[AirPods Pro]# [DEL] Transport /org/bluez/hci0/dev_AB_CD_EF_00_01_02/sep2/fd20 
	[AirPods Pro]# [DEL] Endpoint /org/bluez/hci0/dev_AB_CD_EF_00_01_02/sep1 
	[AirPods Pro]# [DEL] Endpoint /org/bluez/hci0/dev_AB_CD_EF_00_01_02/sep2 
	[AirPods Pro]# [DEL] Endpoint /org/bluez/hci0/dev_AB_CD_EF_00_01_02/sep3 
	[AirPods Pro]# [CHG] Device 2C:18:09:DD:08:F4 ServicesResolved: no
	[AirPods Pro]# [CHG] Device 2C:18:09:DD:08:F4 Connected: no

	putting airpods in case so that they automatically disconnect:
	[AirPods Pro]# [DEL] Transport /org/bluez/hci0/dev_AB_CD_EF_00_01_02/sep2/fd21 
	[AirPods Pro]# [DEL] Endpoint /org/bluez/hci0/dev_AB_CD_EF_00_01_02/sep1 
	[AirPods Pro]# [DEL] Endpoint /org/bluez/hci0/dev_AB_CD_EF_00_01_02/sep2 
	[AirPods Pro]# [DEL] Endpoint /org/bluez/hci0/dev_AB_CD_EF_00_01_02/sep3 
	[AirPods Pro]# [CHG] Device 2C:18:09:DD:08:F4 ServicesResolved: no
	[AirPods Pro]# [CHG] Device 2C:18:09:DD:08:F4 Connected: no

	autoconnect after taking airpods out of case:
	[AirPods Pro]# [NEW] Endpoint /org/bluez/hci0/dev_AB_CD_EF_00_01_02/sep1 
	[AirPods Pro]# [NEW] Endpoint /org/bluez/hci0/dev_AB_CD_EF_00_01_02/sep2 
	[AirPods Pro]# [NEW] Endpoint /org/bluez/hci0/dev_AB_CD_EF_00_01_02/sep3 
	[AirPods Pro]# [NEW] Transport /org/bluez/hci0/dev_AB_CD_EF_00_01_02/fd22 
	[AirPods Pro]# [CHG] Transport /org/bluez/hci0/dev_AB_CD_EF_00_01_02/fd22 Delay: 0x05dc (1500)

	waiting for 2 minutes, nothing happens...

	try changing volume, no volume change indicator sound
	[AirPods Pro]# [CHG] Transport /org/bluez/hci0/dev_AB_CD_EF_00_01_02/fd22 State: active
	[AirPods Pro]# [CHG] Transport /org/bluez/hci0/dev_AB_CD_EF_00_01_02/fd22 Volume: 0x0061 (97)

	that will trigger automatic disconnect and attempted reconnect:
	[AirPods Pro]# [CHG] Device 2C:18:09:DD:08:F4 Connected: no
	[bluetooth]# [DEL] Transport /org/bluez/hci0/dev_AB_CD_EF_00_01_02/fd22 
	[bluetooth]# [DEL] Endpoint /org/bluez/hci0/dev_AB_CD_EF_00_01_02/sep1 
	[bluetooth]# [DEL] Endpoint /org/bluez/hci0/dev_AB_CD_EF_00_01_02/sep2 
	[bluetooth]# [DEL] Endpoint /org/bluez/hci0/dev_AB_CD_EF_00_01_02/sep3 
	[bluetooth]# [CHG] Device 2C:18:09:DD:08:F4 Connected: yes
	[AirPods Pro]# [NEW] Endpoint /org/bluez/hci0/dev_AB_CD_EF_00_01_02/sep1 
	[AirPods Pro]# [NEW] Endpoint /org/bluez/hci0/dev_AB_CD_EF_00_01_02/sep2 
	[AirPods Pro]# [NEW] Endpoint /org/bluez/hci0/dev_AB_CD_EF_00_01_02/sep3 
	[AirPods Pro]# [NEW] Transport /org/bluez/hci0/dev_AB_CD_EF_00_01_02/fd23 
	[AirPods Pro]# [CHG] Transport /org/bluez/hci0/dev_AB_CD_EF_00_01_02/fd23 Delay: 0x05dc (1500)

	and nothing happens until I try to change the volume again...

Vudentz, 2025-04-24:
@russnes will probably need the btmon traces to check what is going at HCI level.

pv, 2025-04-24:
Probably this one again: https://gitlab.freedesktop.org/pipewire/pipewire/-/commit/c5e57a062d8dec68e0cba61e88fd4c8b34b501f5 --- if so you'd need to wait for pw 1.2.8

closed 2025-09-06 by github-actions[bot]