bluez/bluez #1178: Connection Delay Issue on Raspberry Pi  (HCI RAW Socket `write()` Hangs)

stoprocent, 2025-04-22:
#### Description

I'm experiencing a persistent issue on the Raspberry Pi 3B+ where Bluetooth connections via USB HCI dongles and built-in BLE hang for ~40 seconds before eventually succeeding. This behavior is completely **random**—at times everything works flawlessly, while at other times the connection gets stuck for an extended period.

Upon investigation, I found that the call which hangs is the `write()` syscall on the **HCI RAW socket** when sending the `HCI_LE_Create_Connection` command. The call blocks for up to 40 seconds before completing.

Interestingly, if I run:

```bash
sudo systemctl restart bluetooth
```

…while the `write()` call is still hanging, it immediately returns and the connection succeeds. This suggests that something internal is holding or blocking the connection request until the restart triggers a release.

#### Additional Context

- This happens **regardless of the USB HCI dongle** used. I’ve tested five different models (including CSR-based ones and other 3rd-party dongles) and all exhibit the same behavior on the Raspberry Pi.
- However, **when I use the exact same dongles on my Ubuntu 22.04 desktop**, the issue **does not occur**. Connections succeed immediately 100% of the time.
- This confirms the dongles themselves are not at fault, and the issue seems specific to the Raspberry Pi environment (or possibly the BlueZ build/config on Raspberry Pi OS).

We have an issue that I will reference https://github.com/stoprocent/noble/issues/12

#### Steps to Reproduce

1. Connect a USB HCI dongle to Raspberry Pi 3B+.
2. Attempt to connect to a BLE peripheral using a library like `noble` (Node.js), which writes `HCI_LE_Create_Connection` via an HCI RAW socket.
3. Sometimes the connection works immediately.
4. Other times, the `write()` syscall blocks for ~40 seconds.
5. While the `write()` call is blocked, run `sudo systemctl restart bluetooth`.
6. The `write()` call returns immediately, and the BLE connection is established successfully.

#### Observations

- **Random behavior**: Immediate success vs 40-second delays.
- **Restarting the Bluetooth service reliably unblocks the hanging `write()` call**.
- The **same userland code and same dongle work perfectly on Ubuntu 22.04 desktop**, indicating it's not a problem with `noble`, the dongles, or the application logic.
- Likely candidates: something in BlueZ's handling of HCI RAW sockets, socket buffer state, or a kernel-level issue specific to Raspberry Pi OS.

#### Questions / Help Needed

- Could this be related to how BlueZ (or the kernel) handles pending HCI commands or socket blocking on Raspberry Pi?
- Are there known issues with HCI RAW socket blocking behavior specific to Raspberry Pi OS or ARM devices?
- Would enabling debug logs or capturing traffic with `btmon` help identify the root cause?
- Any guidance on how to trace what might be blocking the `write()` call?

Vudentz, 2025-04-25:
> #### Description
>
> I'm experiencing a persistent issue on the Raspberry Pi 3B+ where Bluetooth connections via USB HCI dongles and built-in BLE hang for ~40 seconds before eventually succeeding. This [...]

Well you are not suppose to open a raw socket and send command while bluetoothd is running, if you want to take control over HCI then it probably better to use the HCI_CHANNEL_USER:

https://github.com/bluez/bluez/blob/master/doc/hci.rst#socket-address

Im really not sure what people keep doing this workarounds instead of just using the D-Bus interface, these custom setup will never gonna work without pairing, and all GAP handling, that is done by bluetoothd, raw socket is just for testing.

closed 2025-04-25 by Vudentz

stoprocent, 2025-04-27:
There are multiple reasons for not using dbus. It's not great, and we can discuss more if you want. I will try using the `USER` channel to see the difference in the behavior.