bluez/bluez #1193: Unable to pair to Oticon More 2 hearing aids

shermp, 2025-04-24:
When attempting to test ASHA support, I encountered an issue where I was unable to pair with my Oticon More 2 hearing aids. Older versions of BlueZ would work fine, but not `master`, or the latest version available in Fedora 41/42.

The behavior is that during the pairing process, my hearing aids terminate the connection. I have attached the bluetoothd log and a wireshark capture.

[oticon-pairing-bluetoothd.txt](https://github.com/user-attachments/files/19900422/oticon-pairing-bluetoothd.txt)

[wireshark-oticon-pairing.zip](https://github.com/user-attachments/files/19900423/wireshark-oticon-pairing.zip)

Since I was compiling BlueZ for ASHA testing, I was able to `git bisect` to the most likely bad commit. The attached logs were at this commit:

```
sherman@fedora:~/Development/ASHA/bluez$ git bisect bad
5df9521ce4d50ffa48153503bc5156c11e6ed26b is the first bad commit
commit 5df9521ce4d50ffa48153503bc5156c11e6ed26b (HEAD)
Author: Luiz Augusto von Dentz <luiz.von.dentz@intel.com>
Date:   Wed Mar 26 11:52:07 2025 -0400

    device: Elevate bt_att security if bonding is in progress
    
    When bonding is in progress it means Device.Pair has been called thus
    the user indicated the intend to pair/bond from the start so the
    security of bt_att shall be set in order for it to wait until pairing
    process complete before attempting to proceed with GATT discovery and
    any other traffic.
    
    Fixes: https://github.com/bluez/bluez/issues/1125

 src/device.c | 5 +++++
 1 file changed, 5 insertions(+)
```

Vudentz, 2025-04-25:
> When attempting to test ASHA support, I encountered an issue where I was unable to pair with my Oticon More 2 hearing aids. Older versions of BlueZ would work fine, but not `master`, or the latest [...]

I would prefer if you use btmon to collect the logs in btsnoop format.

shermp, 2025-04-25:
Hi @Vudentz 

Here is a `btmon` log, using current `master` branch.

[bluez-oticon-pair-btmon.log](https://github.com/user-attachments/files/19915549/bluez-oticon-pair-btmon.log)

shermp, 2025-04-25:
Just an update, I tried pairing with my TP-Link UB500 USB adapter (Realtek chipset) and my hearing aids paired fine, so the controller seems to play a role in it as well.

shermp, 2025-04-25:
For reference, the controller that is having the pairing issues is the Intel AX210 - the controller that keeps on giving...

ValeZAA, 2025-04-27:
Wait, does TP-Link UB500 USB even support LE audio??

btmgmt info should say 

>wide-band-speech cis-central cis-peripheral

shermp, 2025-04-27:
No, it does not. But my hearing aids also do not support LE audio.

I am testing the in-development ASHA support, which is an older audio streaming solution created by Google for use in Android, It uses L2CAP credit oriented connections. See the [ASHA spec](https://source.android.com/docs/core/connect/bluetooth/asha) for more details about the protocol.

ValeZAA, 2025-04-27:
>No, it does not. But my hearing aids also do not support LE audio.

I googled it and apparently  RTL8852B supports le audio https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/commit/?id=c1a6a1a2030f34c5a895b6e18110c36d380c0669 (IF that is what is used in your dongle.)

BE200 and others I still was not able to make it pair to my LE aduio buds. My LE audio buds appear to have LL privacy too, so they change their MAC addres in the left earbud all the time... That commit you referenced axtually fixes a bug IN LE AUDIO. And it is not needed in your cases, I imagine.

ValeZAA, 2025-04-27:
> I'm more leaning to the idea that the Intel AX210 is just plain buggy. It's not the first time I've had issues with it and BlueZ.

I updated kernel to 6.13.11 and bluez to master and now pactl list cards shows:

a2dp-sink-sbc a2dp-sink-sbc headset-head-unit-cvsd headset-head-unit-msbc  headset-head-unit (LC3-SWB) and bap-duplex!! bap-duplex is LE Audio! It works now, will do more tests

ValeZAA, 2025-04-28:
Just to clarify: this commit is what fixed this issue https://github.com/bluez/bluez/issues/1125  where sometimes internally FIRMWARE of Intel chip was breaking and absent GATT data was not allowing stuff to propely initialise and thus was failing first try pairing. So it just makes it to pair 2nd time only, meh, not such a big issue. Probably the issue is not even that commit, but since byvdefault bluetoothctl does not run scan le, see my newest issue that I will open just now. You can safely revert it... Am I correct here @Vudentz

If you carefully read the issue it is obvious this is not needed for ASHA (probably).

Note for myself: Also no sound in my case can be caused by this thing too. I need to test.

Vudentz, 2025-04-28:
> Just to clarify: this commit is what fixed this issue [#1125](https://github.com/bluez/bluez/issues/1125) where sometimes internally FIRMWARE of Intel chip was breaking and absent GATT data was not allowing stuff to propely initialise and thus was failing first try pairing. So it just makes it to pair 2nd time only, meh, not such a big issue. Probably the issue is not even that commit, but since by default bluetoothctl does not run scan le, see my newest issue that I will open just now. You can safely revert it... Am I correct here [@Vudentz](https://github.com/Vudentz)

Not following, revert what? We want Pair to behave like Connect when it comes to GATT transactions, so Im not reverting that change if that what you want, that said you started by saying it had fixed this issue, if it is fixed please close.

> If you carefully read the issue it is obvious this is not needed for ASHA (probably).
> 
> Note for myself: Also no sound in my case can be caused by this thing too. I need to test.

Although the intention was to workaround controller that get confused with the stack sending data while pairing procedure is in progress, the fact that it align Device.Pair/Connect procedures actually valid by itself.

ValeZAA, 2025-04-28:
>Not following, revert what?

Revert 5df9521ce4d50ffa48153503bc5156c11e6ed26b

ValeZAA, 2025-04-28:
>that said you started by saying it had fixed this issue, if it is fixed please close.

No, it fixed issue #1125

Vudentz, 2025-04-28:
Looks like a connection timeout:

```
> HCI Event: Disconnect Complete (0x05) plen 4                                                                           #392 [hci0] 59.208475
        Status: Success (0x00)
        Handle: 2048 Address: 56:BE:15:91:BF:B1 (Resolvable)
        Reason: Connection Timeout (0x08)
@ MGMT Event: Device Disconnected (0x000c) plen 8                                                                    {0x0001} [hci0] 59.208518
        LE Address: 56:BE:15:91:BF:B1 (Resolvable)
        Reason: Connection timeout (0x01)
@ MGMT Event: Command Complete (0x0001) plen 10                                                                      {0x0001} [hci0] 59.208530
      Pair Device (0x0019) plen 7
        Status: Failed (0x03)
        LE Address: 56:BE:15:91:BF:B1 (Resolvable)

```

Now why would it work if there is GATT traffic that is sort of puzzling, well except if there is something else going on at Link-Layer which we cannot see in the HCI trace.

ValeZAA, 2025-04-29:
Why cannot you disable what 5df9521ce4d50ffa48153503bc5156c11e6ed26b does when connecting to ASHA devices? ASHA was just reverse engineered, right... There may be something missing.

shermp, 2025-04-29:
> Why cannot you disable what [5df9521](https://github.com/bluez/bluez/commit/5df9521ce4d50ffa48153503bc5156c11e6ed26b) does when connecting to ASHA devices? ASHA was just reverse engineered, right... There may be something missing.

A few reasons:

1. BlueZ probably has no idea what type of device it is attempting to pair to at that point. This is not at all specific to ASHA.
2. It's quite likely this is specific to my particular hearing aids and bluetooth controller. Other hearing aids might pair just fine with the AX210
3. Adding device specific hacks is generally something to be avoided if at all possible.