bluez/bluez #1202: Methods to Get Bluetooth MAC Address

bam80, 2025-04-26:
Here is the article describing ["3 Methods to Get Bluetooth MAC Address in Linux"](https://lindevs.com/methods-to-get-bluetooth-mac-address-in-linux):
`hcitool` command, `hciconfig` command and `hci0/identity` file.

AFAIK, the `hci*` tools are deprecated now and absent in base offering of some distributions,
the `/sys/kernel/debug/bluetooth/hci0/identity` file needs `CONFIG_BT_DBUGFS` kernel flag sometimes not enabled also.

What is the actual ways left to gather the MAC Address required?

ValeZAA, 2025-04-27:
btmgmt info

>CONFIG_BT_DBUGFS kernel flag which is not always enabled also.

It mostly always enabled.

bam80, 2025-04-27:
> btmgmt info

Does it need bluez daemon?

pv, 2025-04-27:
If running bluetoothd: get it via DBus.
If no bluetoothd: https://github.com/bluez/bluez/blob/master/doc/mgmt-api.txt#L276

bam80, 2025-04-27:
Thanks.
The `btmgmt` output doesn't look as machine-readable, though:
```
$ btmgmt --index 0 info 
hci0:	Primary controller
	addr 00:11:22:33:44:55 version 6 manufacturer 10 class 0x7c0104
...
```
I could parse it in shell, of course, but it's not very convenient.
So an output formatting option for using in scripts would be welcomed.

Or maybe easier way to get the address exists?
I don't have bluez daemon.

Vudentz, 2025-05-02:
> Thanks. The `btmgmt` output doesn't look as machine-readable, though:
>
> ```
> $ btmgmt --index 0 info
> hci0:	Primary controller
> 	addr 00:11:22:33:44:55 version 6 manufacturer 10 class 0x7c0104
> ...
> ``` [...]

Well if you don't have the daemon then this probably don't belong here, if you wish to use just the kernel then it probably is a linux-bluetooth only thing, otherwise the right way to access the controller/adapter address is:

https://github.com/bluez/bluez/blob/master/doc/org.bluez.Adapter.rst#string-address-readonly

closed 2025-05-06 by Vudentz