bluez/bluez #1207: Post S5 the next iteration of remote wake from connected standby fails

satijav, 2025-04-28:
Test steps:

1. echo enabled >  /sys/bus/usb/devices/3-6/power/wakeup
2. Connect LE mouse
3. Put the DUT to connected standby (sudo rtcwake -m mem-s 60)
4. Remote wake the DUT by clicking/moving the LE mouse -> Works fine at this stage
5. Perform s5 (reboot)
6. echo enabled >  /sys/bus/usb/devices/3-6/power/wakeup
7. Click/move the LE mouse to connect back to DUT
8. Put the DUT to CS
9. Try to wake DUT from CS by clicking/moving the mouse --> FAILS here

Vudentz, 2025-04-29:
```
@ MGMT Event: Controller Suspended (0x002d) plen 1                                                                   {0x0001} [hci0] 53.961732
        Suspend state: Page scanning and/or passive scanning (2)
...
< HCI Command: LE Add Device To Accept List (0x08|0x0011) plen 7                                                       #6624 [hci0] 115.430958
        Address type: Random (0x01)
        Address: CF:AC:A6:79:3F:3A (Static)
> HCI Event: Command Complete (0x0e) plen 4                                                                            #6625 [hci0] 115.431934
      LE Add Device To Accept List (0x08|0x0011) ncmd 1
        Status: Success (0x00)
```

So this tells me that the device is in fact in the Accept List, so except if this is not the right logs ([btmon_s5_snoop_fail.log](https://github.com/user-attachments/files/19939952/btmon_s5_snoop_fail.log)) then something else is going on.

Vudentz, 2025-04-29:
@satijav please try with the fix above

satijav, 2025-04-30:
Thanks luiz.
Above fix is working.

[btmon_s5.log](https://github.com/user-attachments/files/19976645/btmon_s5.log)

closed 2025-04-30 by BluezTestBot