bluez/bluez #1209: Towards dual mode: BAP full duplex breaks A2DP (audio starts artefacting)

ValeZAA, 2025-04-28:
I managed to find out how to force LE audio to work. So it appears my genius Buds 3 Pro use LE Privacy RPA. So I had to activate privacy "device" in main.conf. It still did not work. Then I thought: my device is already paired, what if it accepts the same authorisation in LE audio automatically? So I did not just enable ISO, I enabled dual mode. THEN it still did not work, so I decided to do this: when I boot up my PC I first do scan le in bluetoothctl, and then in GNOME settings (sic) I pair. And I recorded btmon bap negotiation! Yay! Sets keys are there now!

Just to clarify: LE mode only does not work, you need to use dual, because MAC address it likes is the Bluetooth Standard one. And you need to scan le first, which makes no sense 💀💀💀. GATT info maybe? You need to fix that, but that brings me to new issue that has to be fixed first:

Now the issue is that in that mode A2DP SBC does not work properly anymore. Yes, it just completly breaks the connection all the time and sounds starts constant artefacting every half a second. Any idea how to fix that? You probably ask too much from the dual connected device even when I switch to A2DP, and thus LE audio connected is not needed to be constantly pinged. You see this needs to be fixed before defaulting to scan le or people will have their A2DP broken.

Kernel is 6.13.11, bluez is master (debian bluez testing is broken anyway right now since a week ago). I also have latest firmwares for everything, BE200 from 3 days ago, and I have new driver from Intel backport tree that makes my Intel Killer work properly with Wifi 7 and recognises it as BE200 and not BE202 (you probably want to use it too in your tests, so git checkout 82d656e4fa337f124dbdca4885d2b81beb338fdb && make && make install) https://git.kernel.org/pub/scm/linux/kernel/git/iwlwifi/backport-iwlwifi.git/log/?h=config-cleanup

Some proof:

pactl list cards now shows

bap-duplex.

I also see this:

[bluetoothctl]> info C4:77:64:6C:84:8E

Sets.Key: /org/bluez/hci1/set_6c6477c424ad456477c48e846c6477c4
Sets.Value.Rank: 0x01 (1)
PreferredBearer: last-seen

What is sets.key? Is the thing 6c6477c424ad456477c48e846c6477c4 the key? Or there is some file for it? Also it still does not do default-agent stuff when it allows to connect 2nd airbud...

I also see this LE audio QOS data.

ValeZAA, 2025-04-28:
[secretstff.zip](https://github.com/user-attachments/files/19941184/secretstff.zip)

btmon Log is here, 25 MB :(

ValeZAA, 2025-04-29:
Actually A2DP appears to work fine when switching from BAP if you use pavucontrol and not gnome controls

LMAO

pv, 2025-05-03:
As for BAP
```
> ACL Data RX: Handle 2048 flags 0x02 dlen 15                                                                                                                                                                                                               #22531 [hci1] 562.6>
      ATT: Handle Value Notification (0x1b) len 10
        Handle: 0x0043 Type: ASE Control Point (0x2bc6)
          Data[8]: 0302030000010e0b
            Opcode: Enable (0x03)
            Number of ASE(s): 2
            ASE: #0
            ASE ID: 0x03
            ASE Response Code: Success (0x00)
            ASE Response Reason: None (0x00)
            ASE: #1
            ASE ID: 0x01
            ASE Response Code: Unspecified Error (0x0e)
            ASE Response Reason: Reserved (0x0b)
```
Probably the device does not support duplex 48kHz out / 16 kHz in. Most devices only seem to have 32/32kHz in/out. The specification doesn't have a way to determine which duplex combinations are supported, and fallback to different configuration on "Unspecified Error" is not implemented in bluez/pw so far.

You can do something like
```
# ~/.config/wireplumber/wireplumber.conf.d/bap-sink.conf
monitor.bluez.properties = { bluez5.roles = [ bap_source a2dp_sink a2dp_source hfp_hf hfp_ag ] }
```

ValeZAA, 2025-05-03:
Nope, there is still no sound even with bap-sink.conf... New btmon log, see the very end... Also I am pretty sure my Buds3 Pro is state of the art and has 48khz LE audio on windows (well, not with BE200, since Intel does not enable LE driver for BE200 devices, weird, it does pair though on windows very well if you install that driver manually, animation to connect second earbud and all, no sound though). What can be a problem is 2 output channels per earbud...

Oh and also how do I make sure both earbuds are connected? How do I see that coordinated set works?

Maybe now it is time to make controlller le mode only? Or dual would work anyway, here, since you can see LE start encryption and Encryption Change do work... Only rarely, BTW, most of the time unless you scan le it will not do LE start encryption.

Here is what is printed in bluetoothctl when I manage to activate bap:

```
HG] Device 7C:C9:7B:2D:5E:20 RSSI: 0xffffffcb (-53)
[NEW] Transport /org/bluez/hci0/dev_C4_77_64_6C_84_8E/sep1/fd2 
[CHG] Transport /org/bluez/hci0/dev_C4_77_64_6C_84_8E/sep1/fd2 Delay: 0x0960 (2400)
[CHG] Transport /org/bluez/hci0/dev_C4_77_64_6C_84_8E/sep1/fd2 Volume: 0x0044 (68)
[DEL] Device E3:65:0A:B7:BB:55 E3-65-0A-B7-BB-55
[CHG] Device C4:77:64:6C:84:8E ServicesResolved: yes
[CHG] Device C4:77:64:6C:84:8E Modalias: bluetooth:v0075pA013d0100
[NEW] Endpoint /org/bluez/hci0/dev_C4_77_64_6C_84_8E/pac_source0 
[NEW] Endpoint /org/bluez/hci0/dev_C4_77_64_6C_84_8E/pac_source1 
[NEW] Endpoint /org/bluez/hci0/dev_C4_77_64_6C_84_8E/pac_sink0 
[NEW] Endpoint /org/bluez/hci0/dev_C4_77_64_6C_84_8E/pac_sink1 
[NEW] Endpoint /org/bluez/hci0/dev_C4_77_64_6C_84_8E/pac_sink2 
Endpoint: SelectProperties
	Properties.Endpoint: /org/bluez/hci0/dev_C4_77_64_6C_84_8E/pac_source1
	Properties.Capabilities:
  03 01 a4 00 02 02 03 02 03 01 05 04 1a 00 9b 00  ................
  02 05 01 00 03 01 a4 00 02 02 03 02 03 01 05 04  ................
  1a 00 9b 00 02 05 01                             .......         
	Properties.Locations: 0x00000002 (2)
	Properties.ChannelAllocation: 0x00000002 (2)
	Properties.Metadata:
  03 01 06 00 00 03 01 06 00                       .........       
selecting 16_2_1...
	Configuration.#0: len 0x02 type 0x01
	Configuration.Sampling Frequency: 16 Khz (0x03)
	Configuration.#1: len 0x02 type 0x02
	Configuration.Frame Duration: 10 ms (0x01)
	Configuration.#2: len 0x03 type 0x04
	Configuration.Frame Length: 40 (0x0028)
	Configuration.#3: len 0x05 type 0x03
	Configuration.Location: 0x00000002
	Configuration.Location: Front Right (0x00000002)
Framing 0x00
PresentationDelay 40000
TargetLatency 0x02
Interval 10000
PHY 0x02
SDU 40
Retransmissions 2
Latency 10
[NEW] Transport /org/bluez/hci0/dev_C4_77_64_6C_84_8E/pac_source0/fd3 
Endpoint: SetConfiguration
	Transport /org/bluez/hci0/dev_C4_77_64_6C_84_8E/pac_source0/fd3
	Properties.Device: /org/bluez/hci0/dev_C4_77_64_6C_84_8E
Auto Accepting...
[NEW] Transport /org/bluez/hci0/dev_C4_77_64_6C_84_8E/pac_sink0/fd4 
[CHG] Transport /org/bluez/hci0/dev_C4_77_64_6C_84_8E/pac_source0/fd3 Links: /org/bluez/hci0/dev_C4_77_64_6C_84_8E/pac_sink0/fd4
[CHG] Transport /org/bluez/hci0/dev_C4_77_64_6C_84_8E/pac_sink0/fd4 Links: /org/bluez/hci0/dev_C4_77_64_6C_84_8E/pac_source0/fd3
[NEW] Transport /org/bluez/hci0/dev_C4_77_64_6C_84_8E/pac_sink1/fd5 
Endpoint: SetConfiguration
	Transport /org/bluez/hci0/dev_C4_77_64_6C_84_8E/pac_sink1/fd5
	Properties.Device: /org/bluez/hci0/dev_C4_77_64_6C_84_8E
Auto Accepting...
[NEW] Primary Service (Handle 0x0001)
	/org/bluez/hci0/dev_C4_77_64_6C_84_8E/service0001
	00001800-0000-1000-8000-00805f9b34fb
	Generic Access Profile
[NEW] Characteristic (Handle 0x0002)
	/org/bluez/hci0/dev_C4_77_64_6C_84_8E/service0001/char0002
	00002a00-0000-1000-8000-00805f9b34fb
	Device Name
[... 310 lines cut ...]
	/org/bluez/hci0/dev_C4_77_64_6C_84_8E/service007d/char007e
	00008fca-0000-1000-8000-00805f9b34fb
	Unknown
[NEW] Characteristic (Handle 0x0080)
	/org/bluez/hci0/dev_C4_77_64_6C_84_8E/service007d/char0080
	00008fcb-0000-1000-8000-00805f9b34fb
	Unknown
[NEW] Descriptor (Handle 0x0082)
	/org/bluez/hci0/dev_C4_77_64_6C_84_8E/service007d/char0080/desc0082
	00002902-0000-1000-8000-00805f9b34fb
	Client Characteristic Configuration
[NEW] Characteristic (Handle 0x0083)
	/org/bluez/hci0/dev_C4_77_64_6C_84_8E/service007d/char0083
	00008fcc-0000-1000-8000-00805f9b34fb
	Unknown
[NEW] Primary Service (Handle 0x0085)
	/org/bluez/hci0/dev_C4_77_64_6C_84_8E/service0085
	00001853-0000-1000-8000-00805f9b34fb
	Common Audio
[CHG] Transport /org/bluez/hci0/dev_C4_77_64_6C_84_8E/sep1/fd2 Volume: 0x0066 (102)
[DEL] Device 77:2A:52:85:77:E1 77-2A-52-85-77-E1
hci0 7C:C9:7B:2D:5E:20 type LE Random connect failed (status 0x08, Timeout)
```

[btmonnew.txt](https://github.com/user-attachments/files/20025174/btmonnew.txt)

ValeZAA, 2025-05-03:
Okay, I suppose I will have to compile pipewire. Android uses 32k in, 32k out, but I will try 16k in, because that is what said by Samsung in SWB option https://github.com/bluez/bluez/issues/713#issuecomment-1896643401

And btw, here is:

```
[Валерий's Buds3 Pro]> transport.show /org/bluez/hci0/dev_C4_77_64_6C_84_8E/pac_source0/fd3
Transport /org/bluez/hci0/dev_C4_77_64_6C_84_8E/pac_source0/fd3
	UUID: Sink PAC                  (00002bc9-0000-1000-8000-00805f9b34fb)
	Codec: 0x06 (6)
	Configuration.#0: len 0x02 type 0x01
	Configuration.Sampling Frequency: 16 Khz (0x03)
	Configuration.#1: len 0x02 type 0x02
	Configuration.Frame Duration: 10 ms (0x01)
	Configuration.#2: len 0x05 type 0x03
	Configuration.Location: 0x00000002
	Configuration.Location: Front Right (0x00000002)
	Configuration.#3: len 0x03 type 0x04
	Configuration.Frame Length: 40 (0x0028)
	Configuration.#4: len 0x02 type 0x05
	Configuration.Frame Blocks per SDU: 1 (0x01)
	Device: /org/bluez/hci0/dev_C4_77_64_6C_84_8E
	State: idle
	Volume: 0x0088 (136)
	Endpoint: /org/bluez/hci0/dev_C4_77_64_6C_84_8E/pac_source0
	QoS.CIG: 0x00 (0)
	QoS.CIS: 0x00 (0)
	QoS.Framing: 0x00 (0)
	QoS.PresentationDelay: 0x00009c40 (40000)
	QoS.Interval: 0x00002710 (10000)
	QoS.Latency: 0x000a (10)
	QoS.SDU: 0x0028 (40)
	QoS.PHY: 0x02 (2)
	QoS.Retransmissions: 0x02 (2)
	Location: 0x00000002 (2)
	Links: /org/bluez/hci0/dev_C4_77_64_6C_84_8E/pac_sink0/f
```

ValeZAA, 2025-05-05:
>but I will try 16k in

Apparently [that is the default](https://gitlab.freedesktop.org/pipewire/pipewire/-/commit/89077d16ec5902e528fe9a8e8b37b8598a51b781), because pv's buds 2 pro mic do not work besides 16 kHz... @pv Did you try newer firmware? Maybe they fixed 32kHz mic, since that is the default in Android linux kernel? Again, on my Buds 3 pro LC3-SWB works just fine (on Debian testing Linux) and that is 32k mono stream. Also, I will check whether mic stream in LE audio is mono on Buds 3 pro, as defined by new revision of specification. https://gitlab.freedesktop.org/pvir/pipewire/-/commit/0b5f71652668e7c4f9304c5a190fd84136f23117

Okay, I will revert this commit in part https://gitlab.freedesktop.org/pvir/pipewire/-/commit/c3f1fcca4b1b4b223669cc1ba18bfb62f416ea54 and hope this will make a difference. I will also remove bap-sink.conf hack, of course.