bluez/bluez #1226: BlueZ can miss a disconnect event if it happens before the Event Mask is set

RyuzakiKK, 2025-04-29:
## Steps to reproduce

- Enable Bluetooth wake from suspend
- Connect a Nintendo SNES gamepad
- Suspend the PC
- Wake the gamepad

## Expected result

The PC should resume, and the gamepad should either successfully connect or fail to connect and being reported as "disconnected"

## Actual result

With a variable repro rate, the gamepad fails to connect while BlueZ reports it as `Connected: yes` indefinitely.

Trying to manually disconnect it results in `Failed to disconnect device: Disconnected (0x0e)`

## Issue details

Apparently this device can send a disconnect event after only 0.2 seconds of establishing a connection.

As you can see in the attached btmon log, it seems that the L2CAP disconnect event happens before the `Set Event Mask`, and this means that BlueZ doesn't get the expected HCI `Disconnect Complete` event. This timing issue could be related to the fact that the PC was still busy resuming from suspend.

After the timeout is reached, an HCI `Disconnect` command gets sent, but the controller already removed that device and replies with `Unknown Connection Identifier (0x02)`.

This leaves BlueZ status out of sync with the controller. BlueZ keeps thinking that the device is connected when it actually isn't.

```
@ MGMT Event: Device Connected (0x000b) plen 30       {0x0001} [hci0] 35.076960
        BR/EDR Address: 78:20:A5:4A:DF:28 (Nintendo Co.,Ltd)
        Flags: 0x00000000
        Data length: 17
        Name (complete): SNES Controller
[...]
> ACL Data RX: Handle 3 flags 0x02 dlen 12                #613 [hci0] 35.334462
      L2CAP: Disconnection Request (0x06) ident 5 len 4
        Destination CID: 65
        Source CID: 72
< ACL Data TX: Handle 3 flags 0x00 dlen 12                #614 [hci0] 35.334515
      L2CAP: Disconnection Response (0x07) ident 5 len 4
        Destination CID: 65
        Source CID: 72
> ACL Data RX: Handle 3 flags 0x02 dlen 12                #616 [hci0] 35.343173
      L2CAP: Disconnection Request (0x06) ident 6 len 4
        Destination CID: 64
        Source CID: 71
< ACL Data TX: Handle 3 flags 0x00 dlen 12                #617 [hci0] 35.343195
      L2CAP: Disconnection Response (0x07) ident 6 len 4
        Destination CID: 64
        Source CID: 71
< HCI Command: Set Event Mask (0x03|0x0001) plen 8        #619 [hci0] 35.514430
        Mask: 0x3dbff807fffbffff
          Inquiry Complete
          Inquiry Result
          Connection Complete
          Connection Request
          Disconnection Complete
          Authentication Complete
[...]
< HCI Command: Disconnect (0x01|0x0006) plen 3            #643 [hci0] 39.501645
        Handle: 3 Address: 78:20:A5:4A:DF:28 (Nintendo Co.,Ltd)
        Reason: Remote User Terminated Connection (0x13)
> HCI Event: Command Status (0x0f) plen 4                 #644 [hci0] 39.502155
      Disconnect (0x01|0x0006) ncmd 1
        Status: Unknown Connection Identifier (0x02)
```

RyuzakiKK, 2025-04-29:
I'm wondering if the status `Unknown Connection Identifier (0x02)` should be forwarded to BlueZ to correctly clean up that device as if the disconnect command was successful.

Vudentz, 2025-04-29:
> I'm wondering if the status `Unknown Connection Identifier (0x02)` should be forwarded to BlueZ to correctly clean up that device as if the disconnect command was successful.

Yeah, most likely we need to threat it as if it were successful since the controller don't know about its handle anymore there is no reason for the host to keep it, do you care to send a patch addressing it?

Vudentz, 2025-05-06:
> Thanks, I'll try to make a patch for it.

Any update on this?

RyuzakiKK, 2025-05-07:
Sorry, I was going to write an update before the end of the week.

So was I only had time to test it with this trivial change:
```diff
---
 net/bluetooth/hci_event.c | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/net/bluetooth/hci_event.c b/net/bluetooth/hci_event.c
index 1d8616f2e740..03636fa50cb6 100644
--- a/net/bluetooth/hci_event.c
+++ b/net/bluetooth/hci_event.c
@@ -2726,7 +2726,7 @@ static void hci_cs_disconnect(struct hci_dev *hdev, u8 status)
 	if (!conn)
 		goto unlock;
 
-	if (status) {
+	if (status && status != HCI_ERROR_UNKNOWN_CONN_ID) {
 		mgmt_disconnect_failed(hdev, &conn->dst, conn->type,
 				       conn->dst_type, status);
 
-- 
2.49.0
```

And I could see the device being correctly removed from the list of "connected" devices. In the error free case we also expected a followup event that triggers `hci_disconn_complete_evt()`, but the two functions seems quite similar anyway.

The only problem is that the gamepad isn't able to reconnect. I'm assuming this is happening because we are not cleaning up correctly the `profiles/input/device.c` instance, effectively missing the interrupts there to trigger the disconnect and let it trigger `input_device_enter_reconnect_mode()`.

Unfortunately I'm now out of office, and I'll return the second week of next month. @Vudentz if you want to work on it, that would be great. I can still help you with the actual testing, in case you don't have a device affected by this issue. It may take some time for me to reply, but I should be able to do it.