bluez/bluez #1233: Extremely complicated to cross-compile - and non-trivial versioning

danergo, 2025-04-30:
We are building on a trimmed version of Ubuntu with crosstool-ng (in Docker), purely command-line, targeting a RPi which is also purely command-line only.

Downloading BlueZ 5.78 version from kernel.org: "http://www.kernel.org/pub/linux/bluetooth/bluez-5.78.tar.xz":

Facts:

1. We **do** have `./configure`
2. We **can** successully call it.

Cloning BlueZ 5.78 branch from here

Facts:

1. We **do NOT** have `./configure` ?
2. We do have bootstrap-configure.sh, but we **can NOT** successfully call it, because it demands various things which doesn't exists (and we don't want them on our build image). (for example: `configure: error: elfutils support is required`)

Why there are two versions with the very same version number?

Vudentz, 2025-04-30:
> We are building on a trimmed version of Ubuntu with crosstool-ng (in Docker), purely command-line, targeting a RPi which is also purely command-line only.
>
> Downloading BlueZ 5.78 version from [...]

What 2 version are you talking about? Or you are refering to https://github.com/bluez/bluez/releases/tag/5.78? I think the github version is not the same as to the kernel.org since the later is probably the result of make dist but it seems the github doesn't do that it seems, it just pack the current tree, perhaps I can start adding the tarball manually going forward.

danergo, 2025-05-06:
`./bootstrap` adds `configure`. :)

closed 2025-05-06 by Vudentz