bluez/bluez #1240: Not builded on Arch Linux

Boria138, 2025-05-04:
I'm trying to build bluez 5.82 on Arch Linux but I get this error

 src/shared/shell.c: In function 'rl_cleanup':
src/shared/shell.c:1424:9: error: too many arguments to function 'rl_message'; expected 0, have 1
 1424 |         rl_message("");
      |         ^~~~~~~~~~ ~~
In file included from src/shared/shell.c:29:
/usr/include/readline/readline.h:410:12: note: declared here
  410 | extern int rl_message ();
      |            ^~~~~~~~~~
make[1]: *** [Makefile:8706: src/shared/libshared_mainloop_la-shell.lo] Error 1

Vudentz, 2025-05-06:
@Boria138 weird, this seems to be a different error than fc42 which was fixed by the patch bellow:

```
commit e501c8ab3023155700bc752d318b1ba3480311da
Author: Luiz Augusto von Dentz <luiz.von.dentz@intel.com>
Date:   Thu Apr 17 15:58:19 2025 -0400

    shared/shell: Fix build errors in fc42
    
    This fixes the following errors:
    
    src/shared/shell.c: In function 'rl_cleanup':
    src/shared/shell.c:1429:20: error: zero-length gnu_printf format string [-Werror=format-zero-length]
     1429 |         rl_message("");
          |                    ^~

```

Boria138, 2025-05-06:
> @Boria138 weird, this seems to be a different error than fc42 which was fixed by the patch bellow:
>
> ```
> commit e501c8ab3023155700bc752d318b1ba3480311da
> Author: Luiz Augusto von Dentz [...]

I tried to build version 5.79, which was definitely built before and it also did not build, maybe it's not even in the bluez code itself, but in an update of the compiler or some other component in the system

404NotFoundJS, 2025-05-13:
readline.h:
```c
#if defined (USE_VARARGS) && defined (PREFER_STDARG)
extern int rl_message (const char *, ...)  __attribute__((__format__ (printf, 1, 2)));
#else
extern int rl_message ();
#endif
```
Seem like bluez intends to use variadic `rl_message` but does not define `USE_VARARGS` and `PREFER_STDARG`

substring, 2025-05-23:
Looks like this is a GCC15 problem, it builds fine with GCC14

Boria138, 2025-06-06:
> Also present on Gentoo [LINK](https://gitweb.gentoo.org/repo/gentoo.git/commit/?id=6b9a8501590c4b1fa99ff64229a50618ec060db1). They have a patch for readline, but I fixed it with a dirty fix by removing the lines highlighed by [@404NotFoundJS](https://github.com/404NotFoundJS), with the exception of: `extern int rl_message (const char *, ...) __attribute__((__format__ (printf, 1, 2)));`

As it turns out, the same patch as in Gentoo was added to Arch and bluez 5.83 successfully built, but I'll probably leave the bug open.

Xeonacid, 2025-07-08:
Refresh