bluez/bluez #1247: Won't connect to Q16 earbuds:

linuxguy123, 2025-05-05:
Failure to connect to Q16 earbuds via bluetoothctl.   The earbuds pair fine via the Bluetooth GUI in Fedora 41/KDE.  

```
[bluetoothctl]> devices
Device 90:10:18:8C:0A:24 Q16
Device F2:01:04:EC:C9:13 HW706-0030652
Device 20:16:04:00:0F:94 LUXA2 Lavi S

[bluetoothctl]> info 90:10:18:8C:0A:24
Device 90:10:18:8C:0A:24 (public)
        Name: Q16
        Alias: Q16
        Class: 0x00240404 (2360324)
        Icon: audio-headset
        Paired: yes
        Bonded: yes
        Trusted: yes
        Blocked: no
        Connected: no
        LegacyPairing: no
        UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        Modalias: bluetooth:v05D6p000Ad0240

[bluetoothctl]> connect 90:10:18:8C:0A:24 
Attempting to connect to 90:10:18:8C:0A:24
Failed to connect: org.bluez.Error.Failed br-connection-page-timeout

[bluetoothctl]> connect 90:10:18:8C:0A:24    200~0000110b-0000-1000-8000-00805f9b34fb
Attempting to connect profile "200~0000110b-0000-1000-8000-00805f9b34fb" to 90:10:18:8C:0A:24
Failed to connect: org.bluez.Error.InvalidArguments br-connection-invalid-argument

[bluetoothctl]> connect 90:10:18:8C:0A:24  0000110b-0000-1000-8000-00805f9b34fb
Attempting to connect profile "Audio Sink" to 90:10:18:8C:0A:24
Failed to connect: org.bluez.Error.Failed br-connection-page-timeout

[bluetoothctl]> connect 90:10:18:8C:0A:24  0000110e-0000-1000-8000-00805f9b34fb
Attempting to connect profile "A/V Remote Control" to 90:10:18:8C:0A:24
Failed to connect: org.bluez.Error.Failed br-connection-create-socket

[bluetoothctl]> connect 90:10:18:8C:0A:24  00001200-0000-1000-8000-00805f9b34fb
Attempting to connect profile "PnP Information" to 90:10:18:8C:0A:24
Failed to connect: org.bluez.Error.Failed br-connection-page-timeout

[bluetoothctl]> connect 90:10:18:8C:0A:24  0000110b-0000-1000-8000-00805f9b34fb
Attempting to connect profile "Audio Sink" to 90:10:18:8C:0A:24
Failed to connect: org.bluez.Error.Failed br-connection-page-timeout
```

Vudentz, 2025-05-06:
@linuxguy123 please add the btmon -w logs, from the looks of it is page timeout which normally is due to the remote device being powered off or something like it, but you are saying that with other clients you managed to connect to it?

linuxguy123, 2025-05-07:
Solved.   Somehow the WiFi (also Bluetooth) antenna was not connected to my PC.  I connected the antenna and restarted Bluetooth and now everything works.

closed 2025-05-07 by linuxguy123