bluez/bluez #1251: After recent commits info after bluetoothctl causes instant segfault

ValeZAA, 2025-05-06:
root@kali:/opt/SourceCode/bluez# /usr/bin/bluetoothctl
Agent registered
AdvertisementMonitor path registered
[bluetoothctl]> info
Missing device address argument
Segmentation fault

Vudentz, 2025-05-07:
[bluetoothctl]> info
Missing device address argument
DeviceSet (null) not available

Doesn't crash for me

ValeZAA, 2025-05-07:
You need Privacy = device 
Experimental = true
KernelExperimental = 6fbaf188-05e0-496a-9885-d6ddfdb4e03e

It does not crash on default main.conf

```
root@kali:~# gdb bluetoothctl 
GNU gdb (Debian 16.3-1) 16.3
Copyright (C) 2024 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from bluetoothctl...
(gdb) r
Starting program: /usr/bin/bluetoothctl 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Agent registered
hci0 new_settings: powered bondable ssp br/edr le secure-conn privacy wide-band-speech cis-central cis-peripheral 
[CHG] Controller E8:BF:B8:E4:A5:6A Pairable: yes
AdvertisementMonitor path registered
[bluetoothctl]> info
Missing device address argument

Program received signal SIGSEGV, Segmentation fault.
__strcmp_avx2 () at ../sysdeps/x86_64/multiarch/strcmp-avx2.S:287
warning: 287	../sysdeps/x86_64/multiarch/strcmp-avx2.S: No such file or directory
(gdb) r
```

ValeZAA, 2025-05-16:
Experimental = true
KernelExperimental = 6fbaf188-05e0-496a-9885-d6ddfdb4e03e

are enough to reproduce. Still happens on latest commit @Vudentz

ValeZAA, 2025-05-16:
yes, your patch fixes it.

ValeZAA, 2025-05-18:
>issue info command without giving an address, or first connecting a device?

@Vudentz it is not supposed to cause segfault because you just typed info there

closed 2025-05-19 by BluezTestBot