bluez/bluez #1259: Keyboard fails to reconnect after sleep/disconnect with `org.bluez.Error.Failed br-connection-create-socket`

39george, 2025-05-12:
Logitech K380 fails to reconnect after sleep/disconnect with `org.bluez.Error.Failed br-connection-create-socket`

### **Environment**:  
- **OS**: Arch Linux (latest)  
- **BlueZ version**: `5.82` 
- **Kernel**: `6.14.6-arch1-1`
- **Device**: Logitech K380 (Bluetooth keyboard)  

### **Issue Description**:  
The keyboard pairs successfully and works initially, but after disabling Bluetooth or turning the keyboard off/on, it fails to reconnect with the following error:  

```
org.bluez.Error.Failed br-connection-create-socket
```

**Observed behavior** (`btmon` output shows disconnection after encryption):  
```plaintext
> HCI Event: Encryption Change (0x08) plen 4               #53 [hci0] 23.788027
        Status: Success (0x00)
        Handle: 3
        Encryption: Enabled with E0 (0x01)
...
> HCI Event: Disconnect Complete (0x05) plen 4             #60 [hci0] 23.951307
        Reason: Remote User Terminated Connection (0x13)
```

### **Troubleshooting Attempted**:  
- Checked `rfkill`, and `bluetooth.service` → all active.  

### **Additional Notes**:  
- Works fine on Windows/macOS (firmware issue unlikely).

### **Request**:  
Why BlueZ fails to re-establish the BR/EDR socket connection after disconnection. Is this a kernel (Bluetooth stack) or BlueZ (profile handling) issue?

FakeMichau, 2025-05-13:
I'm seeing the same behavior with a dualsense controller on Arch 6.14.6.arch1-1. The success rate of a reconnect is very low but can happen. 

On linux-lts 6.12.28-1 the success rate is way higher, around 75% I'd say. But btmon shows a different error in that case so maybe it's unrelated to the issue on 6.14.
```
Result: Connection refused - security block (0x0003)
Status: No further information available (0x0000)
```

Downgrading bluez to 5.79 doesn't help for me

Vudentz, 2025-05-15:
@39george try collecting the HCI trace with btmon and attach it here.

39george, 2025-05-15:
@Vudentz, Here it is. Recorded starting from pairing, then successfully connecting. After that, I disabled the keyboard and re-enabled it, but now it fails to reconnect (it briefly connects but immediately disconnects).
[btmon_log.txt](https://github.com/user-attachments/files/20233287/btmon_log.txt)

Vudentz, 2025-05-15:
> [@Vudentz](https://github.com/Vudentz), Here it is. Recorded starting from pairing, then successfully connecting. After that, I disabled the keyboard and re-enabled it, but now it fails to reconnect [...]

It is most likely the same issue to https://bugzilla.kernel.org/show_bug.cgi?id=220061 which was fixed last week so you may need to update your kernel.

FakeMichau, 2025-05-15:
From what I can tell it's not fixed yet, moreover the progress seems stalled. Tho some talk is held on the [Arch side](https://gitlab.archlinux.org/archlinux/packaging/packages/linux/-/issues/137) to apply the patch for their builds.

Nevertheless it's not an issue with bluez but the kernel.

39george, 2025-05-16:
Thanks! So I will close on that :)