bluez/bluez #1263: btusb: Intel HCI reset failure results in permanent loss of controller until reboot

dineththilina, 2025-05-13:
Summary:
On systems using Intel Bluetooth adapters (e.g., 0a5c:5842 or similar), the Bluetooth controller occasionally fails during normal operation (e.g., Bluetooth headphone disconnect event). The kernel attempts an HCI reset, but this fails with -19 (ENODEV). After this point, the Bluetooth controller is no longer available in the system — it does not re-register as hci0, cannot be reloaded via modprobe, and USB unbind/rebind has no effect. The only recovery is a full system reboot.

🔍 Environment
Hardware: Dell Precision 3571 (12th Gen Intel)

Bluetooth Device: 0a5c:5842 Broadcom Corp. 58200 (USB)

Kernel version: 6.8.x (Fedora 40)

BlueZ version: 5.81

Driver: btusb, btintel

💥 Trigger
Bluetooth headphones disconnect mid-session (possibly due to idle, power saving, or manual action)

Kernel logs show hci0 trying to reset

Firmware re-init fails, resulting in device disappearance

🧾 dmesg Output (Key Segment)
plaintext
Copy
Edit
Bluetooth: hci0: Device revision is 0
Bluetooth: hci0: Secure boot is enabled
Bluetooth: hci0: OTP lock is enabled
Bluetooth: hci0: API lock is enabled
Bluetooth: hci0: Debug lock is disabled
Bluetooth: hci0: Found device firmware: intel/ibt-0040-0041.sfi
Bluetooth: hci0: Waiting for firmware download to complete
Bluetooth: hci0: Firmware loaded in 2.2 seconds
Bluetooth: hci0: Device booted
Bluetooth: hci0: Applying Intel DDC parameters completed
Bluetooth: hci0: Firmware SHA1: 0xc115e35a
Bluetooth: hci0: Fseq status: Success (0x00)

<=== event trigger: headphone disconnect ===>

Bluetooth: hci0: sending frame failed (-19)
Bluetooth: hci0: Opcode 0x0c1a failed: -19
Bluetooth: hci0: Error when powering off device on rfkill (-19)
Bluetooth: hci0: HCI reset during shutdown failed
After this, bluetoothctl list shows:

arduino
Copy
Edit
No default controller available
🔁 Recovery Attempts (All Failed)
modprobe -r btusb btintel && modprobe btusb → No effect

USB unbind/rebind of /sys/bus/usb/devices/3-8 → No effect

Restarting systemd services (bluetooth.service, bluetoothd) → No effect

btmgmt, bluetoothctl, hcitool, hciconfig → All fail with No controller

Device still visible in lsusb, but kernel refuses to re-register it

✅ Workaround
Adding the following boot parameter prevents the crash entirely:

ini
Copy
Edit
btintel.enable_hci_reset=0
After adding it and rebooting, the device no longer enters the failure state even after disconnections and idle.

🧠 Analysis
This appears to be a firmware handling regression introduced in recent kernels (likely ≥ 6.6), or an overly strict reset handling failure in the btintel HCI driver.

Given that:

The device is still visible via lsusb

Reloading btusb + btintel has no effect

Firmware upload is not retried

The HCI interface is not recreated after failure

...the kernel is not giving userspace any way to recover from a soft failure.

📣 Why This Matters
This isn’t just a desktop annoyance — in my case, this happens on a machine acting as a server to a Bluetooth-dependent app. Rebooting isn't viable without affecting uptime.

🙏 Request
Please consider adding a runtime recovery path or a sysfs debug trigger (e.g., /sys/class/bluetooth/hciX/reset_firmware)

Or allow USB-level reprobe or btusb-level reset after -19 failures

At the very least, retry firmware upload once on HCI reset failure

Vudentz, 2025-05-13:
@dineththilina have you tried sysfs reset method added in the following change:

```
commit 0f8a0013741163459fc6cf4610bb9ed2ec25fb94
Author: Hsin-chen Chuang <chharry@chromium.org>
Date:   Wed Jan 8 20:24:44 2025 +0800

    Bluetooth: Allow reset via sysfs
    
    Allow sysfs to trigger hdev reset. This is required to recover devices
    that are not responsive from userspace.
    
    Signed-off-by: Hsin-chen Chuang <chharry@chromium.org>
    Signed-off-by: Luiz Augusto von Dentz <luiz.von.dentz@intel.com>
```

That said perhaps that is not available for something like FC40, another option is perhaps using something like usbreset:

https://www.man7.org/linux/man-pages/man1/usbreset.1.html

Vudentz, 2025-05-13:
> 💥 Trigger Bluetooth headphones disconnect mid-session (possibly due to idle, power saving, or manual action)
> <=== event trigger: headphone disconnect ===>
> 
> Bluetooth: hci0: sending frame failed (-19) Bluetooth: hci0: Opcode 0x0c1a failed: -19 Bluetooth: hci0: Error when powering off device on rfkill (-19)

There is a single entry to the log above, which is in hci_rfkill_set_block, so the disconnection is very likely being caused by something using rfkill, so perhaps the problem is actually that the flag HCI_RFKILLED is set, so try checking if unblocking makes it works again.

closed 2025-09-04 by github-actions[bot]

sashoalm, 2026-04-10:
I found the following way to "restart" the bluetooth controller if `lsusb | grep -i "Intel"` doesn't show anything:

```
# Since the device is gone from the USB list, we have to tell the PCI USB Controller to look for new devices again.
# Note: This might momentarily disconnect your USB mouse or keyboard.
for dev in /sys/bus/pci/drivers/[uoex]hci_hcd/*:*; do [ -e "$dev" ] && echo -n "${dev##*/}" | sudo tee "${dev%/*}/unbind" > /dev/null && echo -n "${dev##*/}" | sudo tee "${dev%/*}/bind" > /dev/null; done
```

After that it should show in the list again:

```
$ lsusb | grep -i "Intel"
Bus 001 Device 009: ID 8087:0aaa Intel Corp. Bluetooth 9460/9560 Jefferson Peak (JfP)
```

Linux dell 6.14.0-37-generic #37~24.04.1-Ubuntu SMP PREEMPT_DYNAMIC