bluez/bluez #1266: Multiple reconnects during power-on with Bose QuietComfort Ultra headphones

c0f, 2025-05-13:
I am having problems with a long and messy connection process for my Bose QuietComfort Ultra headphones. The headphones are running the latest firmware as confirmed by the Bose IOS app.

Pairing the headphones works without any issues. The headphones pair as a trusted device.

For the tests below the headphones are factory reset then connected to my Kubuntu 24.10 PC via a TP-Link USB Bluetooth adapter. The headphones are only connected to my PC, they are not paired with any other devices.

This is what happens when I power-on my headphones:

- 01 Play video
- 02 Press power-on button on headphones
- 03 Headphones play power on sound
- 04 Headphones announce battery level
- 05 Headphones play connect sound
- 06 Bluetooth settings in KDE show headphones connected but no sound is playing through headphones
- 07 Headphones play connect sound
- 08 Headphones announce battery level
- 09 Headphones play connect sound
- 10 Bluetooth settings in KDE show headphones connected but no sound is playing through headphones
- 11 Wait ~30s, no sound is coming through headphones
- 12 Headphones play connect sound
- 13 Bluetooth settings in KDE show headphones connected but no sound is playing through headphones
- 14 Headphones announce battery level
- 15 Bluetooth settings in KDE show headphones connected and audio is now playing

This process takes ~60 seconds. This process is longer if audio is not playing before I power on the headphones.

Once they connect the headphones work fine.

Tests (I factory reset the headphones between each test):

- **Same PC, different headphones**: I've tried different Bluetooth headphones with the same Kubuntu PC and USB adapter and they work as expected and connect within a few seconds. (For reference the other headphones were AfterShokz OpenMove and Apple Airpods 1). I think this rules out the USB adapter.
- **Same headphones, different PC, different OS**: I've tried the headphones with an IOS device and Windows PC and they connect quickly without any issues. I think this rules out a fault with the headphones.
- **Same headphones, different PC and distro**: I've tried the headphones on an Ubuntu laptop with a different, on-board, Bluetooth controller and I get the same connection issues as with Kubuntu.
- **Tested with BlueZ 5.82**. I booted to a live install of the daily build of Ubuntu 25.10 which uses BlueZ 5.82. I had the same connection problems although this time the headphones played a partial 'call disconnected' message part way through the connection process.

bluetoothctl device info:
```
Device BC:87:FA:9F:ED:20 (public)
        Name: Bose QC Ultra Headphones
        Alias: Bose QC Ultra Headphones
        Class: 0x002c0404 (2884612)
        Icon: audio-headset
        Paired: yes
        Bonded: yes
        Trusted: yes
        Blocked: no
        Connected: yes
        LegacyPairing: no
        UUID: Vendor specific           (00000000-deca-fade-deca-deafdecacaff)
        UUID: Serial Port               (00001101-0000-1000-8000-00805f9b34fb)
        UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
        UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: Advanced Audio Distribu.. (0000110d-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: Vendor specific           (0000eb10-d102-11e1-9b23-00025b00a5a5)
        UUID: Qualcomm Technologies I.. (0000fd92-0000-1000-8000-00805f9b34fb)
        UUID: Google                    (0000fe2c-0000-1000-8000-00805f9b34fb)
        UUID: Bose Corporation          (0000febe-0000-1000-8000-00805f9b34fb)
        UUID: Vendor specific           (9b26d8c0-a8ed-440b-95b0-c4714a518bcc)
        UUID: Vendor specific           (df21fe2c-2515-4fdb-8886-f12c4d67927c)
        Modalias: bluetooth:v009Ep4066d0167
```

Adapter details from btmgmt info:
```
[... 116 lines cut ...]
bluez-obexd/oracular-updates,now 5.77-0ubuntu2 amd64 [installed,automatic]
bluez/oracular-updates,now 5.77-0ubuntu2 amd64 [installed,automatic]
gstreamer1.0-pipewire/oracular,now 1.2.4-1ubuntu1 amd64 [installed,automatic]
libbluetooth3/oracular-updates,now 5.77-0ubuntu2 amd64 [installed,automatic]
libkf5bluezqt-data/oracular,oracular,now 5.116.0-0ubuntu1 all [installed,automatic]
libkf5bluezqt6/oracular,now 5.116.0-0ubuntu1 amd64 [installed,automatic]
libkf6bluezqt-data/oracular,oracular,now 6.6.0-0ubuntu2 all [installed,automatic]
libkf6bluezqt6/oracular,now 6.6.0-0ubuntu2 amd64 [installed,automatic]
libkpipewire-data/oracular,oracular,now 6.1.5-0ubuntu1 all [installed,automatic]
libkpipewire6/oracular,now 6.1.5-0ubuntu1 amd64 [installed,automatic]
libkpipewiredmabuf6/oracular,now 6.1.5-0ubuntu1 amd64 [installed,automatic]
libkpipewirerecord6/oracular,now 6.1.5-0ubuntu1 amd64 [installed,automatic]
libpipewire-0.3-0t64/oracular,now 1.2.4-1ubuntu1 amd64 [installed,automatic]
libpipewire-0.3-common/oracular,oracular,now 1.2.4-1ubuntu1 all [installed,automatic]
libpipewire-0.3-modules/oracular,now 1.2.4-1ubuntu1 amd64 [installed,automatic]
libspa-0.2-bluetooth/oracular,now 1.2.4-1ubuntu1 amd64 [installed,automatic]
libwireplumber-0.5-0/oracular,now 0.5.6-1ubuntu1 amd64 [installed,automatic]
pipewire-bin/oracular,now 1.2.4-1ubuntu1 amd64 [installed,automatic]
pipewire-pulse/oracular,now 1.2.4-1ubuntu1 amd64 [installed,automatic]
pipewire/oracular,now 1.2.4-1ubuntu1 amd64 [installed,automatic]
qml-module-org-kde-bluezqt/oracular,now 5.116.0-0ubuntu1 amd64 [installed,automatic]
qml6-module-org-kde-bluezqt/oracular,now 6.6.0-0ubuntu2 amd64 [installed,automatic]
qml6-module-org-kde-pipewire/oracular,now 6.1.5-0ubuntu1 amd64 [installed,automatic]
wireplumber/oracular,now 0.5.6-1ubuntu1 amd64 [installed,automatic]
```

Vudentz, 2025-05-13:
@c0f please add the btmon logs so we can check why it is disconnecting.

c0f, 2025-05-13:
Hi, btmon and bluetoothd -d outputs attached.

[bluetoothd-debug.txt](https://github.com/user-attachments/files/20195704/bluetoothd-debug.txt)
[btmon-output.txt](https://github.com/user-attachments/files/20195703/btmon-output.txt)

Vudentz, 2025-05-13:
Looks like a connection timeout:

```
> HCI Event: Disconnect Complete (0x05) plen 4            #359 [hci0] 21.184433
        Status: Success (0x00)
        Handle: 8 Address: BC:87:FA:9F:ED:20 (Bose Corporation)
        Reason: Connection Timeout (0x08)
@ MGMT Event: Device Disconnected (0x000c) plen 8     {0x0001} [hci0] 21.184523
        BR/EDR Address: BC:87:FA:9F:ED:20 (Bose Corporation)
        Reason: Connection timeout (0x01)
```

Maybe some incompatibility of controller and device, anyway we can't do much if the connection just times out.

closed 2025-09-04 by github-actions[bot]

BlueskyFR, 2026-10-31:
Hi there! I had problems with my QC Ultra as well, and it seems since then some fixes were pushed to this repo, but not yet released in a new bluez version.
I the meantime, you can build bluez from master and it will work.

For anyone on NixOS (or not!), you can look at my patch here for instance: https://github.com/BlueskyFR/dotfiles/commit/82bb1e9162e2e69d240ed3bb326f7e6c96a9dfd8