bluez/bluez #1282: BASS can't sync with the K70's local broadcast audio

chinli, 2025-05-21:
DUT environment：BlueZ 5.82+pipewire 1.3.83+kernel 6.12
BIS source: Redmi K70 phone

On the K70 mobile phone, connect to the DUT and enable local audio broadcasting from the Bluetooth settings screen.
The bc_sid used by iso_listen_bis in kernel is the default value 0x00，btmon log as below:

[btmon.log](https://github.com/user-attachments/files/20366799/btmon.log)

> ACL Data RX: Handle 16 flags 0x01 dlen 3                #434 [hci0] 20.510023
      ATT: Write Command (0x52) len 30
        Handle: 0x0065 Type: Broadcast Audio Scan Control Point (0x2bc7)
          Data[28]: 0201e514cb20085c020b21c901780001ffffffff0703020400020501
            Opcode: Add Source (0x02)
            Source_Address_Type: 1
            Source_Address: 5C:08:20:CB:14:E5
            **Source_Adv_SID: 2**
            Broadcast_ID: 0xc9210b
            PA_Sync_State: Synchronize to PA - PAST available
            PA_Interval: 0x0078
            Num_Subgroups: 1
            Subgroup #0:
              BIS_Sync State: 0xffffffff
              Metadata: #0: len 0x03 type 0x02
              Metadata:   04 00                                            ..              
              Metadata: #1: len 0x02 type 0x05
              Metadata:   01  
.......

< HCI Command: LE Periodic Adver.. (0x08|0x0044) plen 14  #435 [hci0] 20.512677
        Options: 0x0000
        Use advertising SID, Advertiser Address Type and address
        Reporting initially enabled
        **SID: 0x00**
        Adv address type: Random (0x01)
        Adv address: 5C:08:20:CB:14:E5 (Resolvable)
        Skip: 0x0000
        Sync timeout: 20000 msec (0x07d0)
        Sync CTE type: 0x0000

Vudentz, 2025-05-21:
> DUT environment：BlueZ 5.82+pipewire 1.3.83+kernel 6.12 BIS source: Redmi K70 phone
>
> On the K70 mobile phone, connect to the DUT and enable local audio broadcasting from the Bluetooth settings [...]

This has been fixed already:

https://github.com/bluez/bluetooth-next/commit/e2d471b7806b09744d65a64bcf41337468f2443b

chinli, 2025-05-23:
Hi @Vudentz,
Unfortunately, the patch above did not resolve the issue. The btmon log shows that after receiving the extended advertisement from the target device, the SID was not updated, and the PA Create Sync command was issued again.

[btmon_sid.log](https://github.com/user-attachments/files/20404883/btmon_sid.log)
```
< HCI Command: LE Periodic Adver.. (0x08|0x0044) plen 14  #518 [hci0] 36.212659
        Options: 0x0000
        Use advertising SID, Advertiser Address Type and address
        Reporting initially enabled
        **SID: 0x00**
        Adv address type: Random (0x01)
        Adv address: 42:1D:B8:54:0C:C9 (Resolvable)
          Identity type: Random (0x01)
          Identity: FC:5B:8C:F7:5D:FB (Static)
        Skip: 0x0000
        Sync timeout: 20000 msec (0x07d0)
        Sync CTE type: 0x0000
> HCI Event: Command Status (0x0f) plen 4                 #519 [hci0] 36.214135
      LE Periodic Advertising Create Sync (0x08|0x0044) ncmd 1
        Status: Success (0x00)
> HCI Event: LE Meta Event (0x3e) plen 91                 #520 [hci0] 36.352046
      LE Extended Advertising Report (0x0d)
        Num reports: 1
        Entry 0
          Event type: 0x0000
            Props: 0x0000
            Data status: [0;32mComplete[0m
          Address type: Random (0x01)
          Address: 42:1D:B8:54:0C:C9 (Resolvable)
          Identity type: Random (0x01)
          Identity: FC:5B:8C:F7:5D:FB (Static)
          Primary PHY: LE 1M
          Secondary PHY: LE 2M
          **SID: 0x02**
          TX power: 0 dBm
          RSSI: -75 dBm (0xb5)
          Periodic advertising interval: 150.00 msec (0x0078)
          Direct address type: Public (0x00)
          Direct address: 00:00:00:00:00:00 (OUI 00-00-00)
          Data length: 0x41
```
I believe an alternative solution would be to implement the **PAST** (Periodic Advertising Sync Transfer) procedure. This allows the controller to directly exchange and apply PA information, making the process more efficient and reliable.

Vudentz, 2025-05-23:
> Hi [@Vudentz](https://github.com/Vudentz), Unfortunately, the patch above did not resolve the issue. The btmon log shows that after receiving the extended advertisement from the target device, the [...]

You also need the following change to go along with it:

```
commit aedd7b936b7240614a5305a8039a402d2da335ea
Author: Luiz Augusto von Dentz <luiz.von.dentz@intel.com>
Date:   Tue Apr 15 16:00:59 2025 -0400

    bap: Add support for using SID for broadcast receiver
    
    Broadcast receiver needs to set the SID of the advertisement, for now
    that is done by using 0xff which indicates to the kernel to perform to
    discover what is the actual SID in use over the air.

```

Using PAST may actually be a good idea, but controllers are not required to implement it in order to support synchronized received feature.

Vudentz, 2025-05-27:
> Hi [@Vudentz](https://github.com/Vudentz)
> 
> Adding this patch to BlueZ still does not resolve the issue. My current environment is as follows:
> 
>     * **BlueZ:** 5.82 + `bap: Add support for using SID for broadcast receiver`
> 
>     * **Kernel:** 6.14 + multiple patches
>       
>       * `e2d471b7806b` Bluetooth: ISO: Fix not using SID from adv report
>       * `024421cf3992` Bluetooth: hci_conn: Fix not setting timeout for BIG Create Sync
>       * `6d0417e4e1cf` Bluetooth: hci_conn: Fix not setting conn_timeout for Broadcast Receiver
> 
> 
> According to the `btmon` log, after issuing the `HCI_LE_Periodic_Advertising_Create_Sync_Cancel` command, the `PA Create Sync` command was not re-initiated.
> 
> [btmon.log](https://github.com/user-attachments/files/20436584/btmon.log)
> 
> Moreover, as the bc_sid value is already available from the BASS characteristic, it might be more appropriate to configure the SID using setsockopt prior to invoking iso_listen_bis(), similar to the existing approach for QoS configuration.

Hmm, are you trying Broadcast Assistant rather the just Broadcast Sink? Not sure what is the characteristic you are referring to, but if it is under BASS it is not relevant to a pure Broadcast Sink.

Vudentz, 2025-05-28:
> > Hmm, are you trying Broadcast Assistant rather the just Broadcast Sink? Not sure what is the characteristic you are referring to, but if it is under BASS it is not relevant to a pure Broadcast Sink.
> 
> Below is my test environment and the steps taken, focusing exclusively on the Broadcast Sink role.
> 
> ```
> DUT environment：BlueZ 5.82+pipewire 1.3.83+kernel 6.12
> BIS source & Assistant: Redmi K70 phone
> 
> On the K70 mobile phone, connect to the DUT and enable local audio broadcasting from the Bluetooth settings screen.
> ```
> 
> Can you share the btmon.log from your test with this patch? I want to compare it and see if there are any differences. [bluez/bluetooth-next@e2d471b](https://github.com/bluez/bluetooth-next/commit/e2d471b7806b09744d65a64bcf41337468f2443b)

If it is a pure broadcast sink then the bass code is not relevant, all you need to do is what is described in the broadcast-sink.bt:

https://github.com/bluez/bluez/blob/master/client/scripts/broadcast-sink.bt

That will enumerate the discovered broadcast sources as transports, than you do transport.select/transport.acquire.

Vudentz, 2025-05-29:
> > > > Hmm, are you trying Broadcast Assistant rather the just Broadcast Sink? Not sure what is the characteristic you are referring to, but if it is under BASS it is not relevant to a pure Broadcast Sink.
> > > 
> > > 
> > > Below is my test environment and the steps taken, focusing exclusively on the Broadcast Sink role.
> > > ```
> > > DUT environment：BlueZ 5.82+pipewire 1.3.83+kernel 6.12
> > > BIS source & Assistant: Redmi K70 phone
> > > 
> > > On the K70 mobile phone, connect to the DUT and enable local audio broadcasting from the Bluetooth settings screen.
> > > ```
> > > 
> > > 
> > >     
> > >       
> > >     
> > > 
> > >       
> > >     
> > > 
> > >     
> > >   
> > > Can you share the btmon.log from your test with this patch? I want to compare it and see if there are any differences. [bluez/bluetooth-next@e2d471b](https://github.com/bluez/bluetooth-next/commit/e2d471b7806b09744d65a64bcf41337468f2443b)
> > 
> > 
> > If it is a pure broadcast sink then the bass code is not relevant, all you need to do is what is described in the broadcast-sink.bt:
> > https://github.com/bluez/bluez/blob/master/client/scripts/broadcast-sink.bt
> > That will enumerate the discovered broadcast sources as transports, than you do transport.select/transport.acquire.
> 
> This is not a pure Broadcast Sink, it uses a mobile phone as the Assistant to configure BASS and synchronize with the BIS source.

And what made you think that a Linux system would need a mobile phone to assist? I don't think we even have supported role for BASS to act as a BASS server.

chinli, 2025-05-29:
> And what made you think that a Linux system would need a mobile phone to assist? I don't think we even have supported role for BASS to act as a BASS server.

I am currently developing a Bluetooth speaker product that already supports the BASS server role — that is, the phone acts as the Assistant to configure the BIS source. However, synchronization fails when the PA SID of the BIS source is non-zero. Resolving this issue is the focus of the current investigation.

closed 2025-05-29 by BluezTestBot