bluez/bluez #1284: Remove CIG is send before HCI disconnect complete event

VinitMehta-NXP, 2025-05-21:
Hello,

I am validating the BAP source using BlueZ. Below are the test setup details:

- Test Case: 2-CIS
- DUT: BlueZ Host (BAP Source)
- Reference Device: Samsung Galaxy Buds 2 Pro (BAP Sink)
- BlueZ Version: 5.79

Currently, the kernel sends the HCI LE Remove CIG command immediately after BAP socket cleanup, without waiting for the HCI Disconnection Complete event for the last CIS. If the remote device is out of range, the HCI disconnection for a CIS is delayed until the supervision timeout completes. This causes the controller to respond with "command disallowed" as the CIG state machine is still active, leading to inconsistencies between the host and controller.

Would it be advisable to modify the kernel's CIS cleanup process to wait for the HCI Disconnection Complete event for the last CIS before sending the Remove CIG command?

Please provide any comments or suggestions.

Regards,
Vinit

Vudentz, 2025-05-21:
@VinitMehta-NXP any reason not to be using 5.82, or better the latest git?

VinitMehta-NXP, 2025-05-22:
@Vudentz The DUT platform is based on Arch and is still running 6.6.52 kernel. I do see some compatibility issue and intermittent bluetoothd crash if using 5.82 or latest bluez. Will plan to upgrade to more recent kernel version and will give it a try using latest git.

One question though, AFAIK once the CIS socket is shut down and the CIS cleanup function is called it has implementation to remove CIG function if cleanup is done for last CIS withing a CIG. 
https://github.com/bluez/bluetooth-next/blob/17487bc76dcf0b3bd9dc49a64c9f44066b0200f0/net/bluetooth/hci_conn.c#L896C2-L896C19

I was not very sure if upgrading bluez to latest may change this behavior and hence raised the query if this is a valid issue or no. Please correct if I am missing something. 

I will anyways try to upgrade the kernel and then using latest bluez and verify again.

Thanks

pv, 2025-06-14:
> Currently, the kernel sends the HCI LE Remove CIG command immediately after BAP socket cleanup, without waiting for the HCI Disconnection Complete event for the last CIS

The cleanup is called when `hci_conn` is deleted, but that waits for Disconnect completion. Eg. here, https://github.com/bluez/bluetooth-next/blob/17487bc76dcf0b3bd9dc49a64c9f44066b0200f0/net/bluetooth/hci_sync.c#L5585 and iirc closing ISO socket ends up here via hci_conn_drop() -> hci_conn_timeout()

closed 2025-06-19 by VinitMehta-NXP