bluez/bluez #1290: AVRCP doesn't work after set ControlerMode as dual

xihua13104, 2025-05-26:
DUT environment：BlueZ 5.82+pipewire 1.3.83+kernel 6.12
Audio source: Redmi K70 phone

After I set ControllerMode to dual, I turned on Classic visibility and ble advertisement at the same time, and used K70 to connect to the DUT. Both Classic and ble were connected. At this time, I used the player subcommand in bluetoothctl to control the audio track, and it prompted "No default player available". To track the btmon, i got the error notice as below, the DUT reject the volume change RegisterNotification.

[k70-dual-btmon.txt](https://github.com/user-attachments/files/20439041/k70-dual-btmon.txt)

```shell
> ACL Data RX: Handle 51 flags 0x02 dlen 22               #559 [hci0] 27.064618
      Channel: 67 len 18 [PSM 23 mode Basic (0x00)] {chan 3}
      AVCTP Control: Command: type 0x00 label 0 PID 0x110e
        AV/C: Notify: address 0x48 opcode 0x00
          Subunit: Panel
          Opcode: Vendor Dependent
          Company ID: 0x001958
          AVRCP: RegisterNotification pt Single len 0x0005
            EventID: 0x0d (EVENT_VOLUME_CHANGED)
            Interval: 0x00000000 (0 seconds)
bluetoothd[2295]: = profiles/audio/avrcp.c:handle_vendordep_pdu()..   27.065031
bluetoothd[2295]: = profiles/audio/avrcp.c:handle_vendordep_pdu()..   27.065226
bluetoothd[2295]: = profiles/audio/avrcp.c:avrcp_handle_register_..   27.065344
bluetoothd[2295]: = profiles/audio/avrcp.c:handle_vendordep_pdu()..   27.065458
bluetoothd[2295]: < ACL Data TX: Ha.. flags 0x00 dlen 18  #560 [hci0] 27.065375
      Channel: 69 len 14 [PSM 23 mode Basic (0x00)] {chan 3}
      AVCTP Control: Response: type 0x00 label 0 PID 0x110e
        AV/C: Rejected: address 0x48 opcode 0x00
          Subunit: Panel
          Opcode: Vendor Dependent
          Company ID: 0x001958
          AVRCP: RegisterNotification pt Single len 0x0001
            Error: 0x01 (Invalid Parameter)
```  

I think there is a conflict between classic audio and ble audio, because when I set ControllerMode to single mode (bredr or le), AVRCP & MCP can work normally.

Vudentz, 2025-05-27:
@xihua13104 perhaps media_transport_get_device_volume is failing, that said Im not sure what you would like to have both BR and LE connected simultaneously?

xihua13104, 2025-05-28:
Hi @Vudentz 

Thanks for looking at this! My goal is to allow a phone to remain paired over both Classic (BR/EDR) and LE Audio simultaneously—so that the user can switch back and forth between A2DP and LE Audio without having to re-connect.

closed 2025-05-28 by BluezTestBot