bluez/bluez #1293: Does not enable auto_connect during Pair process

WadeWangHP, 2025-05-27:
During pair with HID LE device like mouse / keyboard, no one enables "auto connect", it causes mouse / keyboard can't auto connect back after link lost.
This issue was introduced by commit 507ba12483c3b6504d7fc0abfbbe1b4a0c8fa006 by luiz.von.dentz@intel.com since bluez 5.69, this patch "temporary mark auto_connect as disabled" during scanning process, and hope "when the device is connected it attempts to enable it.", but this patch does not add "device_set_auto_connect(device, TRUE)" call during Pair process, cause no one enabling "auto connect" until disconnect then connect, or load_devices when bluetooth service restart.
Can you fix this issue ASAP?

Vudentz, 2025-05-27:
@WadeWangHP what version are you using? Afaik the auto_connect list gets updated based on drivers, so when a driver is marked to auto_connect then the device should auto_connect.

WadeWangHP, 2025-05-29:
No this issue on 5.82. and a check the commit history, it should be fixed by [39467578207889fd015775cbe81a3db9dd26abea](https://git.kernel.org/pub/scm/bluetooth/bluez.git/commit/?id=39467578207889fd015775cbe81a3db9dd26abea)

closed 2025-05-29 by Vudentz