bluez/bluez #1319: 5.83 regression: btmgmt: unable to send *cmd

NekoCWD, 2025-06-04:
Regression commit: [2d2bc7150 client: Port "mgmt" menu to pre_run](https://github.com/bluez/bluez/commit/2d2bc715025caac8530982190fc4336fdfb6a50c)

Description:
	btmgmt upgrade causes error on any btmgmt commands.

Tested commands
- `btmgmt info` -  `Unable to send read_info cmd`
- `btmgmt config` - `Unable to send unconf_index_list cmd`
- `btmgmt revision` - `Unable to send read_version cmd`
- `btmgmt power on` - `Unable to send Set Powered cmd`

Regression since: 5.82

vipaoL, 2025-06-04:
I also encountered this on xiaomi-pipa (Xiaomi Pad 6) with postmarketOS, and some people with xiaomi-pipa on Arch Arm also reported this error

ausil, 2025-06-19:
I am seeing this on Fedora on a lenovo x13s it has a wcn6855 device

NekoCWD, 2025-06-22:
Should i send this issue to some mailing list? Or how can i communicate with bluez devs?

NekoCWD, 2025-07-05:
# Issue wasn't fixed:
Interactive mode now works fine, but it still fail non-interactive checks

## Non interactive (still broken):
```
$ btmgmt -i 0 info
Unable to send read_info cmd
```
## Interactive (fixed):
```
$ btmgmt
[mgmt]> info
Index list with 1 item
hci0:	Primary controller
	addr AA:BB:CC:DD:EE:FF version 12 manufacturer 93 class 0x6c010c
	supported settings: powered connectable fast-connectable discoverable bondable link-security ssp br/edr le advertising secure-conn debug-keys privacy static-addr phy-configuration wide-band-speech cis-central cis-peripheral 
	current settings: powered ssp br/edr le secure-conn wide-band-speech cis-central cis-peripheral iso-broadcaster sync-receiver 
	name Meow
	short name 
```

konzerterlebnisberichte, 2025-09-25:
I can't use either the interactive mode or the non interactive mode, always get unable to send error. ArchARM on Lenovo THinkpad X13S

NekoCWD, 2025-09-25:
> I can't use either the interactive mode or the non interactive mode, always get unable to send error. ArchARM on Lenovo THinkpad X13S

What bluez version?

konzerterlebnisberichte, 2025-09-26:
> What bluez version?

5.83-1