bluez/bluez #1395: Unable to pair Wii remote (Authentication Failed)

camthehaxman, 2025-07-05:
I am trying to pair a Wii remote with a Raspberry Pi running Void Linux (Linux kernel version 6.6.78, bluez version 5.78), but I am unable to pair the Wii remote due to some authentication failure.

## Steps to reproduce (all commands run as root):
1. Install the bluez package (`xbps-install bluez`)
2. Enable the bluetoothd service (`ln -s /etc/sv/bluetoothd /var/service`) and verify that it is running (`sv status bluetoothd`) 
3. Enable the hid-wiimote kernel module (`modprobe hid-wiimote`)
4. Run `bluetoothctl` and enter the following commands:
```
default-agent
scan on
```
5. Press the sync button of the Wii remote (inside the battery cover) or press the 1 and 2 buttons together. This will put the Wii remote into discoverable mode for 20 seconds.
6. Amidst the never ending spam of discovered devices, you should see the Wii remote listed as `Nintendo RVL-CNT-01` along with its MAC address.
```
[bluetooth]# [NEW] Device 00:1A:E9:F1:F1:86 Nintendo RVL-CNT-01
```
7. Attempt to pair the device. You may have to press the Wii remote's sync button or 1+2 again before doing so. This fails with status 0x05 (Authentication Failed).
```
[bluetooth]# pair 00:1A:E9:F1:F1:86
Attempting to pair with 00:1A:E9:F1:F1:86
[bluetooth]# hci0 device_flags_changed: 00:1A:E9:F1:F1:86 (BR/EDR)
[bluetooth]#      supp: 0x00000001  curr: 0x00000000
[bluetooth]# hci0 type 7 discovering off
[bluetooth]# hci0 00:1A:E9:F1:F1:86 type BR/EDR connected eir_len 26
[bluetooth]# [CHG] Device 00:1A:E9:F1:F1:86 Connected: yes
[Nintendo RVL-CNT-01]# hci0 00:1A:E9:F1:F1:86 auth failed with status 0x05 (Authentication Failed)
[Nintendo RVL-CNT-01]# hci0 00:1A:E9:F1:F1:86 type BR/EDR disconnected with reason 0
[Nintendo RVL-CNT-01]# [CHG] Device 00:1A:E9:F1:F1:86 Connected: no
```

## Expected behavior
Wii remote should pair. Bluez has code to automatically generate a PIN for the Wii remote (which is the bytes of the MAC address in reverse).

## Actual behavior
Pairing fails with status 0x05 (Authentication Failed)

rctgamer3, 2025-09-06:
Still an issue

fonikz, 2025-11-16:
Everything I've ever read says to pair the wiimote with 1+2. But on first pair, I am finding that the red sync button will connect but not 1+2. And from then on just pressing any button will reconnect. I recently migrated to Arch so I should be on the latest bluez now.

traverseda, 2026-01-17:
Related to #911 . It looks like some code was merged in to fix this. What release is the referenced fix landing in? @Vudentz , sorry for the ping but can you provide some insight into what version I could expect this fix to be a part of?