bluez/bluez #1417: Latest version of Bluez breaks connection with JBL LIVE650

TheDokT0r, 2025-07-10:
- BlueZ version: 5.83 and master 
- Distro: Arch
- Desktop Environment:
  * KDE Plasma 6.4.2
  * Gnome 48
- Kernel version:
6.15.5-arch1-1
- `lsusb`:
```
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 14ed:29b6 Shure Inc. X2u Adapter
Bus 001 Device 003: ID 0b05:19af ASUSTek Computer, Inc. AURA LED Controller
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 002: ID 04f2:3f41 Chicony Electronics Co., Ltd HP Wired Desktop 320K Keyboard
Bus 003 Device 003: ID 2357:0604 TP-Link TP-Link UB500 Adapter
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 007 Device 002: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
```
- Bluetooth devices:

```
Device 2C:FD:B3:28:B7:B3 JBL LIVE650BTNC
Device E8:9E:13:47:22:7E WF-C700N
```

## Description of Problem:
Before the newest version of bluez, my JBL headphones worked just fine. But after the new version, every time I try and switch to the JBL headphones as the audio source, it automatically disconnects. The issue also persists with the git version. I've tried to downgrade bluez  to 5.72 and everything works just fine. The issue consists with both pipewire and pulseaudio. Btw, my other Bluetooth headphones (Sony) works just fine. the issue is only with the JBL headphones.

## How Reproducible:
1. Connect the Bluetooth headphones
2. go the the audio settings tray in Plasma or Gnome. You can also just try and change the audio volume from your keyboard shortcuts. 

### Actual Results:
Bluetooth headphones disconnects after the system tries to switch to them as the audio source.

### Expected Results:
The headphones should work without an issue, as they did before the last release.

pv, 2025-07-10:
Pipewire issue with the btmon dump: https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/4789#note_2996800

TheDokT0r, 2025-07-10:
Yeah, that's an issue I've made on the PipeWire repo. It was only afterwards I've realised the issue was with bluez.

bjo81, 2025-07-14:
Same issue here with a LIVE650, downgrading the kernel to 6.15.4 like in https://www.reddit.com/r/archlinux/comments/1lukcs9/linux_6155arch1_broke_bluetooth_headphones/ helps.

bitfroster, 2025-07-16:
The issue still persists in 6.15.6

salfum, 2025-07-18:
Same issue with JBL Flip 7 bluetooth speaker.
System: Manjaro Linux
Kernel: 6.9.9-zen1
Audio: Pipewire 1.4.1-1
Bluez: 5.72-2, 5.83-1
Devices:
```
bluetoothctl devices
Device 50:1B:6A:D3:01:AC JBL Flip 7
Device 90:89:5F:C9:3C:C8 Wireless Controller
Device AC:80:0A:A3:DA:B5 WF-1000XM5
```
Trying to connect:
```
Attempting to connect to 50:1B:6A:D3:01:AC
[NEW] Media /org/bluez/hci0 
	SupportedUUIDs: 0000110a-0000-1000-8000-00805f9b34fb
	SupportedUUIDs: 0000110b-0000-1000-8000-00805f9b34fb
[CHG] Device 50:1B:6A:D3:01:AC Connected: yes
[NEW] Endpoint /org/bluez/hci0/dev_50_1B_6A_D3_01_AC/sep2 
[NEW] Endpoint /org/bluez/hci0/dev_50_1B_6A_D3_01_AC/sep1 
[NEW] Endpoint /org/bluez/hci0/dev_50_1B_6A_D3_01_AC/sep4 
[NEW] Endpoint /org/bluez/hci0/dev_50_1B_6A_D3_01_AC/sep3 
[DEL] Endpoint /org/bluez/hci0/dev_50_1B_6A_D3_01_AC/sep4 
Failed to connect: org.bluez.Error.Failed br-connection-unknown
```

DaggLo, 2025-07-20:
Seem like related:

- https://github.com/bluez/bluez/issues/1422
- https://gitlab.archlinux.org/archlinux/packaging/packages/linux/-/issues/149

AMDBartek, 2025-08-13:
Experiencing this issue on a custom kernel (`6.15.7`) on CachyOS which I manually built with `0001-Bluetooth-L2CAP-Fix-attempting-to-adjust-outgoing-MT.patch` (made from https://github.com/bluez/bluetooth-next/commit/514f29ce5bf37160f6f4e124a9cc06ca77cd5ab1) and the issues do not seem to be related.

It happens with my Moondrop Space Travel but not with my Fosi Audio BT30D PRO. The two main surface level discrepancies are that the Space Travel support the AAC codec while the Fosi uses SBC or good old phono in.

Using `sudo downgrade bluez bluez-utils bluez-hid2hci bluez-libs` and downgrading all of these packages to `5.82` fixes the problem, meaning that this had to have been introduced in BlueZ itself somewhere between the bump to `5.83`.

closed 2025-09-27 by github-actions[bot]

CollieIsCute, 2025-11-04:
Me too. The JBL flip7 speaker failed to connect to my machine. After trying lots of method, I finally used wireplumber script to force the whole bluetooth service to use SBC as the only codec. 

However, today I found out my JBL flip7 works well without those workaround script, and can pair with my machine natively. Maybe it's time to roll your system and try again :)

```fish
~
❯ paru -Q bluez bluez-utils
bluez 5.84-1
bluez-utils 5.84-1

❯ fastfetch
                  -`                     collie@archlinux
                 .o+`                    ----------------
                `ooo/                    OS: Arch Linux x86_64
               `+oooo:                   Kernel: Linux 6.17.6-arch1-1
              `+oooooo:                  Uptime: 16 mins
              -+oooooo+:                 Packages: 1048 (pacman), 11 (flatpak-user)
            `/:-:++oooo+:                Shell: fish 4.1.2
           `/++++/+++++++:               Display (XV272U RV): 2560x1440 @ 1.25x in 27", 170 Hz [External]
          `/++++++++++++++:              Display (PG27AQDM): 2560x1440 @ 1.25x in 27", 240 Hz [External]
         `/+++ooooooooooooo/`            Display (MSI MP242): 1920x1080 in 24", 75 Hz [External]
        ./ooosssso++osssssso+`           WM: Hyprland 0.51.1 (Wayland)
       .oossssso-````/ossssss+`          Cursor: Adwaita
      -osssssso.      :ssssssso.         Terminal: ghostty 1.2.3-arch2
     :osssssss/        osssso+++.        Terminal Font: JetBrainsMono Nerd Font (12pt)
    /ossssssss/        +ssssooo/-        CPU: 13th Gen Intel(R) Core(TM) i5-13500 (20) @ 4.80 GHz
  `/ossssso+/:-        -:/+osssso+-      GPU: AMD Radeon RX 9070 XT [Discrete]
 `+sso+:-`                 `.-/+oso:     Memory: 5.22 GiB / 31.08 GiB (17%)
`++:.                           `-/+/    Swap: Disabled
.`                                 `/    Disk (/): 581.90 GiB / 1.86 TiB (31%) - btrfs
                                         Local IP (wlp0s20f3): 192.168.68.109/24
                                         Locale: en_US.UTF-8
```