bluez/bluez #1444: x86_64 5.83-2.fc41 - spa.bluez5.sink.media error 24 - Philips TAH4205

Sir-DotJ, 2025-07-17:
Having issues identical to first error at https://forum.manjaro.org/t/bluetooth-headphones-disconnect-randomly-with-pipewire-installed/137025

The error `spa.bluez5.sink.media: ...: error 24` occurs when using my Philips TAH4205 (includes handsfree microphone).

On Fedora Linux 41 (Workstation Edition)

How it occurs:
- Connecting to the device and registering it goes fine;
- When going into sound settings the device gets disconnected and journal prints out the error above;
- When trying to play any audio on Rhythmbox the device gets disconnected, same error.

What I tried to do to fix it:
- Updating everything to the latest version;
- Turning off hands free mode (with systemctl reboot): https://wiki.archlinux.org/title/Bluetooth_headset#Disable_PipeWire_HSP/HFP_profile
   - Additionally tried to switch to bap, didn't work either

This only occurs on the latest version: bluez.x86_64 **5.83-2**.fc41

When downgrading to bluez.x86_64 **5.78-1**.fc41 the issue goes away.

For some reason this only occurs on Philips TAH4205, my earbuds (JBL TUNE215TWS) work just fine on the latest version.

Headphones themselves work just fine on my smartphone.

axelf4, 2025-07-23:
I have a similar issue where the server says it accepts MTUs of 672 bytes, but the bluez client tries to send 969 bytes when first streaming Bisected to commit 04b220a1e485deb3bf025cdc706a4c3958a1c294, which makes https://github.com/bluez/bluez/blob/52de8da1a99cd33b517284f87d231055a0dee6bb/profiles/audio/avdtp.c#L2526 actually turn on auto-IMTU (uncommenting this line is a workaround for me). Commit torvalds/linux@d24e4a7fedae121d33fb32ad785b87046527eedb may be the fix, by not using our large 1021 IMTU for the implied 672 OMTU (?).

LinkUpGames, 2025-07-28:
Same issue here. I use Fedora and the last stable version without this error was kernel v6.15.4