bluez/bluez #1452: Wrong AcquireNotify socket is closed when a device disconnects

john-allthenticate, 2025-07-21:
When multiple unbonded clients are subscribed to notifications for a GATT characteristic via `AcquireNotify`, and one of the devices disconnects, sometimes the notify socket for one of the other clients is closed.

Using BlueZ 5.83 on Ubuntu 24.04 (kernel 6.11.0-29-generic).

[bluez.log](https://github.com/user-attachments/files/21351902/bluez.log)

As far as I can tell, this is happening because `ccc_write_cb` is called with a null `op` when a device disconnects. This causes the first client in `chrc->notify_ios` to always get removed, regardless of which device actually disconnected.

EZIMM, 2025-08-25:
I am seeing the same/similar issue using BlueZ 5.78 on Linux kernel 6.6.48. The code in question in 5.83 mentioned by @john-allthenticate appears to be the same as 5.78.

What I see is with multiple connected clients subscribed to a GATT characteristic via `AcquireNotify`, when one client disconnects and reconnects one other connected client stops receiving notifications.

When this happens for two connected clients, the notifications only work with the client that reconnected most recently, resulting in notifications "ping-ponging" between them with each disconnect/reconnect. When this happens for three connected clients, only one additional client loses notifications (not all clients) when one client disconnects and reconnects.

EZIMM, 2025-09-29:
I have been running with this patch on ccc_write_cb() in gatt-database.c for about a month and have not seen any issues with the wrong client losing its notifications. Occasionally I have seen that a client disconnect event does not trigger when it disconnects, but I am not sure if that is related to this or if it is another issue.

Are there any plans to address this officially?

Vudentz, 2025-10-01:
> I have been running with this patch on ccc_write_cb() in gatt-database.c for about a month and have not seen any issues with the wrong client losing its notifications. Occasionally I have seen that [...]

It would help if you share the change and other details about how to test them.

EZIMM, 2025-10-01:
Attached is the patch I generated for 5.78, simple null checks and return to prevent the call to queue_remove_if with null parameters. As I am not fully aware of any implications this may have on other use-cases, I went with the least invasive change for my use case.

I register a characteristic that is set for indicate and connect 2 mobile devices in sequence and have them enable notifications on that characteristic. AcquireNotify is triggered and threads are spun up that write unique data to each individual device through the FD pipe. I then disconnect the second device which triggers the disconnect calls. At this point the first device stops receiving the indications. I then reconnect the second device and it enables notifications and it is fine. The first device must be disconnected, reconnected, and enable notifications again to get them, but in doing so, the second device stops getting notifications.

With this patch on my environment, I no longer see the wrong device stop receiving notifications when not intended.

[0001-fix-acquirenotify-socket-closed-on-disconnect.patch](https://github.com/user-attachments/files/22648720/0001-fix-acquirenotify-socket-closed-on-disconnect.patch)

Thanks

closed 2025-11-06 by github-actions[bot]