bluez/bluez #1457: `bluetoothctl help --help` kinda stucks

ratijas, 2025-07-23:
If you run `bluetoothctl help --help` it prints the following and then just kinda hangs until you hit Ctrl+C:

```
❯ bluetoothctl help --help
Display help about this program
Usage:
         help                                             
[NEW] Media /org/bluez/hci0 
        SupportedUUIDs: 0000110a-0000-1000-8000-00805f9b34fb
        SupportedUUIDs: 0000110b-0000-1000-8000-00805f9b34fb
```

PranavBhattarai, 2025-08-24:
In my case, it didn't even print out any UUDI.
```
shivoham@x ~> bluetoothctl help --help
Display help about this program
Usage:
	 help
^C⏎
```
This project needs a decent number of devs and funding. There is barely any discussion or response, or engagement on any issues. 

The development and progress are there, though. Slow-paced.

ratijas, 2025-09-27:
bluetoothctl is kinda stale, the project will be closed in 5 days

sysAdminPranav, 2025-09-28:
no it won't @ratijas

closed 2025-10-01 by Vudentz

ratijas, 2025-10-01:
> [image]

it took only 4 days, huh

> That is a lot better than most tools to be honest, try that on something like ip(8).

yeah, well, "others are even worse" is a decent rationale, I guess, maybe

> And Im not really sure is the help --help suppose to do here

at the very least not get stuck?  take a look at actual enterprise tools, for example `docker`:

```
❯ docker help
Usage:  docker [OPTIONS] COMMAND

A self-sufficient runtime for containers

[...wall of text...]

Run 'docker COMMAND --help' for more information on a command.

For more help on how to use Docker, head to https://docs.docker.com/go/guides/

❯ docker help --help
Usage:  docker help [command]

Help about the command
```

Vudentz, 2025-10-01:
> > <img alt="Image" width="579" height="110" src="https://private-user-images.githubusercontent.com/6737986/496373016-0089e38b-6cad-47d9-a701-314b7e7ad702.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTkzNTMyNjMsIm5iZiI6MTc1OTM1Mjk2MywicGF0aCI6Ii82NzM3OTg2LzQ5NjM3MzAxNi0wMDg5ZTM4Yi02Y2FkLTQ3ZDktYTcwMS0zMTRiN2U3YWQ3MDIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MTAwMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTEwMDFUMjEwOTIzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9N2YzNzBkN2Q1Y2Q4NmVkNTViZDUyNDMyZDU0ZWFhZGQ0ZjVmOWYzZTU4OTczMGY2OWMzMjg1YTljNDVkMDJkNCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.e43SQkCMjkZdcn0R8Dk3Rpdg_RZy-XrU4esznrzHn-0">
> 
> it took only 4 days, huh
> 
> > That is a lot better than most tools to be honest, try that on something like ip(8).
> 
> yeah, well, "others are even worse" is a decent rationale, I guess, maybe
> 
> > And Im not really sure is the help --help suppose to do here
> 
> at the very least not get stuck? take a look at actual enterprise tools, for example `docker`:

That is a bug, not lack of documentation.

> ```
> ❯ docker help
> Usage:  docker [OPTIONS] COMMAND
>
> A self-sufficient runtime for containers
>
> [...wall of text...]
>
> Run 'docker COMMAND --help' for more information on a command.
>
> For more help on [...]

Really, you want to compare BlueZ with docker? 

https://www.docker.com/ - 20M+ monthly developers

ratijas, 2025-10-01:
> That is a bug, not lack of documentation.

I never said this issue is about lack of documentation (although I can't help but admit that it could be interpreted as implied)