bluez/bluez #1466: Part of gatt data is lost.

takeshi-dodo, 2025-07-30:
### problem details
- part of gatt data is lost if I try to send one packet larger than 512byte with D-bus API "AcquireWrite".
  - https://github.com/bluez/bluez/blob/5.72/doc/org.bluez.GattCharacteristic.rst#fd-uint16-acquirewritedict-options-optional

### example
- If I set 600 byte data with "AcquireWrite", I can send 512byte, but 88byte is lost.  

### cause
- It's because of the buffer size to read is 512 bytes in sock_read().
  - https://github.com/bluez/bluez/blob/770ad5614e7e8074133e6f563495ce4822f63fe4/src/gatt-client.c#L1118 
- target fd is datagram socket(SOCK_SEQPACKET).
  - https://github.com/bluez/bluez/blob/770ad5614e7e8074133e6f563495ce4822f63fe4/src/gatt-client.c#L1202  
- If all the data in one packet cannot be read in one recvmsg, the unread data will be lost.
  - https://github.com/bluez/bluez/blob/770ad5614e7e8074133e6f563495ce4822f63fe4/src/gatt-client.c#L1130 

### used version
- bluez 5.72

Vudentz, 2025-07-30:
> ### problem details
> 
>     * part of gatt data is lost if I try to send one packet larger than 512byte with D-bus API "AcquireWrite".
>       
>       * https://github.com/bluez/bluez/blob/5.72/doc/org.bluez.GattCharacteristic.rst#fd-uint16-acquirewritedict-options-optional
> 
> 
> ### example
> 
>     * If I set 600 byte data with "AcquireWrite", I can send 512byte, but 88byte is lost.
> 
> 
> ### cause
> 
>     * It's because of the buffer size to read is 512 bytes in sock_read().
>       
>       * [bluez/src/gatt-client.c](https://github.com/bluez/bluez/blob/770ad5614e7e8074133e6f563495ce4822f63fe4/src/gatt-client.c#L1118)
>             
>             
>                  Line 1118
>               in
>               [770ad56](/bluez/bluez/commit/770ad5614e7e8074133e6f563495ce4822f63fe4)
>             
>           
>           
>             
>         
>                 
>                   
>                    uint8_t buf[512];

That is the maximum size a GATT attribute can be:

```
The maximum length of an attribute value shall be 512 octets.
https://www.bluetooth.com/wp-content/uploads/Files/Specification/HTML/Core-61/out/en/host/attribute-protocol--att-.html#UUID-d24e0156-618e-728d-f19b-01349ced952b[]
```

> 
>     * target fd is datagram socket(SOCK_SEQPACKET).
>       
>       * [bluez/src/gatt-client.c](https://github.com/bluez/bluez/blob/770ad5614e7e8074133e6f563495ce4822f63fe4/src/gatt-client.c#L1202)
>             
>             
>                  Line 1202
>               in
>               [770ad56](/bluez/bluez/commit/770ad5614e7e8074133e6f563495ce4822f63fe4)
>             
>           
>           
>             
>         
>                 
>                   
>                    if (socketpair(AF_LOCAL, SOCK_SEQPACKET | SOCK_NONBLOCK | SOCK_CLOEXEC,
> 
>     * If all the data in one packet cannot be read in one recvmsg, the unread data will be lost.
>       
>       * [bluez/src/gatt-client.c](https://github.com/bluez/bluez/blob/770ad5614e7e8074133e6f563495ce4822f63fe4/src/gatt-client.c#L1130)
>             
>             
>                  Line 1130
>               in
>               [770ad56](/bluez/bluez/commit/770ad5614e7e8074133e6f563495ce4822f63fe4)
>             
>           
>           
>             
>         
>                 
>                   
>                    bytes_read = recvmsg(fd, &msg, MSG_DONTWAIT);
> 
> 
> ### used version
> 
>     * bluez 5.72

We could in theory fragment but that is debatable since this is going against the spec if the intent is to write more data than allowed by the spec, also if the idea is to have a internal protocol that does reconstruct the packets then perhaps you should limit the about of data sent in each chunk to 512 bytes.

takeshi-dodo, 2025-07-31:
I missed the following...
```
The maximum length of an attribute value shall be 512 octets.
https://www.bluetooth.com/wp-content/uploads/Files/Specification/HTML/Core-61/out/en/host/attribute-protocol--att-.html#UUID-d24e0156-618e-728d-f19b-01349ced952b[]
```

I was going to write more data than allowed by the specification.
So I will close this issue.

Thanks, @Vudentz san !!