bluez/bluez #1471: Unable to change A2DP SEP configuration since release 5.80

borine, 2025-08-02:
Since commit aa118e9 A2DP SetConfiguration fails for some remote devices when attempting to change the configuration of a connected SEP.

It appears that some (but not all) devices send the AVDTP_CLOSE accept response before they are yet able to apply a new configuration to the SEP. For such devices sending the SetConfiguration command immediately that the preceding CLOSE has been acknowledged causes the SetConfiguration to be rejected ("already in use"), and as a consequence the A2DP profile connection is closed.

The above mentioned commit says, vaguely, that "the remote end ... may end up disconnecting the session" without giving any concrete examples or referencing any bug reports. In fact, the commit actually **causes** exactly the behavior it claims to be fixing when used with some remote devices.

My own examples are selt-built sinks using Bluez with a Broadcom BCM20702A1 USB adapter and a Realtek RTL rtl8761 USB adapter. Note that the same hosts, running the same software, do not exhibit this issue when used with other adapters, so the issue seems specific to the remote host BT adapter (or firmware). There is also an example commercial speaker device that fails with Bluez 5.80 and later, see arkq/bluez-alsa#774

So, please can commit aa118e9 be reverted? If the original delay of 500ms really does cause problems for some remote devices, then perhaps a shorter delay of 100ms would be acceptable? In my own testing 100ms is sufficient to allow the SetConfiguration command to succeed.

borine, 2025-08-02:
> Do you have a btmon trace to show when it fails?

Attached two logs. The "fail" log is 100% repeatable for a specific sink using using Bluez latest master branch on the A2DP source. The "success" log is using the latest master branch with profiles/audio/a2dp.c patched to wait 100ms after the close is acknowledged before sending the Set Configuration request, and the same specific sink; this is also 100% reproducible.

[setconfig-fail.btsnoop.log](https://github.com/user-attachments/files/21558005/setconfig-fail.btsnoop.log)

[setconfig-success.btsnoop.log](https://github.com/user-attachments/files/21558012/setconfig-success.btsnoop.log)

borine, 2025-09-02:
comment  to remove stale label: this issue is still vali,; latest master branch does not fix it.

pv, 2025-09-05:
One difference in the logs is that in the failing case, AVDTP: Set Configuration is sent before L2CAP: Disconnection Response for CID: 65 which seems to correspond to AVDTP: Open. In the successful case the disconnection completes first.

`profiles/audio/avdtp.c:avdtp_close_resp / transport_cb` seems to wait for the channel HUP before sep->cfm->close() --> a2dp.c:close_cfm() --> setup_reconfigure() but not sure now if kernel sends HUP when sending L2CAP: Disconnection Request, or when L2CAP: Disconnection Response. Looks like the former?

AVDTP v1.3 Fig. 6.13 is relevant, but not clear if spec requires L2CAP channel close is acked before considering stream IDLE.

I guess the 500ms timeout could be put back as a workaround...

pv, 2025-09-14:
@borine: Try https://github.com/pv/bluez/commit/6c7bfeca9b2d5e8853f146134854a45ed60fabb8

borine, 2025-09-14:
So things are changing faster than I can keep up. I've just built from commit pv/bluez@5bf7137 and all is still working.

borine, 2025-09-14:
I can confirm that pv@82a0bae is working correctly.