bluez/bluez #1472: Dualshock 3 pairing with multiple adapters

rijnhard, 2025-08-02:
This is probably a niche case.
But on my steam deck the built in Bluetooth hits a range limit pretty quickly, so the idea is to use a Pluggable BT Adapter and pair the DS3 controllers to that instead of the built in steam BT adapter.

The problem (and I may be missing something silly) is that it always pairs against the built in adapter, even if I set the default controller in bluetoothctl. I've bumble around with disabling the built in adapter as well to no success.

So any idea how to do this?

Incase it matters, I'm using ClassicBondedOnly=false. But no other tweaks above the stock steam os image.

rijnhard, 2025-09-03:
Pretty sure this is a bug... Happy to gather more information just need a bit of guidance

Vudentz, 2025-10-01:
@rijnhard not really sure what you are after here, does the DS3 support multiple pairings? I guess not since I don't think it has any sort of interface for the user to select what it should reconnect to. You can probably redo a cable pairing each time though, is that what you are after?