bluez/bluez #1480: AVRCP volume control fail on BlueZ 5.83

xihua13104, 2025-08-07:
DUT environment：BlueZ 5.83+pipewire 1.3.83+kernel 6.12
Audio source: RedMI K70 phone

AVRCP volume control failed on DUT side.
```shell
bluetoothd[2322]: [:1.6:method_call] > org.freedesktop.DBus.Properties.Set [#35]
bluetoothd[2322]: profiles/audio/transport.c:media_transport_set_volume() Transport /org/bluez/hci0/dev_FC_5B_8C_F7_5D_FB/fd1 level 100
bluetoothd[2322]: profiles/audio/avrcp.c:avrcp_set_volume() EVENT_VOLUME_CHANGED not supported
bluetoothd[2322]: profiles/audio/transport.c:set_volume() Unable to set volume: Operation not supported (-95)
bluetoothd[2322]: [signal] org.freedesktop.DBus.Properties.PropertiesChanged
bluetoothd[2322]: [:1.6:error] < org.bluez.Error.Failed [#35]
``` 

We rejected the EVENT_VOLUME_CHANGED RegisterNotification from smart phone for some reason.

```shell
> ACL Data RX: Handle 51 flags 0x02 dlen 22               #248 [hci0] 19.782276
      Channel: 67 len 18 [PSM 23 mode Basic (0x00)] {chan 3}
      AVCTP Control: Command: type 0x00 label 0 PID 0x110e
        AV/C: Notify: address 0x48 opcode 0x00
          Subunit: Panel
          Opcode: Vendor Dependent
          Company ID: 0x001958
          AVRCP: RegisterNotification pt Single len 0x0005
            EventID: 0x0d (EVENT_VOLUME_CHANGED)
            Interval: 0x00000000 (0 seconds)
bluetoothd[2322]: = profiles/audio/avrcp.c:handle_vendordep_pdu()..   19.782755
bluetoothd[2322]: = profiles/audio/avrcp.c:avrcp_handle_register_..   19.782907
bluetoothd[2322]: < ACL Data TX: Ha.. flags 0x00 dlen 18  #249 [hci0] 19.782961
      Channel: 71 len 14 [PSM 23 mode Basic (0x00)] {chan 3}
      AVCTP Control: Response: type 0x00 label 0 PID 0x110e
        AV/C: Rejected: address 0x48 opcode 0x00
          Subunit: Panel
          Opcode: Vendor Dependent
          Company ID: 0x001958
          AVRCP: RegisterNotification pt Single len 0x0001
            Error: 0x01 (Invalid Parameter)
``` 

[btmon.txt](https://github.com/user-attachments/files/21654842/btmon.txt)

[btsnoop.log](https://github.com/user-attachments/files/21654845/btsnoop.log)

[bluetoothd.log](https://github.com/user-attachments/files/21654834/bluetoothd.log)

thePromger, 2025-09-26:
@xihua13104 

Hi,
Can you share the code to register the app for listening avrcp events/commands directly instead of using mpris?

xihua13104, 2025-09-26:
Hi @thePromger ,

I got this issue with bluetoothctl. I believe this is a compatibility issue of Bluez, it can only be reproduced on some mobile phones

thePromger, 2025-09-26:
@xihua13104 

Actually I was trying to listen for avrcp events/commands in python
In gui desktop, it works fine which uses mpris but it doesn't work in headless os (no gui, just CLI)

Do you have any code to listen for avrcp commands/events using bluez?

xihua13104, 2025-09-26:
Hi @thePromger 

In Bluez, avrcp cmd/evts is encapsulated in the org.bluez.MediaPlayer1 interface. To listen to avrcp evts, you need to listen to the org.bluez.MediaPlayer1 property change signal. The following is sample code in C:

```c

void lm_player_update_property(lm_player_t *player,
        const char *property_name, GVariant *property_value)
{
    lm_log_debug(TAG, "%s property_name:%s",  __func__, property_name);
    if (g_str_equal(property_name, MEDIA_PLAYER_PROPERTY_DEVICE)) {
        if (player->device_path)
            g_free((gpointer)player->device_path);
        player->device_path = g_strdup(g_variant_get_string(property_value, NULL));
        g_assert(g_variant_is_object_path(player->device_path));
        lm_log_info(TAG, "device path '%s'", player->device_path);
    } else if (g_str_equal(property_name, MEDIA_PLAYER_PROPERTY_NAME)) {
        if (player->name)
            g_free((gpointer)player->name);
        player->name = g_strdup(g_variant_get_string(property_value, NULL));
        lm_log_info(TAG, "name '%s'", player->name);
    } else if (g_str_equal(property_name, MEDIA_PLAYER_PROPERTY_TYPE)) {
        if (player->type)
            g_free((gpointer)player->type);
        player->type = g_strdup(g_variant_get_string(property_value, NULL));
        lm_log_info(TAG, "type '%s'", player->type);
    } else if (g_str_equal(property_name, MEDIA_PLAYER_PROPERTY_STATUS)) {
        if (player->status)
            g_free((gpointer)player->status);
        player->status = g_strdup(g_variant_get_string(property_value, NULL));
        lm_log_info(TAG, "status '%s'", player->status);
        if (lm_device_get_active_player(player->device) == player) {
            lm_player_status_change_ind_t ind = {
                .player = player
            };
            lm_app_event_callback(LM_PLAYER_STATUS_CHANGE_IND, LM_STATUS_SUCCESS, &ind);
        }
    } else if (g_str_equal(property_name, MEDIA_PLAYER_PROPERTY_POSITION)) {
        player->position = g_variant_get_uint32(property_value);
        lm_log_debug(TAG, "position %d", player->position);
    } else if (g_str_equal(property_name, MEDIA_PLAYER_PROPERTY_TRACK)) {
        GVariantIter track_iter;
        g_variant_iter_init(&track_iter, property_value);
        char *track_key = NULL;
        GVariant *track_value = NULL;
        while (g_variant_iter_loop(&track_iter, "{sv}", &track_key, &track_value)) {
            if (g_str_equal(track_key, "Title")) {
                if (player->track->title)
                    g_free((gpointer)player->track->title);
                player->track->title = g_strdup(g_variant_get_string(track_value, NULL));
                lm_log_info(TAG, "title name '%s'", player->track->title);
            } else if (g_str_equal(track_key, "Artist")) {
                if (player->track->artist)
                    g_free((gpointer)player->track->artist);
                player->track->artist = g_strdup(g_variant_get_string(track_value, NULL));
                lm_log_debug(TAG, "artist name '%s'", player->track->artist);
            } else if (g_str_equal(track_key, "Album")) {
                if (player->track->album)
                    g_free((gpointer)player->track->album);
                player->track->album = g_strdup(g_variant_get_string(track_value, NULL));
                lm_log_debug(TAG, "album name '%s'", player->track->album);
            } else if (g_str_equal(track_key, "Genre")) {
                if (player->track->gerneral_name)
                    g_free((gpointer)player->track->gerneral_name);
                player->track->gerneral_name = g_strdup(g_variant_get_string(track_value, NULL));
                lm_log_debug(TAG, "gerneral name '%s'", player->track->gerneral_name);
            } else if (g_str_equal(track_key, "NumberOfTracks")) {
                player->track->number_of_tracks = g_variant_get_uint32(track_value);
                lm_log_debug(TAG, "number of tracks 0x%x", player->track->number_of_tracks);
            } else if (g_str_equal(track_key, "TrackNumber")) {
[... 59 lines cut ...]
                lm_player_update_property(player, property_name, property_value);
            }
            lm_device_update_active_player(device);
        }
    }

    if (interfaces)
        g_variant_iter_free(interfaces);
}

static void lm_device_subscribe_signal(lm_device_t *device)
{
   
    device->iface_added = g_dbus_connection_signal_subscribe(device->dbus_conn,
                                                            BLUEZ_DBUS,
                                                            INTERFACE_OBJECT_MANAGER,
                                                            OBJECT_MANAGER_SIGNAL_INTERFACE_ADDED,
                                                            NULL,
                                                            NULL,
                                                            G_DBUS_SIGNAL_FLAGS_NONE,
                                                            on_interface_appeared,
                                                            device,
                                                            NULL);
}
```

thePromger, 2025-09-26:
> Hi [@thePromger](https://github.com/thePromger)
> 
> In Bluez, avrcp cmd/evts is encapsulated in the org.bluez.MediaPlayer1 interface. To listen to avrcp evts, you need to listen to the org.bluez.MediaPlayer1 property change signal. The following is sample code in C:
> 

@xihua13104 
Is it possible to get python code, i don't know c & i tried converting this command but didnt received any events from bluez in my python script

thePromger, 2025-09-26:
I even tried dbus-monitor on org.bluez.MediaPlayer1 interface but unable to receive any avrcp events

When tried on org.bluez.MediaControl1 interface, I was able to receive some events like connection status

@xihua13104